o-gs/dji-firmware-tools #78: Phantom P3P - Motor Obstructed even with new motor

ChosenZero, 2018-06-03:
Hey Guys, 

I have a Phantom pro that was in a low accident but once it hit the floor upside down it went in at full speed. now when I go to use the drone it says motor obstructed. I took the drone apart and I switched the motors and it has the same issue, then i switched it with another CCW motor from the other side of the Mother board and i'm having the same issue. 

This tells me there is something wrong with some chip either the mosfets or something. I am pretty new to the whole electronic side of the business but I want to learn and I feel comfortable soldering and desoldering. Anyways I tested the voltages and everything matches on the M1 M3 and M4 - but when the drone starts I am getting .5 volts at M2 ( which is the motor in question, ) while the rest of the motors are running and at near 8 volts. 

Does anyone have schematics with voltages info? or any idea of what I should test? 

I purchased some new mosfetts and a new driver - but I want to make sure that is the issue before I start removing and reinstalling some chips. 

Thanks a lot.

KreativeLabs, 2018-06-26:
Maybe the ribbon cable has some problems? You should try it with a new cable.

gpulch, 2018-08-23:
Hello, 
I’m having a similar problem with a P3A, may I ask how it turned out for you ?

Thanks!

KreativeLabs, 2018-08-24:
Ignore my previous comment, I thought it was regarding the gimbal. I managed to repair some "obstructed motor" problems on P3P, P3S and usually it was caused by a small damage to some electronics near that arm. Once a prop hit the shell so hard that it punched a hole in it and swooshed away some SMD parts. I had to desolder those parts from another board to make the damaged ESC board work again.