o-gs/dji-firmware-tools #80: p3p firmware on p3a

fomox, 2018-07-27:
So my p3a has gotten p3p firmware on it, are these tools able to help me install p3a firmware on the drone again? if so, how?

mefistotelis, 2018-07-27:
> So my p3a has gotten p3p firmware on it

I've seen many people claim they did it, but as of now noone really did.
Not that it's really hard, it just does not happen without specific actions taken.

For the tools usefulness - depends on what you want to achieve. I think the project wiki may be more useful for you; you need to do some research to figure out what is the real issue.

fomox, 2018-08-11:
sorry for the late reply. When i connect it to a phone (galaxy s8) it shows up as a phantom 3 professional, while the stickers on the aircraft says advanced and it worked as an advanced before. The only thing that isnt working is the camera, its still possible to control the gimbal. I think this makes sence since the only difference between the p3p and the p3a is the camera?

mefistotelis, 2023-01-09:
Not much info here, closing.

closed 2023-01-09 by mefistotelis