o-gs/dji-firmware-tools #81: "Response on parameter 287 indicates end of list despite larger size reported"

sincoder, 2018-07-30:
when use the og tool to list param on Mavic Air ,I got a lot "Response on parameter 471 indicates end of list despite larger size reported"  Message, I think those params are  the param that DJI  do not want them to be public, and need high privilege to read them, maybe we can find a way to bypass it
 

> 
> g_config.flying_limit.radius_limit_enabled|g_config.advanced_function.radius_limit_enabled default = 0 range = < 0 .. 1 >
> Response on parameter 471 indicates end of list despite larger size reported.
> Response on parameter 472 indicates end of list despite larger size reported.
> Response on parameter 473 indicates end of list despite larger size reported.
> g_config.flying_limit.min_height default = 20 range = < 1 .. 20 >
> min_height_user default = 5 range = < 1 .. 20 >
> g_config.flying_limit.user_avoid_ground_enable default = 1 range = < 0 .. 1 >
> g_config.flying_limit.avoid_ground_and_smart_landing_enable default = 1 range = < 0 .. 1 >
> Response on parameter 478 indicates end of list despite larger size reported.
> Response on parameter 479 indicates end of list despite larger size reported.
> limit_height_rel default = 5.000000 range = < 0.000000 .. 30.000000 >
> Response on parameter 481 indicates end of list despite larger size reported.
> Response on parameter 482 indicates end of list despite larger size reported.
> g_config.flying_limit.roof_limit_enable default = 1 range = < 0 .. 1 >

mefistotelis, 2018-08-13:
Specific parameters are "hidden" somehow in Mavic Air. At some point we will look at the FW code and write proper support for these.

jacargentina, 2020-07-11:
@mefistotelis similar/same on the Mavic Air 2
Any advances on adding support for those?

mefistotelis, 2020-07-11:
I'm working on something completely different right now. No ETA from me.

Carlo-Cat, 2020-10-18:
We still waiting for support...;-)