o-gs/dji-firmware-tools #82: GL300a Firmware Flashing

empatid, 2018-08-02:
I have flashed gimbal top board, but i could not find img files of gl300a. I tried various type of OpenSSL decrypte command but all given me error "bad magic number". Please someone help me about how can i find image files for GL300a board. Or how can decrypt orijinal dji bin files. If some have image files for gl300a could share with me? I am trying for 3 weeks, i tried almost all instructions on web but i am done. Thank you.

mefistotelis, 2018-08-02:
So how far exactly are you.

You've downloaded one of C1_FW files from here:
https://github.com/o-gs/dji-firmware-tools/blob/master/supported_firmwares.csv

Then what?

mefistotelis, 2018-08-10:
try idenifying the file format; it's probably not what you expect. Use `file` command, plus look at the header directly.

empatid, 2018-08-11:
I checked one, its starting with `xV4` , i am going to use your `dji_xv4_fwcon.py` and `dji_imah_fwsig.py`, i hope i can achive and get uboot images. Do you have any suggestion else?

closed 2018-08-13 by mefistotelis