o-gs/dji-firmware-tools #90: Using the tools on Ubuntu 16.04

mefistotelis, 2018-10-15:
The tools currently require Python 3.6, and Ubuntu 16.04 only has Python 3.5 in its repositories.

To update the Python version and install necessary packages for hardcoder tools:

```
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt-get update
sudo apt-get install python3.7 python3.7-gdbm
sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.7 30
sudo update-alternatives --install /usr/bin/python3 python3 /usr/bin/python3.5 50
curl https://bootstrap.pypa.io/get-pip.py | sudo python3.7
# Python 3.7 is now functional enough to switch to it
sudo update-alternatives --config python3
# (select Python 3.7 from the list; select back in case of issues)
sudo -H pip3 install keystone-engine
sudo -H pip3 install capstone
```

gaifeng8864, 2020-09-23:
When I use this command ./test_hardcoders_p3x.sh P3X_FW_V01.11.0030.bin to test the firmware of P3X_FW_V01.11.0030.bin, after running for a period of time, an error is reported: Error: name'Ks' is not defined. I am currently running ubuntu16.04.7 x64 system. I installed all the components according to the steps you have in this link https://github.com/o-gs/dji-firmware-tools/issues/90, I have looked for it for a long time, no Solve the problem. Where is this "Ks" named? Where is this error? Attached is the test record.

mefistotelis, 2020-09-23:
Ks is keystone-engine. Did you installed 'keystone' instead?

gaifeng8864, 2020-09-23:
> Ks is keystone-engine. Did you installed 'keystone' instead?

What is certain is that I installed keystone and keystone-engine. And I also suspect that it is a keystone problem, and I compiled and installed it according to the official steps on github. But the problem remains. How does python call keystone? Do I need to make some settings for the connection between python and keystone, such as library files?

mefistotelis, 2020-09-23:
> What is certain is that I installed keystone and keystone-engine.

So if you "import keystone" in a python script, what will be used?

Installing conflicting packages is not a good idea.

gaifeng8864, 2020-09-23:
> > What is certain is that I installed keystone and keystone-engine.
> 
> So if you "import keystone" in a python script, what will be used?
> 
> Installing conflicting packages is not a good idea.
> 
> I just updated the tool so it finds such problem earlier. Try that.

Okay, I will test the new tool, and I will give you feedback if there is any problem.

gaifeng8864, 2020-09-23:
> Okay, I will test the new tool, and I will give you feedback if there is any problem.

I downloaded the latest toolkit you provided, and uninstalled the previously installed keystone and keystone-engine. Then I re-installed keystone-engine according to the above steps you provided. Now they work well. What a great job.
If use the m900.bin module generated after the test, rename it to P1765Fw3.bin. Can I directly use the SD card to write into the drone to increase the launch power of the drone and increase the remote control distance? If the above scheme is effective, do I need to add the <PTF>_FW_DEBUG file to the SD card when flashing? Or do I need to increase the version number deliberately?

mefistotelis, 2020-09-23:
You may increase version number just to be sure that the firmware you modded is in use.

Flashing single module should work. Whether you need a debug file - not sure; try without it, check log. If it refused to flash, add the *_FW_DEBUG file.

gaifeng8864, 2020-09-23:
> You may increase version number just to be sure that the firmware you modded is in use.
> 
> Flashing single module should work. Whether you need a debug file - not sure; try without it, check log. If it refused to flash, add the *_FW_DEBUG file.

Is there a way to check the version of each module in the firmware directly on the drone?

mefistotelis, 2020-09-23:
No automated way to query each module ATM. Though that would be nice functionality.

But for each module you can go with this:
https://github.com/o-gs/dji-firmware-tools#comm_serialtalkpy

gaifeng8864, 2020-09-23:
> No automated way to query every module ATM. Though that would be nice functionality.
> 
> But for each module you can go with this:
> https://github.com/o-gs/dji-firmware-tools#comm_serialtalkpy

Okay, thank you very much for your selfless dedication. I will continue to study the use of related tools. To be honest, although I am not a professional, I am very fascinated by it. I am eager to improve the remote control distance by changing the firmware on my P3A. Although it seems that these operations are unnecessary at times, I never get tired of them.

mefistotelis, 2020-09-23:
Good to hear that. Good luck!

svarteld, 2022-06-12:
> Okay, thank you very much for your selfless dedication. I will continue to study the use of related tools. To be honest, although I am not a professional, I am very fascinated by it. I am eager to [...]

@gaifeng8864 Did you get the m0900 module to add some range/RF power? Details would be appreciated. Looking to do similar things on WM610. Thanks!

gaifeng8864, 2022-06-12:
> 

Unfortunately, I didn't find a solution. I'm already about to give up.

gaifeng8864, 2022-06-12:
> @gaifeng8864 Did you get the m0900 module to add some range/RF power? Details would be appreciated. Looking to do similar things on WM610. Thanks!

Unfortunately, I didn't find a solution. I'm already about to give up.

svarteld, 2022-06-12:
@gaifeng8864 sorry to hear that. Did you manage to extract the human-readable text parameters from m0900? Would be interesting if you could show them.

gaifeng8864, 2022-06-13:
> 

> @gaifeng8864 sorry to hear that. Did you manage to extract the human-readable text parameters from m0900? Would be interesting if you could show them.

I haven't done this exploration in a long time and some have forgotten. I remember that the m900 module can adjust the flight control related parameters, right? For example, height limits, geofencing, etc. If yes, then follow @mefistotelis's tutorial to get various parameters easily.

mefistotelis, 2022-06-14:
```
# grep 'name' ./P3X_FW_V01.08.0080_m0900.json
                "name" : "og_hardcoded.lightbridge_stm32.packet_received_attenuation_override"
                "name" : "og_hardcoded.lightbridge_stm32.packet_received_attenuation_value"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad4_attenuation_tx1_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad4_attenuation_tx2_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad5_attenuation_tx1_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad5_attenuation_tx2_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar6_attenuation_tx1_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar6_attenuation_tx2_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad2_attenuation_tx1_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad2_attenuation_tx2_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar7_attenuation_tx1_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar7_attenuation_tx2_ce"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad4_attenuation_tx1_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad4_attenuation_tx2_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad5_attenuation_tx1_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad5_attenuation_tx2_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar6_attenuation_tx1_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar6_attenuation_tx2_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad2_attenuation_tx1_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ad2_attenuation_tx2_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar7_attenuation_tx1_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.board_ar7_attenuation_tx2_fcc"
                "name" : "og_hardcoded.lightbridge_stm32.mcu_firmware_version"
```

Remember to share the problem and the solution for capstone.

svarteld, 2022-06-14:
Thanks, interesting parameters, promising.

Is it possible for me to get the whole .json file? Preferable from WM610_FC550R_FW_V01.11.01.50.bin, it's here:

https://www.dropbox.com/s/g3qg5bc8w34fc19/WM610_FC550R_FW_V01.11.01.50.bin?dl=0

Will post documentation and any solutions! Thanks once again.

gaifeng8864, 2022-06-15:
> @gaifeng8864 flight params are actually in the m0306 module; flashed those many times, my Inspire 1's are fast and safe now, transformed. Chased military water vehicles recently; soon in air too. Better not crash into those, pretty expensive UAVs...
> 
> Looking to adjust Lightbridge params in the m0900 now; will be plenty of RF disturbances around those military UAVs.
> 
> @mefistotelis, thanks so much for the pointers to #90, I've still not managed to fix the capstone error I get with lightbridge_stm32_hardcoder.py. If you've got time, could you possibly be so kind to show the human-readable output from these?
> 
> WM610_FW_V01.11.01.50_m0900.bin arm_bin2elf.py lightbridge_stm32_hardcoder.py
> 
> Just so I know if there's interesting parameters there. Thanks again for all the work you're sharing.

I'm very sorry, I confused the function of the module. With your reminder, I remembered. I have adjusted the RF parameters of the video transmission on my P3A. The script did not prompt any errors, at least the simple parameter modification should be successful. But I tested it and it doesn't have any effect on the flight distance. I have tried many parameter adjustments, but all have no effect. I think there are two possibilities: 1. The parameter modified in the script is wrong, at least an invalid parameter. 2. If you want to change the performance of image transmission, such as the remote control distance that everyone cares about, you need to change it on the air side and the remote control side at the same time. In the second solution, the m1400 module on the remote control (I think it should be it) cannot modify the parameters under this script. I have tried many times, but it should be that some parameters of the reverse engineering are wrong, and it has not been successful. I'm not an expert on this, so I haven't been able to figure this out. This is also the reason why I finally gave up on modifying the parameters of the image transmission, which is too shocking.