o-gs/dji-firmware-tools #91: phantom 3 standard  07 00 Device not detected

Oscar1983, 2018-10-27:
Hi to all !
I have a DJI p3 standard siting over 2 years. I had a crash my son acidently switch motors of in mid air . The drone fell abaut 15 meters hight , the camera riped of from the gimball , i replaced the broken part's the yaw arm and the flat cable , i repaired my self .
But after the drone dont conects to the remote  i mean the 5.8ghz link is fine drone stil can fly , i have find out that maybe the AR 9342 lost his software due to crash i have tryed updating the firmware with sd card on the drone but every time same 0700 Device not detected !
Than i conected the serial pins of the board to an arduino  and with putty an nothing not a glitch i do the same to the remote but here i used a 3,3volt usb ttl an there worked i mean u boot was loaded fine i dont know where is the problem is maybe my arduino not good becuse is an uno r3 clone with ch 340 chip or the nand loses firmware i dont know but i find out on this site is maybe a solution for problem that i short out pins on the gimball top board BOOT_O with 3V3 and maybe i can upload a mising firmware in to the board somehowe but i need some instructions from you guys i mean i preshiate your work you do a nice job hacking these chinese drones to the bone so if someone is out there that has time an will to help me out that my son can fly again with his drone im a poor man from serbia hey i sold my yugo to by this toy for him ... sory for my english is not the best so if there anyone to want help me out sory i know my post isnt belongs here but i need help ! ASAP

Oscar1983, 2018-10-27:
Ok how can i upload the m0700 ? which sofware i need ?

Oscar1983, 2018-10-27:
i didnt find the m0700 firmware on this site where can i else download ?

jiekchenid, 2018-12-05:
you can wechat me work out together 379826168

jppw, 2021-06-14:
@mefistotelis 

So can you maybe point us how to extract the m0700 firmware out of the firmware update so we can flash it directly to the board?

jppw, 2021-06-14:
[](https://phantompilots.com/threads/07-00-00-device-not-detected-phantom-3-standard.147965/post-1480215)

jppw, 2021-06-14:
[Found something...](https://phantompilots.com/threads/07-00-00-device-not-detected-phantom-3-standard.147965/post-1480215)