o-gs/dji-firmware-tools #92: Looking for BOOTSEL pad on Inspire 1 X3 DM368 board

okstuv, 2018-11-03:
I have a Inspire 1 and x3 cam recently lost video feed to the app. It takes pix and video to SD, but doesn't stream live video to the app. Sound familiar?

Since I've repaired tons of Phantom 2 Vision Plus video transmission modules by reflashing the bootloader on the DM368 board, I suspected it was the same issue on the x3. 

After some soldering, I've confirmed that it's the same issue:

`DM36x initialization passed!
UBL Product Vesion : DJI-NOHDMI-WM610-UBL-1.0-rc0(2015-01-13)
Dji UBL Version: 1.51(Jan 13 2015 - 16:42:11)
Booting Catalog Boot Loader
BootMode = NAND
Starting NAND Copy...
Valid magicnum, 0xA1ACED66, found in block 0x00000019.
Valid magicnum, 0xA1ACED66, found in block 0x0000001D.
No valid boot image found!
NAND Boot failed.
Aborting...`

The problem is I can't find what pads to jumper in order to put the X3's DM368 board into BOOT ME mode. Has anyone succeeded in this before to your knowledge? If so, please ping me so I can wrap this up and report back on the outcome. Thanks!

priegor, 2018-11-03:
Did you try short 3v3 with BOOT pins on Sdcard board?

okstuv, 2018-11-03:
@priegor thanks for the reply and idea.

Yeah. I tried each of those boot pads (yellow in pic) to  3v3 pad (red in pic) one at a time. I also tried jumping them together. None of those 3 variations put the DM368 board into BOOT ME mode...

[Link to SD card board of x3 cam]
(https://photos.app.goo.gl/XoAVt64poTzTdFzq8)

okstuv, 2018-11-03:
@mefistotelis thanks for confirming it's same as PH3 in that regard. 

I've tied BOOT_S to the 3v3 that was highlighted in red in my prior pic, but no luck seeing boot me on the terminal, just the same "bad bootloader" message...

Additionally, under the heatsink on the 4th side of the cube of circuitboards there are these pads:
(https://photos.app.goo.gl/EQHfSpMZA5YNCDYe9)

I've tried tying B01 and B00 to VCC3v3 and still no "BOOT ME" prompt.  Also tried tying B01 and B00 together with no luck.

Any chance these don't get tied to 3v3? Or that it matter which 3v3 they tie to (I can't imagine that would be the case...) 

BTW, My technique for getting to / looking for BOOT ME is the same as I have in my tutorial for repairing Phantom 2 Vision Plus Wifi Modules:
(https://sites.google.com/view/phantom-2-wifi-module-fix/home)