o-gs/dji-firmware-tools #94: P3 gimbal FPC connector

vakarcs, 2018-11-05:
I think the FPC connector on WL320/321 Gimbal top boards is an Omron XF2B-3545-31A connector. https://omronfs.omron.com/en_US/ecb/products/pdf/en-xf2b.pdf

pawelsky, 2018-11-05:
It is not. It uses completely different latch mechanism...

vakarcs, 2018-11-05:
I am speaking about this connector here:
[image]
And this is the Omron XF2B connector:
[image]
Both rotary backlock. Dimensions also match, to the extent of my measurement accuracy...

pawelsky, 2018-11-05:
Oh, I see, you may be right about this one.

I thought you were talking about the one on the other side of the board.

mefistotelis, 2018-11-05:
Great!
Added to parts list on wikis for both Gimbal top boards.
Thank you.

closed 2018-11-05 by mefistotelis