o-gs/dji-firmware-tools #99: Calibration Gimbal Phantom 4 Series

jebtir, 2018-12-11:
HI, I am from Brazil and I am looking for a person who sell the Software for calibration of Gimbal's DJI. I say calibration, when you change some module ESC of the engines. Example: ESC Yaw/Roll Gimbal Phantom 4 or Phantom 4 Pro and the Gimbal goes crooked. I saw that you made a gimbal Spark calibration Software available, would you sell some to calibrate gimbal from the other Drones?

mefistotelis, 2018-12-11:
The method for calibrating each model is probably different. I don't know how to do this exactly, that would require some research. I'm not sure if I'll be looking into that.

Spark has a pre-made auto calibration function which is easy to trigger. Previous platforms seem to have less automated solution, but I never looked at details.

jebtir, 2018-12-11:
And how much would you charge to make a script for calibration of Gimbal's Phantom 4 series, Phantom 3 and perhaps inspire? I pay you to do that, I don't want it for free!

jebtir, 2018-12-11:
The site to be faced with stability issues. I can not access the same! Could you pass your email or WhatsApp? I will help in what possible (testing with drones). About the donations, I will donate yes.

mefistotelis, 2018-12-11:
The "hardware" channel on Slack is the place where many people discuss mods and repairs for DJI products.

But if you can't access it - we can talk here. I don't use any other messaging apps.

jebtir, 2018-12-11:
I'm trying to access it, but I can't. What do you need to give the tip foot in the software for calibration? I have all the drones here to test....

mefistotelis, 2018-12-11:
First I will look at gimbal module (m0400) for various platforms. I have the ones for Ph3, Ph4 and Inspire, but I don't think I have one for Ph2 Vision - and it could be helpful.

I will be searching for m0400 modules which are larger in size - this may indicate a version with more debugging messages left in the code.

I need to get a basic understanding of how the firmware works, then I'll get to figuring out the packets which can be sent to it.

jebtir, 2018-12-11:
I have a great friend who also program in Python, if you need any other help, I can talk to him... I can help with the tests, since I have no advanced knowledge of software.

mefistotelis, 2018-12-11:
It will take some time before I have anything to test. I still need to finish current tasks.

Anyway, at some point I'll want you to run comm_serialtalk.py - so you may want to prepare for it, to the point where the example instruction from README works.

And you can try searching Ph2 firmwares for me.

jebtir, 2018-12-11:
Here on the website of DJI has some firmware of the Phantom 2 Plus which is the latest version of PH2

https://www.dji.com/phantom-2-vision-plus/info#downloads

mefistotelis, 2018-12-26:
Here are ALL packets on which Phantom 3 Gimbal reacts. The table comes from version P3X_FW_V01.11.0030_m0400:

| function | cmdid | cmdset |
| -------- | ----- | ------ |
| packet_exec_gimb_push_aetr |	0x06 | Gimbal |
| packet_exec_gimb_control |	0x01 | Gimbal |
| packet_exec_spec_special_app_control |	0x01 | Special |
| packet_exec_flyc_posture |	0x42 | FlightCtrl |
| packet_exec_gimb_ext_ctrl_accel |	0x0C | Gimbal |
| packet_exec_gimb_thirdp_magn |	0x0E | Gimbal |
| packet_exec_gimb_calibration |	0x08 | Gimbal |
| packet_exec_gimb_adjust_roll |	0x07 | Gimbal |
| packet_exec_gener_version_inquiry |	0x01 | Universal |
| packet_exec_gener_enter_loader |	0x07 | Universal |
| packet_exec_gener_update_confirm |	0x08 | Universal |
| packet_exec_gener_update_transmit |	0x09 | Universal |
| packet_exec_gener_update_finish |	0x0A | Universal |
| packet_exec_gener_reboot_chip |	0x0B | Universal |
| packet_exec_gener_get_device_state |	0x0C | Universal |
| packet_exec_gener_set_device_version |	0x0D | Universal |
| packet_exec_gener_encrypt_config |	0x30 | Universal |
| packet_exec_gimb_cmd20 |	0x20 | Gimbal |
| packet_exec_gimb_ext_ctrl_degree |	0x0A | Gimbal |
| packet_exec_gimb_get_ext_ctrl_status |	0x0B | Gimbal |
| packet_exec_gimb_suspend_resume |	0x0D | Gimbal |
| packet_exec_gimb_get_user_param |	0x10 | Gimbal |
| packet_exec_gimb_set_user_param |	0x0F | Gimbal |
| packet_exec_gimb_save_user_param |	0x11 | Gimbal |
| packet_exec_gimb_cmd21 |	0x21 | Gimbal |
| packet_exec_gimb_resume_default_param |	0x13 | Gimbal |
| packet_exec_gimb_cmd14 | 0x14 |	Gimbal |

Now we can test what each of these does and find out which is used to calibration or position setting.
Some we already know - I will update then with descriptions soon.

mefistotelis, 2018-12-29:
Disclamer: This can damage the gimbal. I don't know what it does. I will just provide packets which look like they might trigger calibration or control the yaw arm.

Now, first check if taking to gimbal even works (this command is safe):

```
./comm_serialtalk.py COM5 -vv --receiver_type=Gimbal --seq_num=12341 --ack_type=ACK_After_Exec --cmd_set=General --cmd_id=1
```

If it does (versions are shown in response), then we can get to business.

I will provide proper commands as soon as I finish figuring them out. The idea I have is to enter work mode 5 - it looks like a different version of auto-calibration routine (which is work mode 3).

jebtir, 2018-12-29:
Hey, buddy. How you doing, huh? Very good! I will test this command on some Phantom 3

mefistotelis, 2018-12-29:
Please do.

My idea of switching to work mode 5 is not as easy as I anticipated - all packets which allow to control work mode are masking it to 2 bits (allowing only modes 0..3).

I am searching for alternatives.

mefistotelis, 2019-01-09:
Update:
I got all firmwares for Ph2, am looking at them now; will get back to looking at Ph3 gimbal code when I'm done with them.

jebtir, 2019-01-09:
Tell me, my friend, how was your year-end? I wish you much health, peace and a lot of money that year. I arrived on a trip today, I was at the relatives ' house. I'm going to test the Phantom 3 code tomorrow, and I'll give you a return.

jebtir, 2019-01-25:
Hi, how are you? I went to work. I'm going to take the test this week on a Phantom 3 Pro. It would be interesting if I had any other way to talk to you, I have a good proposition for this project.

mefistotelis, 2019-01-25:
You can write me an e-mail.
mefistotelis@gmail.com

I am still working on pre-Ph3 firmwares; have limited time for that, so it takes quite long.

cimona, 2019-01-25:
can be contacted via viber or whatsApp or messenger. +79219329356
https://www.facebook.com/profile.php?id=100011587383998
cimona84@list.ru  

>
>Hi, how are you? I went to work. I'm going to take the test this week on a Phantom 3 Pro. It would be interesting if I had any other way to talk to you, I have a good proposition for this project.

jebtir, 2019-04-18:
Hey, did you abandon the  "project? "

mefistotelis, 2019-04-18:
I am not working on it now. May be back when I have more time.

TheonlyAnkit, 2019-06-30:
Hello everyone, 
I always repair drones, this time I have one issue in Mavic air. Mavic air started showing gimbal motor overload, I replaced the gimbal by new one, I saw the Mavic air mainboard and the old gimbal had been paired when in production. Any other gimbal will not work until the paired mainboard and gimbal both replaced. But I saw new gimbal is working but not calibrating, it’s tilt left and looks upward slightly. There may be some kind program or software like thing which can calibrate this new gimbal and the gimbal will look straight. Any help or suggestion will be highly appreciated.

0n3-70uch, 2019-09-21:
Hi, did anyone found a solution? I replaced my YAW motor ad now I need to calibrate the motor.

TheonlyAnkit, 2019-09-21:
Yes, the complete solution is here. You just have to run one simple python
program in root unit of Mavic air drone. You can find one video over
YouTube

0n3-70uch, 2019-09-21:
Thanks for your quick answere.
Ok thats for mavic air right? Is there a solution for Phantom 4 Pro?

0n3-70uch, 2019-09-21:
PS: Can you link me the video pls. Many wrong videos there...

TheonlyAnkit, 2019-09-21:
Let me check, i will send here

0n3-70uch, 2019-09-21:
Great thanks. I found this video for the spark: https://www.youtube.com/watch?v=LdQtvJjjqeI
This solution works for the mavic air too... but the PH4PRO is different.

0n3-70uch, 2019-09-21:
> Disclamer: This can damage the gimbal. I don't know what it does. I will just provide packets which look like they might trigger calibration or control the yaw arm.
>
> Now, first check if taking to [...]

Here is my response:
```
Prepared binary packet:
55 0d 04 33 0a 04 35 30 40 00 01 91 94
Sending packet...
Waiting for reply...
Retrieved 16 packets (886b), dropped 0 fragments (0b)
Received response packet:
55 2c 04 36 04 0a 35 30 80 00 01 00 01 6e 6f 6e 65 00 b3 09 f2 36 35 00 55 66 3f 00 00 00 00 0c 01 11 0d 32 01 01 00 00 00 01 96 98
Parsed response  - DJIPayload_General_VersionInquiryRe:
        unknown0: 0
        unknown1: 1
      hw_version: b'none'
     ldr_version: 17563648
     app_version: 20057361
       unknown1A: 1
       unknown1E: 1
```

TheonlyAnkit, 2019-09-21:
Yes correct this is the video, just follow the steps in the video
carefully. This will work for Mavic air too. This works for me when I
installed new gimbal

0n3-70uch, 2019-09-21:
Yeah this works for Spark and Mavic Air.. but not for Phantom 4. 
The topic ist "Calibration Gimbal Phantom 4 Series". Thats why I ask for the PH4PRO ;-).

I used following trick now:
https://youtu.be/uWCm0QlHTTM

TheonlyAnkit, 2019-09-21:
You have replaced the complete motor or only winding part??

0n3-70uch, 2019-09-21:
Only the winding part. The cable was demaged. Now it's running fine and well centered.

TheonlyAnkit, 2019-09-21:
Okay, let me know any help you need

0n3-70uch, 2019-09-21:
Ok thank you. Have a nice weekend.

lensherm, 2019-09-21:
Well well.... Although calibration kept starting but timing out on my Spark, regardless of the combinations of components I tried it with, including original gimbal/cam, different ribbon cable, brand new gimbal motor, brand new gimbal/cam; as soon as I got a replacement board, it worked like a charm. Looks like gremlins were out to get me and my 90 bucks :-). At least all's well with the world in the end.

0n3-70uch, 2019-09-21:
Yeah the motor drivers are placed on the core board. Very bad solution... The same situation like the mavic air.

TheonlyAnkit, 2019-09-21:
Keep the original magnet over the winding when replace the motor. Also it
Depends upon quality of winding whether the motor will work or not in case
of phantom 4 pro

vserena82, 2020-02-15:
Hello, there is a solution for this problem?
I need to calibrate gimbal in P4Pro, I replaced yaw motor? On crash magnet of old yaw motor has broken so I can't use it with new winding part.
There is a way to calibrate gimbal or change its  default parameteres (angles)? after replacing motors?
Thank you

TheonlyAnkit, 2020-02-18:
Sorry for the late reply, I was busy in some work.

Make sure the original magnet of yaw motor is fine. And replace the motor
winding by new one. After installation of new winding part of yaw motor,
make sure every part is fit correctly. When you start your p4 pro the
gimbal/camera may be not in center.
Don’t worry, That will be in center you just have to do several
calibration— gimbal calibration in several position.

Thanks

vserena82, 2020-02-18:
Thank you for your answer but I cant swap magnet: winding bumped into magnet ring and magnet has been broken, so I cant use it.
I tryed with another yaw motor but it is not in correct position.
I have also another gimbal to repair: I changed esc board for yaw motor, also in this case yaw motor is in wrong position.
Is it possible to make a "deep" calibration using some scripts?
Or is it possible to force yaw parameters manually on fw?

If there are no solutions I have to buy others yaw motors and hope the will be in correct position...

Thank you

TheonlyAnkit, 2020-03-27:
Any issues you are facing?

sh88hoseini, 2020-05-21:
Hello
I also encountered this problem, but I can't open this help link.

TheonlyAnkit, 2020-05-31:
Sorry for delayed in response, due to pandemic   My work is paused for a
while.

Can you elaborate what issues you are facing?

fargnoli1969, 2020-06-01:
Hi does anyone here able to write a simple program to calibrate after replacing esc or motors each or any of the dji range ?? Donations for programs on their way if you can help . Thanks 
Fargnoli@scotlandunmanned.co.uk

TheonlyAnkit, 2020-08-01:
Hie everyone, is there any programming script to calibrate the P4 Pro gimbal after replacing some damaged/fractured spares from the gimbal??

krixiicsz, 2021-02-16:
Hello! I have tried to find some information, but haven't got anything. Now as I understand, that I had to replace magnets from old motor to new, but I have already put my old motor in trash bin. So. Is there any other solution that can work to center my gimbal. My drone have crashed, I have replaced yaw motor, flex cable and also esc board for gimbal. Is there some software where can manually calibrate motor?

mEfis-afk, 2021-08-07:
> Hi does anyone here able to write a simple program to calibrate after replacing esc or motors each or any of the dji range ?? Donations for programs on their way if you can help . Thanks
> [Fargnoli@scotlandunmanned.co.uk](mailto:Fargnoli@scotlandunmanned.co.uk)

How much are you selling for?