o-gs/dji-firmware-tools #103: Help with Log Files

Steven22728, 2019-01-16:
Hi,

I have a team building a flight risk management platform and we are at the stage where we are decoding the DJI flight logs. This does not appear to be an easy task, but i see you have some code that looks like it decodes the log.

Would you be interested in assisting for a small fee?

Kind Regards
Steve

Weanuts, 2019-01-16:
Most of what I gleaned was offered by mefistotelis, as you must have 
seen on the web.

I am afraid I do even know (basically) what I performed, albeit it 
worked. So to try and help I believe, would be fruitless endeavor.

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

Steven22728, 2019-01-16:
ok, I appreciate your time in responding.  Kind Regards Sp

kovar-ursa, 2019-01-16:
Greetings,

We provide this as a SaaS offering if you are interested. Our API is here:

https://ursa-inc.github.io/API/

We have a major commercial customer using it and the same engine is in use in other products.

-David

David Kovar
650-278-1774
www.ursasecure.com
dkovar@ursasecure.com

Steven22728, 2019-01-16:
Ive dropped you an email

https://ursa-inc.github.io/API/

dkovar@ursasecure.com<mailto:dkovar@ursasecure.com>