o-gs/dji-firmware-tools #105: Button combinations

bonaime, 2019-01-31:
Hi,
I found a button combinations in a forum that is not describe in the Wiki. it is : 
Press and hold the C1 button and the shutter, then press the ignition button
The main led with turn blue. 

I don't know what is this mode. 

Can you add this to the wiki ?

mefistotelis, 2019-01-31:
I assume you mean GL300 page:
https://github.com/o-gs/dji-firmware-tools/wiki/GL300-Main-board#button-combinations

1. Did you tested the combination?
2. What is ignition button? you mean RC power button?
3. is there any audio queue played when you use that combination?

bonaime, 2019-01-31:
Yes it is this wiki page.
1) Yes I tested the combination with a GL300C
2) ignition is RC power button. 
3) It is the same sound as when you power the RC

mefistotelis, 2019-01-31:
Thank you, I will add it.

mefistotelis, 2019-02-14:
Added.

closed 2019-02-14 by mefistotelis