o-gs/dji-firmware-tools #108: phantom 4 esc/motor upgrade to bigger using pwm signal for esc??????

sakis08, 2019-02-18:


sakis08, 2019-02-18:
please any body know how to get pwm signal from p4 fc to use the frame as fc with djgo fuctions but with diffrent motors and escs????

mefistotelis, 2019-02-18:
Besides PWM, Flight Controller in DJI drones connects to ESC via uart. To make custom ESC work, you'd have to talk to FC like original ESCs do.

So finding the PWM pin should be the easiest part, you just need a multimeter and datasheet of the motor control chip (to know which pin is the PWM input).

sakis08, 2019-02-18:
> Besides PWM, Flight Controller in DJI drones connects to ESC via uart. To make custom ESC work, you'd have to talk to FC like original ESCs do.
>
> So finding the PWM pin should be the easiest part, [...]

on p4 fc there is pins for pwm but im not sure if its working alone or beside uart tx and rx because p4 esc has more then one pin 
lets say i found the pwm signal out from fc 
what is the next step ?

and thanks alot for replaying

sakis08, 2019-02-18:
my project is to use p4 frame as flight controller and LT2 osd waypoint etc fuctions on bigger motors and esc !!!!???

mefistotelis, 2019-02-18:
Look at the ESC boards:
https://github.com/o-gs/dji-firmware-tools/wiki/WM330-ESC-Center-Left-Right-board

At bottom, you'll find a Dji-branded microcontroller. It's really some variant of TMS320F. We know the exact model for Ph3 (you can find it on ph3 esc wiki), but not for ph4.

That uC talks to the FC via UART. Some of the messages they exchange are contained within Wireshark dissectors.

You either need to program your own uC to mimic these, or tap the uC from Dji ESC to your own motor driver.
Not so easy either way.

sakis08, 2019-02-18:
looks hard to do that
p3p is the same ?or harder

sakis08, 2019-02-18:
any idea of wich esc has already uart ?

mefistotelis, 2019-02-19:
> p3p is the same ?or harder

p3p is almost the same as p4, internally. Components are just placed on different shape boards.

closed 2021-06-15 by mefistotelis