o-gs/dji-firmware-tools #112: DJI Go App will not stay on screen long enough

Weanuts, 2019-03-08:
Enclosed  a brief explanation of DJI Go app not staying on screen.

https://youtu.be/TgxjnEJDKMg

Weanuts, 2019-03-08:
Does this log look OK?

[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/2944682/P3X_FW_LOG_AB.txt)

mefistotelis, 2019-03-08:
Looks ok to me.

If the app crashes, I'd look for an issue on mobile device side.

Weanuts, 2019-03-09:
It turns out it was a soft usb connection at a controller. Tried 4 cables and finally got one to stay connected.

Now that I can see the version(s) screen, I see I have (+) beside my downgraded 1.7.60 firmware downgrade. Could you please help with suggestion(s) on getting rid of this problem. 
thank you once again

mefistotelis, 2019-03-09:
Why is it a problem?

If you want to get rid of it for whatever reason, first check which module is not in the same version which fw 1.7.60 provides.

Weanuts, 2019-03-09:
Imu is not right. If I can re calibrate without trying to remove the + I would rather do that way.

Thanks John

Weanuts, 2019-03-09:
Ok re calibrated. Appears ok.  

Thank you again for all your assistance.

John