o-gs/dji-firmware-tools #113: identifying correct board.

Cooperjf, 2019-03-10:
Complete rookie here so go easy on me,
My Sons phantom 3 standard crashed into shallow water, Wifi module or Camera failed to work since.
Everything else is working fine.
I dissembled and cleaned up the circuit boards, connectors and gimbal components.

Does the Gimbal/ Camera board also control the wifi? I see a MAC number stamped on the outside of the casing?
any help greatly appreciated.

mefistotelis, 2019-03-10:
> Does the Gimbal/ Camera board also control the wifi?

There are two radio channels in Ph3 Std. Video feed one is completely on gimbal. This is why you connect two of the antennas to [Gimbal Top Board](https://github.com/o-gs/dji-firmware-tools/wiki/WM321-Gimbal-top-board).

Cooperjf, 2019-03-10:
Fantastic! Thanks for the reply.

This is the board that I suspected to be damaged however the gimbal is still functional.
And I have no wifi signal.

closed 2021-06-15 by mefistotelis