o-gs/dji-firmware-tools #114: DJI Mavic Air gimbal calibration

davidhaluska, 2019-03-14:
Is there possibility to add support for service gimbal calibration after replacement for Mavic Air? The same functionality is already available for SPARK but need this for Mavic Air. Thanks.

supreme2, 2019-03-15:
I‘m also looking for the same solution. Camera Error 3 after replacing Gambal & camera. DJI‘s only offer is sending it in. Assistent 2 doesn’t bring the answer also. Hope someone is in a good mood to help us.

bunchofbradys, 2019-03-31:
Heyo there, just tested it out and calibrated the gimbal on Mavic Air without issue. If you run the GimbalCoarse calibration with MAVIC then it calibrated no worries. Have just installed a new gimbal myself and was searching for calibration software, this worked perfectly.

mefistotelis, 2019-03-31:
Thank you for sharing; good to know it works. For Mavic Air, you should use WM230 as platform code (though currently it follows the same path as MAVIC).

bunchofbradys, 2019-03-31:
Is separation of VM230 and MAVIC (and their other equivalents) something that is going to occur in the future?

mefistotelis, 2019-03-31:
It's WM. Yes it might occur. The current path doesn't work for Mavic.

davidhaluska, 2019-04-16:
Bunchofbradys, could you please confirm how you performed calibration? You run the same command as for SPAK for gimbal coarse? Or was there needed something to modify? Thanks.

bunchofbradys, 2019-04-16:
@davidhaluska I used the MAVIC as the platform code (use WM230 as discussed above) and the Gimbal Coarse calibration which worked with my Mavic Air without and trouble.

davidhaluska, 2019-04-16:
Could you please specify how to use WM230 as platform code? How I need to configure that? Thanks.

mefistotelis, 2019-04-16:
Command line help and readme should give you everything you need:
```
./comm_og_service_tool.py --help
```
https://github.com/o-gs/dji-firmware-tools#comm_og_service_toolpy

Lawesslee, 2019-05-24:
Hello there;
You can use the repair tools program for DJI spark and mavic air models.
Drive link; https://drive.google.com/open?id=1nJ59_l6FTYC9brDbShZC_V4D9JWRVy7L
[image]

fargnoli1969, 2019-05-29:
Hi is this still working ?? I had to install a new maincore board on mavic air and cant get the gimbal to work now with the new board . Anyone ??

Lawesslee, 2019-05-30:
Yes program is running. You can calibrate your mavic air device after the new gimbal tactics. but there is a situation. you need to use new product. If you use the used gimbal, there may be an activation problem.

fargnoli1969, 2019-05-31:
Hi it keeps failing . Have you got any instructions ?? How to setup and use this tool ?? 
Also i have changed the maincore board but my used camera and cant get to work . Camera is skewed but was working with older board . 
Thanks

Lawesslee, 2019-05-31:
no need to install the program. Run the repair tool exe and press the join coarse calib button from the gimbal calib section. after a while the calibration will finish but it will give an error. This program is designed for DJI Spark model working in MAVIC air.

zew302, 2019-06-04:
Hi, I replaced gimbal purchased from Aliexpress. After use RepairTool 2.2.0.7 gimbal working but in DjiGo app is attention with camera error 3. Is there any way how to reset errors in Mavic Air?

Lawesslee, 2019-06-04:
The newly purchased gimballer runs smoothly. But it doesn't work on gimballer mavic air models that are used or disassembled on the device. According to the information I received recently, the dji has made such an application that the dismantled gimballer sold in the market does not work.

zew302, 2019-06-05:
It doesn't make sense because gimbal and camera works perfectly. Only problem is error attention on DjiGo app. There must be any solution to write serial number of camera and gimbal to main board on Mavic Air.

mefistotelis, 2019-06-05:
Just modify the firmware to ignore the error and not set the error flag in packet for the application.
That way it won't matter whether the part is new or used, whether it's SN pairing or encryption pairing - it will just work.

This is what we did in Ph3 gimbal encryption pairing (Ambarella-DaVinci). The tool dm3xx_encode_usb_hardcoder.py can be used there to modify DaVinci firmware so that it sets "success" no matter what real verfication result was.

djicypservice, 2019-06-09:
Just like zew302 said, its been a while almost 2 months we have been waiting solution for Activation Error after replacing dji mavic air gimbal with the used(already activated device) gimbal. Dji go 4 app in andriod rarely requires activation, however the app on ios restrictedly ask for activation after raplacing gimbal or core board on Mavic air.  
We figure out that we can calibrate gimbals by using Repair Tool that dji released for dji sparks. Even if the program gives fail error, calibration works as perfect as we expect it. 

it was nice to see that you gave us a light of possibility in case @mefistotelis. 

Would you please help us how to modify firmware so that we pass through the activation screen. Because everything works good. U can record video u can fly it but cant see the live feed. 

Thanks in advanced for ur helps. 

Any resources how to modify firmware accordingly would be really helpfull if you can provide us.

mefistotelis, 2019-06-09:
I never analyzed Mavic Air firmware. Steps would be:
1. Capture communication between the drone and mobile device
2. Figure out which packets contains the information about activation issue
3. Analyze the firmware, and find that packet/information there
4. Figure out the variable which involved into that issue, and where it is set
5. Change the function which sets it so that error is never visible
That will allow fixing the issue for one specific firmware version. To fix it for all versions, you'll also need:
6. Write a tool similar to existing 'hardcoders' which finds specific assembly pattern in the firmware and modifies it.

jebtir, 2019-08-20:
> Olá; 
> Você pode usar o programa de ferramentas de reparo para modelos de ar de faísca e de mavic DJI. 
> Link da unidade; https://drive.google.com/open?id=1nJ59_l6FTYC9brDbShZC_V4D9JWRVy7L
> [image]

Any way to get in touch with you?

djicypservice, 2019-08-20:
Ofcourse.

Here is my phone numer. You can reach me out from whatsapp.

+90 533 336 93 15

şunu yazdı:

guewald, 2019-09-18:
have anyone solved the activation issue?

zew302, 2019-09-18:
I'm still waiting

robxs1, 2019-10-20:
Re: activation issue.
While this may not help, these are the ways i know to resolve the issue
1. write the s/n from a new non-activated gimbal to your used gimbal ( flash tool)
2. use the dji tool  - this is a tool dji released for the MA & M2 - not easy to get but it is alive.
3. replace with new .. which sometimes does not work.
..if there is other ways, i would be very keen to know.

guewald, 2019-10-20:
> Re: activation issue.
> While this may not help, these are the ways i know to resolve the issue
>
> 1. write the s/n from a new non-activated gimbal to your used gimbal ( flash tool)
> 2. use the dji tool [...]

Do you have this DJI TOOL?

joww1, 2019-10-20:
What tool are you talking about? And do you have any other tool to make the
flash of SN?

joww1, 2019-10-20:
> Re: questão de ativação.
> Embora isso possa não ajudar, estas são as maneiras que conheço para resolver o problema
> 
> 1. escreva s / n de um novo cardan não ativado no seu cardan usado (ferramenta flash)
> 2. use a ferramenta dji - esta é uma ferramenta dji lançada para o MA & M2 - não é fácil de obter, mas está viva.
> 3. substitua por novo .. que às vezes não funciona. 
>    ..se houver outras maneiras, eu gostaria de saber.

What tool are you talking about? And do you have any other tool to make the flash of SN?

robxs1, 2019-10-20:
No, but there is a guy on slack who may help https://dji-rev.slack.com/archives/C64R5L1HN/p1566640302001700

davidhaluska, 2020-01-26:
> No, but there is a guy on slack who may help https://dji-rev.slack.com/archives/C64R5L1HN/p1566640302001700

I'm unable to login to that slack workspace any idea how to get in touch with that guy?

akome-29, 2020-02-21:
> I‘m also looking for the same solution. Camera Error 3 after replacing Gambal & camera. DJI‘s only offer is sending it in. Assistent 2 doesn’t bring the answer also. Hope someone is in a good mood to help us.

Dismantle the camera sensor and return the previous sensor...

akome-29, 2020-02-21:
> Just like zew302 said, its been a while almost 2 months we have been waiting solution for Activation Error after replacing dji mavic air gimbal with the used(already activated device) gimbal. Dji go [...]

Just return the previous camera sensor and install it to a new gimbal motor

Menz82, 2020-03-01:
hello guys, I loaded the program Repair Tool V2 from the Google folder.  tried to start the repairtoolV2.exe.  when I click on the exe, windows want me to allow the program, I click yes.  but nothing happens and does not start.  can someone help me?

Menz82, 2020-03-01:
The Repairtool works perfectly with my Mavic Air! 

Problem is camera error 3. I can‘t downgrade from V01.00.0620. 

Does anyone have a solution?

BensTV, 2020-03-12:
> The Repairtool works perfectly with my Mavic Air!
> 
> Problem is camera error 3. I can‘t downgrade from V01.00.0620.
> 
> Does anyone have a solution?

Take a look here:
https://phantompilots.com/threads/old-software-links-etc-downgrades.123249/

BensTV, 2020-03-13:
> Just like zew302 said, its been a while almost 2 months we have been waiting solution for Activation Error after replacing dji mavic air gimbal with the used(already activated device) gimbal. Dji go [...]

Hi Guys.
Is there any new informations about this issue?
I just replaced the Gimbal &  Camera but i can't bypass the activation pupup in the Dji App.
Thanks in advance !!

Menz82, 2020-03-13:
Has anyone tried this with the bug with Dumldore 3.0?  
Would like to try the days down from the current firmware to get the Camera Error 3 away!

Menz82, 2020-03-15:
I tried to downgrade from v01.00.0620 to 01.00.0520 and 01.00.0500 with Dumldore 3.20 not possible! 
I was hoping to delete the camera error 3 with it,
unfortunately did not work

mauriciocruzdrones, 2020-06-30:
Hola. La solución es usar el sensor de la cámara original en el gimbal nuevo. Hay que desarmar el gimbal nuevo, quitar el sensor de la cámara nuevo y colocar el sensor de la cámara original. Luego volver a armar el gimbal. De esta manera se soluciona el problema.
Saludos cordiales.
Soy mauriciocruzdrones.com

nettunoroberto, 2020-09-06:
> I tried to downgrade from v01.00.0620 to 01.00.0520 and 01.00.0500 with Dumldore 3.20 not possible!
> I was hoping to delete the camera error 3 with it,
> unfortunately did not work

You have found any solutions for this error?

mingtaoxin, 2020-09-06:
Just buy or find free vpn and dealer software DJI , and you can erase error code 3..

mingtaoxin, 2020-09-06:
Look software on ebay

nettunoroberto, 2020-09-06:
> Just buy or find free vpn and dealer software DJI , and you can erase error code 3..

Can you help me?

nettunoroberto, 2020-09-06:
> Look software on ebay

My searches on the web and on ebay have not been successful

mingtaoxin, 2020-09-06:
http://rover.ebay.com/rover/1/711-53200-19255-0/1?icep_ff3=2&pub=5575378759&campid=5338273189&customid=&icep_item=402360616745&ipn=psmain&icep_vectorid=229466&kwid=902099&mtid=824&kw=lg&toolid=11111

joww1, 2020-09-06:
> [http://rover.ebay.com/rover/1/711-53200-19255-0/1?icep_ff3=2&pub=5575378759&campid=5338273189&customid=&icep_=402360616745&ipn=psmain&icep_vectoridto](http://rover.ebay.com/rover/1/711-53200-19255-0/1?icep_ff3=2&pub=5575378759&campid=5338273189&customid=&icep_item=402360616745&ipn=psmain&icep_vectorid=229466&kwid=902099&mtid=824&kw=lg&toolid=11111)

Do you have WhatsApp to contact or email?

mingtaoxin, 2020-09-06:
I just share link, so i not super expereinced in this tools

joww1, 2020-09-06:
> Acabei de compartilhar o link, então não tenho muita experiência com essas ferramentas

I know. I was supposed to ask you about something else. Anyway, does this tool work on Mavic Air and Mavic mini? or just mavic 2?

joww1, 2020-09-09:
Has anyone managed to take this VPN and use it?

coptersafe, 2020-10-20:
https://www.ebay.com/itm/402504779633

pilgrimage80, 2020-11-05:
Mavic mini bought a gimbal from aliexpress. I cannot calibrate this gimbal. Can anyone fix this problem?

robxs1, 2020-11-05:
The tool works. Note that the mini port closes in 10seconds. You need to have the script ready to go as soon as the mini boots. Check the device manager and you will see the usb port open and close. otherwise,  Its possible that the video cable, gimbal or board is faulty. In our experience the refurbished mini parts are 50% defective.

pilgrimage80, 2020-11-05:
My gimbal was working fine. Suddenly the gimbal got stuck. Then there was no image on the camera. I just changed the Mavic mini Ptz cable. Does this prevent the Gimbal from working correctly?

https://tr.aliexpress.com/item/4001109332615.html?spm=a2g0s.9042311.0.0.3d734c4d7Hd7Hr

pilgrimage80, 2020-11-05:
If you mean the repair tool, DJI is giving a virtual com error. Why is it giving this? It appears as USB Serial (COM4) in the blue Mini Ports section. Repair Tool does not work. How did you make it work?

robxs1, 2020-11-05:
if no image, then replace cable. Its easy to damage the core board when replacing this cable, be careful and check connection to core board. fix video issue 1st.  When video & the gimbal is working, but gimbal is not pointing straight, check your drivers / reinstall tool etc & calibrate.

pilgrimage80, 2020-11-06:
I changed my cable. no problem with image transfer. The problem is that the Gimbal does not move when I enter gimbal calibration commands using dji firmwaretool. However, when the mavic mini is first opened, the gimbal movement moves left and right. It just doesn't move more or less at all. What could be the reason for this?

When I enter this command, it acts as follows.
py comm_og_service_tool.py COM4 WM160 GimbalCalib JointCoarse

https://www.youtube.com/watch?v=4Lw4iCAnmy0&ab_channel=AlperAYKUT

However, when I enter this command, it doesn't do anything.
py comm_og_service_tool.py COM4 WM160 GimbalCalib LinearHall

Is this problem gimbal issue or cable related? Couldn't solve this?

thankyou.

robxs1, 2020-11-06:
Before entering the second command you might need to reboot the drone and connect to the computer. The com port may have closed.

The start up sequence may not need the ptz cable to work 100%. So the issue could still be the cable. 

If the gimbal does not move with the drone movements, the issue could be 1. Ptz cable, 2. gimbal 3. 3:1 cable. 4. Core board connector. 
Carefully inspect the fine wires of the ptz cable , if any have a sharp angle they could be damaged.

mihadongle, 2021-01-15:
Do any one have the dji dealer software or have a link for me i can buy it

tecdrone, 2021-03-23:
> Alguém tem o software do revendedor dji ou tem um link para mim, posso comprá-lo?

Eu também preciso

rsrmdp, 2021-03-23:
https://github.com/o-gs/dji-firmware-tools

[13 more replies]