o-gs/dji-firmware-tools #116: P4P GIMBAL

fengzhonggeinichengnuo, 2019-03-27:
Hello, I am a drone repairman from China. I am not good at English communication, so I wrote this article with translation software. If the meaning is not accurate, please forgive me. The problem I asked is the universal alignment of the P4P series. I have seen that you have been very useful for the program designed by SPARK, but it is not suitable for three-axis cloud platforms such as P4P. I do not know much about software programming. I want to ask you if you can do some research in this area. In the question bar, I saw your previous conversation with another person. I don't know why you didn't respond. I also did some understanding. The Yaw axis data should exist in the IESC1000 on the tone. I tried to remove the camera part. Yaw can still maintain the correct position. This seems to prove that there is a fixed program in the iesc 1000 during the boot process that can directly control YAW. No need for STM32 intervention, If you can continue your research in this area, I can pay for it or provide a platform. If the research is completed, you can make a encryption tool and sell it to me. When you see it, please give me an answer. Thank you.

mefistotelis, 2019-03-27:
I am sorry, but my schedule is full - I have no time for a paid task. I am planning to get back to looking at the gimbals, but no idea when.

fengzhonggeinichengnuo, 2019-03-27:
Well, I respect your choice, and I 'd like to get the calibration tool for the P4P universal section, and after you're done with it, study the P4P universal section. You're the best technical engineer I 've seen on DJI firmware. Thank you. When you study this problem, please contact me. Thank you very much for your reply. If possible, could you tell me an approximate time.

jebtir, 2019-04-22:
> Bem, eu respeito sua escolha, e gostaria de obter a ferramenta de calibração para a seção universal P4P, e depois de terminar, estude a seção universal P4P. Você é o melhor engenheiro técnico que vi no firmware do DJI. Obrigado. Quando você estudar esse problema, entre em contato comigo. Muito obrigado pela sua resposta ... Se possível, você poderia me dizer um tempo aproximado.

I'm also looking for someone to do that. Did you get it?

0n3-70uch, 2019-09-21:
+1 Would be so great :-)

joww1, 2019-09-25:
> +1 seria tão bom :-)

What?

0n3-70uch, 2019-09-25:
> > +1 seria tão bom :-)
> 
> What?

A software solution to calibrate the gimbal.

closed 2021-03-10 by mefistotelis

tecdrone, 2021-04-14:
> Olá, sou um reparador de drones da China. Não sou bom em comunicação em inglês, então escrevi este artigo com um software de tradução. Se o significado não estiver correto, por favor, me perdoe. O problema que perguntei é o alinhamento universal da série P4P. Eu vi que você foi muito útil para o programa desenvolvido por SPARK, mas não é adequado para plataformas de nuvem de três eixos, como P4P. Não sei muito sobre programação de software. Eu gostaria de perguntar se você pode fazer alguma pesquisa nesta área. Na barra de perguntas, vi sua conversa anterior com outra pessoa. Não sei por que você não respondeu. Eu também fiz alguns entendimentos. Os dados do eixo Yaw devem existir no IESC1000 no tom. Tentei remover a parte da câmera. Yaw ainda consegue manter a posição correta. Isso parece provar que há um programa fixo no iesc 1000 durante o processo de inicialização que pode controlar diretamente o YAW. Não há necessidade de intervenção STM32, se você puder continuar sua pesquisa nesta área, posso pagar por isso ou fornecer uma plataforma. Se a pesquisa for concluída, você pode fazer uma ferramenta de criptografia e vendê-la para mim. Quando você vir, por favor me dê uma resposta. Obrigada.
您好，早安朋友。 需要您的幫助。還是可以出售您的知識來進行phantom 4系列萬向節校準？  porfavou我真的需要您的幫助。我從巴西說話，您可以通過電子郵件或whatsapp與我聯繫+55 8198734-0010 tecdronedji@gmail.com