o-gs/dji-firmware-tools #118: DJI Phantom 3 Pro: IMU calibration - unknown error at 2%

Fred0212, 2019-04-10:
Hello,
I am having a IMU problem with a Phantom 3 Pro. 
The IMU shows good readings on X and Y axis, but Z keeps showing -1.5. Calibration of the IMU progresses to 2% and displays "Unknown error".
Interestingly I am sometimes calibrate the IMU if I place it on "wrong" side, but this is not obviously a good option.

What I've tried so far with no success:
-Dismantling the IMU and giving it few good knocks.
-Half an hour in ultrasonic cleaner with IPA.

My next thought was to replace the MP-6500 MEMS component with a new one by soldering.

I would seek your advice: is replacing the MP-6500 a good option, any other options?
If replacing the MP-6500 does not work, what would the best source for completely new DJI IMU component?

mefistotelis, 2019-04-10:
I don't have experience with replacing IMU. Looks like there can be two causes of the issue:
- damaged MPU-6500
- damaged connection between IMU and FC

So replacing the MEMS chip might work. It isn't that hard to do, so you might try that.
If not - try buying a crashed drone. Most are crashed by user error, so IMU should be working, and you'll have a new Flat Film Wire. Note that IMUs of all Ph3s are the same - you may even use one from "standard".

Fred0212, 2019-04-11:
I think I will try to replace the MEMS chip once I get the replacement.
I was wondering what could be causing the "Unknown error" during the IMU calibration? Does anyone have any idea what are the requirements for this error message to appear? And is there any way to "reset" the calibration readings?

I was able to progress the IMU calibration beyond 2%, but it stalled at 44%. Now all the IMU axis readings are completely wrong.. Damn. I was thinking that zeroing the IMU calibration results could help to "start from clean table".

Any ideas and help is highly appreciated.

mefistotelis, 2019-04-11:
If you want to diagnose software, you have two ways:

1. Open DAT log downloaded from the AC and search for messages about reason of the error.
There are usually text messages logged in such cases.

2. Check which field in which packet causes the error to be shown in Dji Go, capture DUML communication from FC, then find the related packets in Wireshark.

In both cases, you'll then have more info about the error, and in case that isn't enough - you will know which packet type and field informs about it, chich opens way for further analysis, this time within FC firmware.

Fred0212, 2019-04-12:
After a bit of research (I am not that familiar with DJI products, terminology etc. takes a while to learn), I decided to try option 1 that @mefistotelis recommended.

I believe these are the relevant entries regarding the IMU problem from the .DAT log file I was able to extract:

``
7.187 :       0 IMU need temper cali 0 55
7.187 :       0 IMU need base cali 0
7.187 :       0 mis cali 55 55 99b7 fff0 cnt:0
7.187 :       0 imu status:3
7.187 :       0 [hardfault]:******************check fault info and trace ************
7.187 :       0 [hardfault]:-----fault is null: addr(0x40024000),flag(0xa7b689a0)-----
7.187 :       0 [hardfault]:-----fault is null: addr(0x400241a0),flag(0xdaf63f29)-----
7.187 :       0 [hardfault]:-----task info  is null: addr(0x40024340),flag(0xc50a222e)-----
7.187 :       0 [hardfault]:-----trace info  is null: addr(0x400244d0),flag(0xb5a17364)-----
7.187 :       0 [hardfault]:******************check last trace ******************
7.187 :       0 [hardfault]:-----trace info  is null: addr(0x40024790),flag(0x2d7a185d)-----
7.187 :       0 [hardfault]:-----wdg_time_info is null: addr(0x40024a50),flag(0x4917dd68)-----
7.187 :       0 startup:7.84244||
7.187 :       0 Board:"wm320v2"
7.192 :       0 imu err status 1003

9.212 :     101 [FDI AHRS[1]]:ahrs_init begin
9.227 :     102 [FDI AHRS[1]]:bias fdi turn on
9.227 :     102 [FDI AHRS[1]]:init fdi turn on
9.227 :     102 [FDI AHRS[1]]:wait for sensor check
9.407 :     111 [LED] changed: imu error

10.410 :     161 Eeprom write offset:1d0||
11.147 :     198 [FDI AHRS[1]]:(wxyz)gyro bias large!!!
11.147 :     198 [FDI AHRS[1]]:(accxyz)acc bias large!!!
12.708 :     276 sub_cmd:9,data_len:14

19.930 :     521 Eeprom write offset:1d0||
26.652 :     857 iwdg_set_max_timeout befor set swdg_timeout(0x00000120)!
26.652 :     857 iwdg_set_time_out time_out(0x00000ffa)!
26.652 :     857 iwdg_set_swdg_time_out time_out(4090) set g_swdg_timeout_max(0x000007ab) ||
26.653 :     857 iwdg_set_default_timeout befor set swdg_timeout(0x000007ab)!
26.653 :     857 iwdg_set_time_out time_out(0x00000a23)!
26.653 :     857 iwdg_set_swdg_time_out time_out(2595) set g_swdg_timeout_max(0x00000120) ||
36.732 :    1361 iwdg_set_max_timeout befor set swdg_timeout(0x00000120)!
36.732 :    1361 iwdg_set_time_out time_out(0x00000ffa)!
36.732 :    1361 iwdg_set_swdg_time_out time_out(4090) set g_swdg_timeout_max(0x000007ab) ||
36.733 :    1361 iwdg_set_default_timeout befor set swdg_timeout(0x000007ab)!
36.733 :    1361 iwdg_set_time_out time_out(0x00000a23)!
36.733 :    1361 iwdg_set_swdg_time_out time_out(2595) set g_swdg_timeout_max(0x00000120) ||
36.768 :    1361 iwdg_set_max_timeout befor set swdg_timeout(0x00000120)!
36.768 :    1361 iwdg_set_time_out time_out(0x00000ffa)!
36.768 :    1361 iwdg_set_swdg_time_out time_out(4090) set g_swdg_timeout_max(0x000007ab) ||
36.770 :    1361 iwdg_set_default_timeout befor set swdg_timeout(0x000007ab)!
36.770 :    1361 iwdg_set_time_out time_out(0x00000a23)!
36.770 :    1361 iwdg_set_swdg_time_out time_out(2595) set g_swdg_timeout_max(0x00000120) ||
36.805 :    1361 iwdg_set_max_timeout befor set swdg_timeout(0x00000120)!
36.805 :    1361 iwdg_set_time_out time_out(0x00000ffa)!
36.805 :    1361 iwdg_set_swdg_time_out time_out(4090) set g_swdg_timeout_max(0x000007ab) ||
36.807 :    1361 iwdg_set_default_timeout befor set swdg_timeout(0x000007ab)!
36.807 :    1361 iwdg_set_time_out time_out(0x00000a23)!
36.807 :    1361 iwdg_set_swdg_time_out time_out(2595) set g_swdg_timeout_max(0x00000120) ||
43.767 :    1710 [FDI AHRS[1]]:(accxyz)trans acc large when init``

Only things I am able to pick up from the log is that FDI AHRS seems to be something related to the IMU calibration. Also the accxyz (IMU accelerometer) values seem to have too large bias.

I do not know if the hardfault-reports at tick 7.187 have anything to do with the IMU problem? What about the iwdg_set-reports in the end of my pasted log?

Is anyone more professional able to pick up other clues from there?