o-gs/dji-firmware-tools #119: sys_hardcoder can't find values

nicolov, 2019-04-14:
Thanks a lot for the amazing work! I was preparing to extract the bitrate tables on my Zenmuse X5, but noticed that `amba_sys_hardcoder` doesn't work for the PX3 either. Here's what I did:

```
./dji_xv4_fwcon.py -v -x -p P3X_FW_V01.01.0008.bin
./amba_fwpak.py -v -x -m P3X_FW_V01.01.0008_m0100.bin
./amba_sys2elf.py -vv -e -l 0x6000000 -p P3X_FW_V01.01.0008_m0100_part_sys.a9s
./amba_sys_hardcoder.py -vv -x --elffile P3X_FW_V01.01.0008_m0100_part_sys.elf
```

and this is the output:

```
P3X_FW_V01.01.0008_m0100_part_sys.elf: Opening for extract
P3X_FW_V01.01.0008_m0100_part_sys.elf: Search for DjiMsgSettingsInit ver P3X_FW_V01.01, progress 100%
P3X_FW_V01.01.0008_m0100_part_sys.elf: Pattern of DjiMsgSettingsInit ver P3X_FW_V01.01 was not found; closest was 0 lines at 0x0
P3X_FW_V01.01.0008_m0100_part_sys.elf: Search for DjiUstVideoQualitySetInner ver P3X_FW_V01.01, progress 100%
P3X_FW_V01.01.0008_m0100_part_sys.elf: Pattern of DjiUstVideoQualitySetInner ver P3X_FW_V01.01 was not found; closest was 0 lines at 0x0
P3X_FW_V01.01.0008_m0100_part_sys.elf: Warning: No variant of function 'DjiMsgSettingsInit' was found
P3X_FW_V01.01.0008_m0100_part_sys.elf: Warning: No variant of function 'DjiUstVideoQualitySetInner' was found
P3X_FW_V01.01.0008_m0100_part_sys.elf: Warning: From expacted functions, 2 were not found
Traceback (most recent call last):
  File "./amba_sys_hardcoder.py", line 2746, in <module>
    main()
  File "./amba_sys_hardcoder.py", line 2726, in main
    armfw_elf_ambavals_extract(po, elffh)
  File "./amba_sys_hardcoder.py", line 2493, in armfw_elf_ambavals_extract
    raise ValueError("No known values found in ELF file.")
ValueError: No known values found in ELF file.
```

Thanks!

nicolov, 2019-04-14:
This is no longer needed - I was able to grep for the bitrate values using the disassembly you provided in https://github.com/o-gs/dji-firmware-tools/issues/49 - thanks a lot again.

mefistotelis, 2019-04-14:
Glad you've succeeded.

The issue with your command lines was that you didn't provided sys2elf binary with info about where is the border between `.text` and `.data` sections, and automatic detection returned invalid place. I don't know where the border is exactly in that firmware, but as long as the value is close to where is should be, the tool will work:
```
./amba_sys2elf.py -vv -e -l 0x6000000 --section .ARM.exidx@0x464800:0 -p P3X_FW_V01.01.0008_m0100_part_sys.a9s
```

closed 2019-04-14 by mefistotelis