o-gs/dji-firmware-tools #120: Inspire 1 bitrate change doesn't apply

nicolov, 2019-04-14:
Thanks again! I've managed to unpack, edit, and reflash the m0100 module on my Inspire with X5. I've hex-edited all occurrences of `25000000` (`4078 7d01`) to `60000000` (`0087 9303`) in the `m0100_part_sys.a9s` file.

I was expecting 1080p 24fps videos to now be recorded at 60mbps (from 25mbps), but there seems to be no change.  Do you happen to know if I'm missing anything?

mefistotelis, 2019-04-14:
Try using hardcoder tool - if it exports the table correctly, it will be much easier for you to edit it.

Here's an example from P3X_FW_V01.01.0006_m0100_part_sys - note that a single bit rate seem to be controlled by 3 values:
```
[
	{
		"index" : 0,
		"description" : "AuthorityLevel established when SHA204 communication fail; 0-restricted,1-normal,2-superuser",
		"minValue" : 0,
		"maxValue" : 2,
		"defaultValue" : 0,
		"setValue" : 1,
		"name" : "og_hardcoded.p3x_ambarella.encrypt_query_fail_authority_level"
	},
	{
		"index" : 1,
		"description" : "AuthorityLevel established when encryption keys match; 0-restricted,1-normal,2-superuser",
		"minValue" : 0,
		"maxValue" : 2,
		"defaultValue" : 1,
		"setValue" : 1,
		"name" : "og_hardcoded.p3x_ambarella.verify_state_good_authority_level"
	},
	{
		"index" : 2,
		"description" : "AuthorityLevel established on encryption mismatch; 0-restricted,1-normal,2-superuser",
		"minValue" : 0,
		"maxValue" : 2,
		"defaultValue" : 0,
		"setValue" : 1,
		"name" : "og_hardcoded.p3x_ambarella.verify_state_bad_authority_level"
	},
	{
		"index" : 3,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "30000000 45000000 60000000",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_00"
	},
	{
		"index" : 4,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "30000000 45000000 60000000",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_01"
	},
	{
		"index" : 5,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "30000000 45000000 60000000",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_02"
	},
	{
		"index" : 6,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "30000000 45000000 60000000",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_03"
	},
	{
		"index" : 7,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "30000000 45000000 60000000",
[... 155 lines cut ...]
		"index" : 27,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "1702256979 1702065440 1702043762",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_24"
	},
	{
		"index" : 28,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "1852404852 1948283751 1330782319",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_25"
	},
	{
		"index" : 29,
		"description" : "Bitrates used for h.264 video compression; 3 values: min, avg, max",
		"minValue" : "1000000 1000000 1000000",
		"maxValue" : "64000000 64000000 64000000",
		"setValue" : "1948269133 1768580128 1953723680",
		"name" : "og_hardcoded.p3x_ambarella.vid_setting_bitrates_26"
	}
]
```

nicolov, 2019-04-15:
I think I got the section offset right on my firmware (I used the `0405 0607 0809 0a0b 0c0d 0e0f 0a0b 0c0d` pattern in the dump file, is this a good idea?). However, the script couldn't find the functions, and I ended up writing a python script.

Sadly, it seems that the X5 can't go above 64 mbps (where did you get that magic number, btw?). If I set 80mbps, files are recorded at 15mbps (for all resolutions, so it seems to be some sort of fallback).

Still a big improvement, 1080p/24fps videos are now 63mbps up from 25mbps.

Can you send me a slack invite?

mefistotelis, 2019-04-15:
There's a "Slack" button with invite here:
http://dji.retroroms.info/

If it doesn't know - let me know.

Magic number - we got it via trial and error.

nicolov, 2019-04-16:
Thanks a lot for all your help!

closed 2019-04-16 by nicolov