o-gs/dji-firmware-tools #121: P4P GIMBAL 2

fengzhonggeinichengnuo, 2019-04-17:
Hello, I have been looking for you in the question section some time ago. Do you have time to do the P4P gimbal test recently? I can pay for it.

mefistotelis, 2019-04-17:
No, no time. Sorry.

hanoiflycam, 2019-07-25:
Hi My Idol mefistotelis

hanoiflycam, 2019-07-25:
mefistotelis My Idol Can I help you

mefistotelis, 2019-07-25:
> mefistotelis My Idol Can I help you

Sure, any help on the gimbal is appreciated. I am not working on this yet, though I'm planning to get back to it.

What can you do? Do you have repair skills, or disassembly skills?
Photos of boards within the drones?
Ability to download firmwares from micro-controllers?
Other?

fengzhonggeinichengnuo, 2019-07-25:
Yes, I have the disassembly skills, and I have decrypted and got the firmware of the chip. I have a lot of UAVs to test.

mefistotelis, 2019-07-25:
Here's a simplified Ph3 schematic; ph4 is similar in concept:
https://github.com/o-gs/dji-hardware-schematics/blob/master/phantom_3_pro_overview/phantom_3_pro_overview.pdf

Gimbal is controlled by STM32 uC, but each arm has its own dsPIC uC. We have firmware of the STM32 (without loader; would be nice to have it too, but it's not mandatory for the subject). But we don't have firmware of the dsPIC. You could check if the FW is protected or whether it can be dumped.

fengzhonggeinichengnuo, 2019-07-26:
I got the firmware of the microprocessor, and I could rewrite the firmware into the microprocessor.

mefistotelis, 2019-07-26:
Ok then, can you get the firmware from dsPICs for us?

fengzhonggeinichengnuo, 2019-07-30:
Can we switch platforms, like WeChat?

mefistotelis, 2019-07-30:
If you prefer a chat-style communication, you can find me on Slack. Invite is here:
dji.retroroms.info

fengzhonggeinichengnuo, 2019-07-30:
I can't open this website. Maybe it's due to regional restrictions. What other ways can I keep talking?

mefistotelis, 2019-07-30:
Here's the current link:
https://join.slack.com/t/dji-rev/shared_invite/enQtMjk5OTEyMzcyMjI3LTdlZjY4NzQ5M2M2NmE5ZWM4OTgyNThmZDVmZjdjODE4ODYyNmYwZjYxMDcyYzcxNmZlYzI5ZjI2ZGQ2NGY1ZTc

I'm not using any other chats. For private communication, just email me.

fengzhonggeinichengnuo, 2019-07-30:
I have Y R ESC schematics and FW, but I don't want these resources to appear on public platforms, so I want to go to other platforms to communicate.

fengzhonggeinichengnuo, 2019-07-30:
As you said before, unlike Spark, P4P does not seem to calibrate actively, which can be very difficult.

mefistotelis, 2019-07-30:
Then it would be nest to send those via e-mail, either as attachment or links to any upload service.

email: mefistotelis@gmail.com

joww1, 2019-10-06:
> Eu tenho esquemas YR ESC e FW, mas não quero que esses recursos apareçam em plataformas públicas, então quero ir para outras plataformas para se comunicar.

Can you share it in private? I'm also at the same goal of getting this. I have good experience in testing... Soon, we'll be able to uncover that mystery.

tecdrone, 2021-03-22:
> Peguei o firmware do microprocessador e pude reescrever o firmware no microprocessador.

Olha amigo vc poderia vende as informações para calibra o gimbal P4 p4pro??

fengzhonggeinichengnuo, 2021-03-22:
> > Peguei固件不执行微处理器处理，而reescrever固件不执行微处理器处理。
> 
> Olha amigo vc poderia vende as calipara o gimbal P4 p4pro的信息？WhatsApp + 558199133-0291巴西

Yes, I have completed the PTZ calibration development of P4, P4P. Now this problem can be solved by software

guewald, 2021-03-22:
> > > Peguei固件不执行微处理器处理，而reescrever固件不执行微处理器处理。
> > 
> > 
> > Olha amigo vc poderia vende as calipara o gimbal P4 p4pro的信息？WhatsApp + 558199133-0291巴西
> 
> Yes, I have completed the PTZ calibration development of P4, P4P. Now this problem can be solved by software

I have interest. Can you contact me? +55519999483379 or guewald@gmail.com

joww1, 2021-03-22:
> Podemos mudar de plataforma, como WeChat?

Your WeChat, please?

tecdrone, 2021-03-23:
> > > Peguei 固件 不 执行 微处理器 处理, 而 reescrever 固件 不 执行 微处理器 处理。
> > 
> > 
> > Olha amigo vc poderia vende como calipara o gimbal P4 p4pro 的 信息？ WhatsApp + 558199133-0291 巴西
> 
> Sim, concluí o desenvolvimento da calibração PTZ de P4, P4P. Agora, esse problema pode ser resolvido por software

hello friend you can talk to me by email.  suporte@tecdrone.com.br
or whatsapp +558199133-0291

josjrbr, 2021-11-22:
> 

Hello my friend
I'm interested

josjrbr@gmail.com

Clearvisionsenthil, 2021-12-16:
Hi friend
I am interested. 
Senthilyogan@yahoo.com
WhatsApp-+919842193180

> > > Peguei固件不执行微处理器处理，而reescrever固件不执行微处理器处理。
> > 
> > 
> > Olha amigo vc poderia vende as calipara o gimbal P4 p4pro的信息？WhatsApp + 558199133-0291巴西
> 
> Yes, I have completed the PTZ calibration development of P4, P4P. Now this problem can be solved by software

Hi friend
I am interested. 
Senthilyogan@yahoo.com
WhatsApp-+919842193180

Clearvisionsenthil, 2021-12-21:
[image]
[image]

#
Hi,
 how to read& write binary files in p4p IMU board &yaw motor drive board chips? I have usb to ttl convertor. Which software to use read&write binary files. 
Any other tools for programming? 
Please help. 
I will ready to pay. 
Thanks. 
WhatsApp- +919842193180.

fengzhonggeinichengnuo, 2021-12-21:
他们都是处于加密状态 无法读取 通过特殊手段对芯片开片才可以得到 

发自我的iPhone

mefistotelis, 2021-12-21:
> how to read& write binary files in p4p IMU board &yaw motor drive board chips? I have usb to ttl convertor. Which software to use read&write binary files.

IMU board - that's STM32 processor, so you should use STM32 programmer. You won't be able to read the chip though, because it's protected. Thew firmware probably consists of bootrom part and application part, and only application part is included in FW updates; so reflashing this chip would require dumping bootrom using hacked application part, and then appending proper application part to that bootrom.

motor driver board - the CPU is rebranded as "DJI", but it's just normal uC. We'd have to know the real model of that uC to tap to it.

fengzhonggeinichengnuo, 2021-12-21:
I already have all these data, and I have made them tools and calibrated them.

The attached picture is the calibration software I developed.

发自我的iPhone

Clearvisionsenthil, 2021-12-22:
Hi, thanks for your support.

Clearvisionsenthil, 2021-12-22:
Phantom 4pro series gimbal calibration software any available?

fengzhonggeinichengnuo, 2021-12-22:
Yes, the price is 15,000 RMB.

发自我的iPhone

------------------ Original ------------------
From: Clearvisionsenthil ***@***.***&gt;
Date: Wed,Dec 22,2021 11:13 AM
To: o-gs/dji-firmware-tools ***@***.***&gt;
Cc: fengzhonggeinichengnuo ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [o-gs/dji-firmware-tools] P4P GIMBAL 2 (#121)

Clearvisionsenthil, 2021-12-22:
Only software? Including any tools? Which models compatible? Only for gimbal calibration? Please send full details.

fengzhonggeinichengnuo, 2021-12-24:
请添加我的微信&nbsp;KBDZ-DJI

发自我的iPhone

------------------ Original ------------------
From: Clearvisionsenthil ***@***.***&gt;
Date: Wed,Dec 22,2021 3:56 PM
To: o-gs/dji-firmware-tools ***@***.***&gt;
Cc: fengzhonggeinichengnuo ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [o-gs/dji-firmware-tools] P4P GIMBAL 2 (#121)

Clearvisionsenthil, 2021-12-24:
No wechat. WhatsApp- +91 9842193180

请添加我的微信 KBDZ-DJI

Clearvisionsenthil, 2022-01-07:
Add me to wechat

请添加我的微信 KBDZ-DJI

bugalaws, 2023-04-20:
> Yes, the price is 15,000 RMB. 发自我的iPhone
> […](#)
> ------------------ Original ------------------ From: Clearvisionsenthil ***@***.***&gt; Date: Wed,Dec 22,2021 11:13 AM To: o-gs/dji-firmware-tools [...]

Hello, fengzhonggeinichengnuo how can i contact you? I have the interest into buy the solution to calibrate the phantom 4 pro gimbals

lbacciotti, 2023-05-24:
> Como você disse antes, ao contrário do Spark, o P4P não parece calibrar ativamente, o que pode ser muito difícil.

Hello we could talk by wechat about the phantom 4 my name is Lucas Bacciotti on wechat

nullibyte, 2023-09-13:
@mefistotelis I am also interested in helping with gimbal recalibration, had to swap the yaw motor out of a p4p as the drone was crashed, also the esc board for yaw was missing, but it seems a salvaged board from a p4s does work, but as expected the motor calibration is all wrong. 

One thing I'm uncertain on is does the IMU board store the magnetization data, or is it stored on the esc boards? I have 1 working p4s and the p4p with faulty yaw calibration that I can contribute with as needed

From what I understand if the IMU stores this data, then you could swap the magnet from one motor to another to retain calibration. But if it is the esc board then I might be screwed as I have no idea where the original one is, and only have access to a spare one from a salvage drone. I'd rather help find a software fix than resort to magnet swapping

nullibyte, 2023-09-13:
I also have a zenmuse x3 to help improve the service script. Thus far I got as far as loading the elf binary into iaito for decompiling, but have had no luck sourcing ambarella documentation to convert the assembly into something human readable

lbacciotti, 2023-09-16:
Thanks for answering me, would you have the file to do the recording of
stm32.?