o-gs/dji-firmware-tools #122: Help! Can I use P3 Pro Gimbal in P3 4k?

Spider84, 2019-05-11:
Hello!

My P3 4k gimbal is deaded after crash. I can't find true P3 4k gimbal. But I have many P3 Pro gimbals. Can I use them?
I have readed that I can resolder resistor to switch USB direction. Where I can read what exactly? 

Thx.

Spider84, 2019-05-12:
Gimbal is different.  Connector is different :(
[image]
[image]

mefistotelis, 2019-05-12:
Yup. Phantom 3 Pro and Advanced are very similar and the boards can be switched between them. Other models, like Phantom 3 4K and Phantom 3 Standard are different, don't know if compatible in their own group.

The photos show Phantom 3 Advanced board.

Spider84, 2019-05-12:
> The photos show Phantom 3 Advanced board.
No. At photo P3 4k board

mefistotelis, 2019-05-12:
OK, that is interesting. I didn't knew 4k board looks like Advanced..

So you're saying those are still not compatible? where?

Spider84, 2019-05-12:
First: This board has different 12V connector. If I connect P3 Pro gimbal to P3 4k drone - drone don't turn on. Short circuit issue :( 
Second: At pictures below you can see that PCB is different then Pro/Adv.
Around 2 capacitors different polygon.
Different polygon around connector and different pinout as result.
Additional Transistors (??? SOT-23) near SD Card slot.
Third: Another Camera PCB, but I lost my cam (did broke gimbal and lost Cam on fly...)
[image]
[image]
[image]
[image]

Spider84, 2019-05-12:
Camera from P3 Pro don't work with P3 4K Gimbal - as result in DJI GO Gimbal not detected, and no picture.
BUT! Camera from P3 4K Work on P3 Pro gimbal (in my search of camera I found this example of replacement)

P.S. I'm searching for P3 4K camera :((( Please!

Spider84, 2019-06-01:
I finally resolve problem.
In P3 4k used camera from P3 Adv.
Now All work fine.

closed 2019-06-02 by Spider84