o-gs/dji-firmware-tools #123: Dji osmo action fw support.

nutsey, 2019-05-22:
Starts with xV4. Here is the link:
https://mydjiflight.dji.com/file/links/OSMO_ACTION_V12010_20190516_ZIP

hypoxic, 2019-05-22:
Seems to have some sort of compression on it or scrambling. I'm going with
compression as SJcam's H2 based processor binary is 60+MB. I just can't
locate an appropriate header or algorithm. Thank you for the direct link.

nutsey, 2019-05-22:
At first glance it looks like LZ algo.

pawelsky, 2019-05-22:
LZ4 to be precise :)

hypoxic, 2019-05-22:
Thank you. Don’t know how I missed that. The rtos is compressed again in
section 0

drozdov1984, 2019-10-22:
Supported DJI commands for osmo action
https://github.com/drozdovoleg/DJI-Osmo-Action-command-research

nutsey, 2019-10-22:
> Supported DJI commands for osmo action
> https://github.com/drozdovoleg/DJI-Osmo-Action-command-research
Anything interesting available with 't dji iq'?

drozdov1984, 2019-10-22:
No, nothing happens.

drozdov1984, 2019-10-22:
> > Supported DJI commands for osmo action
> > https://github.com/drozdovoleg/DJI-Osmo-Action-command-research
> 
> Anything interesting available with 't dji iq'?

No, nothing happens