o-gs/dji-firmware-tools #124: P3 MAX Altitude hack. How to use?

Spider84, 2019-06-02:
Hello!  
I patched my firmware by  dji_flyc_param_ed.py  tool and uploaded to device via SDcard and PMCAPPFw3.bin file.  
All looks like I do right. BUT! I can't flight higher 500 meters. How to set higher altitude?  
DJI GO not able to set more then 500m.

mefistotelis, 2019-06-02:
Either use modified Dji Go, or set the parameter on PC - for example using comm_og_service_tool.py.

Spider84, 2019-06-02:
Thx.  
Can you say where to get modified Dji GO for P3?

ferraript, 2019-06-04:
if you can't find (or make yourself) patched DJI GO, change it in firmware's flyc_param_infos
but not only change defaultValue, but change minValue too (for example to 1500), so DJI GO won't be able to overwrite the max height to 500

Spider84, 2019-06-05:
All paches for over 4 version. But I need before 4. Apktool can't disassembly current DJI GO version :(

mefistotelis, 2019-06-05:
You need a 'nosecneo' version.

Go to [dji.retroroms.info](http://dji.retroroms.info/), there's a Slack link in top menu.
When on Slack, join 'spoon-feeding' channel. Ask for nosecneo version there.
The nosecneo version can be easily used with [deejayeye-modder](https://github.com/Bin4ry/deejayeye-modder).