o-gs/dji-firmware-tools #125: P3X Flashing the STM32F427IIH6 on the MC board

DannyBoyCJ, 2019-06-30:
I managed to damage the m0305 and m0306 modules on my Phantom 3 Pro by accidentally uploading encrypted modules. Is there a guide somewhere about reflashing the STM32F427IIH6 chip, both the bootloader and the program? I tried the SD card way with the unencrypted modules (PMCAPPFw3 and PMCLDRFw3), but that doesn't work. They just get renamed to "PMCAPPFw3_damaged.bin" and "PMCLDRFw3_damaged.bin", and the update fails. The log shows:

```
 [00017942]========== remo-con disconnect. boot(15) ============
 [00018026]Firmware [C:\PMCLDRFw3.bin] detected, card sn [0x28537f3c].
 [00018105][03 05] Firmware upgrade start...
 [00060698][03 05] Firmware upgrade finish failed (step = 2, err = 0xaa).
 [00090782]Done.

 [00095289]========== remo-con disconnect. boot(15) ============
 [00095368]Firmware [C:\PMCAPPFw3.bin] detected, card sn [0x28537f3c].
 [00095457][03 06] Firmware upgrade start...
 [00138031][03 06] Firmware upgrade finish failed (step = 2, err = 0xaa).
 [00168116]Done.
```
So what I think I need is the image files that need to be flashed, an ST-Link programmer, and some kind of guide/instructions. I ordered the ST-Link, so if anyone can provide the images and/or instructions, I would be very grateful :). Also, if there's another way of flashing said modules (given the damaged state), please let me know.

mefistotelis, 2019-07-01:
Images - you can extract them from firmware package; just make sure they're decrypted during extraction.
You also need an address at which each image resides in the STM32 internal flash.
Instructions on how to use ST-link - there are lots of videos on that; plus there's original documentation in programming guide of the chip.

Other way of flashing - there might be one. But since we don't know whether it exists or how to use it, it is best to focus on the ST-link way, which some people claimed to use successfully in the past. (The m0305 might be either the only bootloader in use, or a second stage bootloader. If it second stage, then the first stage code (which we don't have) might have firmware loading functionality).

DannyBoyCJ, 2019-07-01:
Thank you for the advice!

> The m0305 might be either the only bootloader in use, or a second stage bootloader. If it second stage, then the first stage code (which we don't have) might have firmware loading functionality)

I'm wondering, if the m0305 bootloader would be second stage, wouldn't the primary bootloader (not touched by the update) act as a failsafe, allowing the reflashing of the modules even after a botched job like I did? I mean via the normal SD card way, with unencrypted modules on it, or even whole firmware. If that makes no sense, just ignore it, I'm not savvy about these things.

> it, it is best to focus on the ST-link way, which some people claimed to use successfully in the past

If you know of a forum/discussion where such people might talk about their deeds, that would help me greatly. Meanwhile, as I'm waiting for my St-link V2, I will have a go at extracting the image files from the modules. Don't really know what I'm doing, but I'll try. About the service pads, I'm thinking:

SWDIO to SWDIO
SWDCLK to SWDCLK
GND TO GND
VCC3V3 TO 3.3V
BOOT0 TO VCC3V3

I don't see a RST pad, but maybe that's not needed. Not sure which software for the ST Link to use yet, but I'll watch more tutorials and use what others are using.  I don't plan on desoldering the MC from the ESC board. I hope I can provide power to the board through the St-Link device, but if that's not enough, I'll work on the powered drone. 
The last piece of the puzzle are the addresses at which the bootloader and the program are to be uploaded. I wonder if these are generic addresses, common to the stm32 family (or the specific STM32f427IIH6 part), or rather they are custom addresses, specific to dji products. If so, would someone with a working unit be able to extract them from a boot log? If you know of a resource for this info, please point me in the right direction.

mefistotelis, 2019-07-01:
> wouldn't the primary bootloader (not touched by the update) act as a failsafe, allowing the reflashing of the modules even after a botched job like I did?

It would allow reflashing, but not necessarily by that method.

> If you know of a forum/discussion where such people might talk about their deeds, that would help me greatly.

Someone claimed that here, within "issues". Can't remember any details though.

> BOOT0 TO VCC3V3

Don't know about this one; I have no idea what this pin does.
Might be explained in AN2606 pdf from ST website.

> The last piece of the puzzle are the addresses at which the bootloader and the program are to be uploaded. 

All I know in regard to this is that the application is visible at RAM address 0x8020000 after the device booted. How this maps to internal flash, I don't know.

DannyBoyCJ, 2019-07-01:
Ok, thanks, I'll see how it works out...

hanoiflycam, 2019-07-25:
> Hướng dẫn cách sử dụng ST-link - có rất nhiều video về điều đó

Hi Man. Can I heip You..

DannyBoyCJ, 2019-07-25:
> Hi Man. Can I heip You..

Yes, I'm looking for a way to "revive" the m0305 and m0306 modules on the STM32F427 controller of MC. As I said, I accidentaly uploaded the encrypted modules, so now I can't update it via SD card. I'm looking for another way, probably by ST-link. I don't have enough experience to do it, so yeah I could use some help. What I need is a step by step procedure for updating via ST link, including the procedure for correctly extracting the necessary files from the modules, the addresses where those files should be uploaded, and the necessary service pad connections on the board. I have a rough idea about the process, but I don't want to risk making a bigger mess of things by going forward like this. So yeah, I would appreciate any help.

MrBurnsAT, 2019-07-25:
I think u have a version mismatch. That's why u couldn't update via SD card.

Check all modules for version numbers. Than bring back the mismatched module to match to the rest of the firmware version. (only one module update via extracted module). After that u should be able to update via SD card and official firmware file

DannyBoyCJ, 2019-07-25:
I'm quite sure I have a version mismatch, but sadly I can't tell the versions on the drone because they don't show up in the app. It has very limited connectivity. I can only update via separate modules, because when I put the official firmware on the SD it just doesn't start the update process. The only module that I haven't been able to update this way (before the current issue) was m0100. This issue was discussed [here](https://github.com/o-gs/dji-firmware-tools/issues/37#issuecomment-372075989). This is also probably why the drone won't accept an official firmware package.

MrBurnsAT, 2019-07-25:
Try an update via SD card. U should get an file where all versions are compared and listed

Should be after the update in an hidden folder (MISC) within should be a folder named LOG within u should get the file.

DannyBoyCJ, 2019-07-29:
Sorry for the late response, I was away from the PC for a few days. I did make a few updates via SD since the issues started, but only from separate modules extracted from official firmware. The thing is, when I do that, the drone doesn't do the version check, it just detects the module on the SD and then starts the update. When I try a full update (with an official firmware bin on the SD), the process just doesn't initialize and nothing gets written in the logs (both the one in MISC and the one in the SD root). It acts like it doesn't know the firmware file is on the SD. I'm attaching a recent log from the MISC folder so you can see what I'm talking about.

[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/3440939/P3X_FW_LOG_AB.txt)

MrBurnsAT, 2019-07-29:
Tried with the Debug File?

DannyBoyCJ, 2019-07-29:
Yes. It makes no difference, no matter what version of firmware I copy on the SD.

MrBurnsAT, 2019-07-29:
I think with the Debug File on the SD there should be something in the hidden log file

DannyBoyCJ, 2019-07-29:
Hmm... Let me try and I'll get back with the results.

mefistotelis, 2019-07-29:
You can't tell the camera to flash FC if FC is not responding. How do you imagine it would work? After m0305 have been damaged, there is no software on FC which would be able to receive DUML messages from the camera. It won't magically come back to life no matter how many files you create on SD-card.

I would get back to the proper fixing path which was discussed here before.

DannyBoyCJ, 2019-07-29:
That makes sense. But maybe I can get a list of software versions for the other modules, to see how far off they are from each other, since the remote doesn't report anything. I guess the software on the drone is a mambo-jumbo now, since some of the modules accepted to update and some didn't. Probably won't work, but it won't cost me anything to try.

DannyBoyCJ, 2019-07-29:
No cigar. It writes no logs (main or hidden) when it has official fw on the card, with or without debug. I tried with modules 08 and 15 on the SD and the update failed, so I guess the NAND flash is bad again and needs to be revived. At least we have a procedure for that. I attached the hidden log, because the main one is unchanged since it stopped accepting official fw, but it contains no firmware version info.
[P3X_FW_LOG_AB.txt](https://github.com/o-gs/dji-firmware-tools/files/3441686/P3X_FW_LOG_AB.txt)

gdanas87, 2020-02-04:
> No cigar. It writes no logs (main or hidden) when it has official fw on the card, with or without debug. I tried with modules 08 and 15 on the SD and the update failed, so I guess the NAND flash is [...]

Hello Danny did you ever manage to fix this using the method you described with mefistotelis?I am having exactly the same problem as you