o-gs/dji-firmware-tools #126: Phantom 3 Pro Camera No longer working.

ramteen85, 2019-07-27:
hi there. i have lost complete vision of the phantom 3 camera.
every time i flick it on, red light. I attempted changing the parameters at 1.11.

Everything installed nicely. all steps were followed nicely.

I followed these instructions.

https://www.youtube.com/watch?v=e00mOOcSi18

But after turning the phantom back on the controller no longer communicates with the camera. Green light is on the RC, but it either says disconnected or weak transmission.
A red light flashes on the gimbal after it is turned on. I can not get it solid green anymore.

When trying to install the original firmware 1.11 that installs as well. But now no matter which I install, ever since attempting this, my vision is completely lost as well as most communication between the rc and phantom 3 pro.

I had  tried this because someone else had said it had worked. It actually was working and then this happened.

If anyone could help me fix this much appreciated as i know python, but have no idea how to reverse engineer a phantom :( im also using the dji go app.

ramteen85, 2019-07-28:
an update: reinstalled the software without configuring no fly zones.
camera was back up and then while doing the imu calibration, the camera went dead again.

please help how do i fix this? is there a way of going back to the original phantom firmware?

mefistotelis, 2019-07-28:
If you want specific help, you need to be more specific in what you did. We're not going to pull your tongue on everything.

If you're updating FC firmware, camera and gimbal are not updated / changed in any way.

ramteen85, 2019-07-28:
pull my tongue. haha thats funny.
thank you for your response.

I started off by installing dji phantom 3 professional firmware 1.11.20 on my aircraft, with the remote mod at 1.9.2. 

Following this guide:

https://www.youtube.com/watch?v=e00mOOcSi18

I managed to perform the following steps with my phantom 3 pro.
I also paid note to the description as well as comments which said that this works with the latest firmware. so i tried it.

My dji go app is at the latest version on android.

When I extracted the firmware tools the first time, after modding, the video signal between the drone and the controller went dead.
The second time i repeated the process and did everything again without touching the no-fly zone.
Worked for a little while until video went dead again.
By that i mean black screen you cannot see any picture on your phone. but remote light is green and says controller connected. although dji go can sometimes say disconnected.

I noticed the remote has to be charged past 50% to keep the video signal good.

Another bug i noticed with the android version of dji go - the latest.
I noticed it now works if everything is fully charged and the controller is on before the phantom.

it seems to fly smoothly but how can i make sure what is stable and what to do different?
Another strange bug i have noticed is when i switch from "usb charging" to "usb file transfer" on a phone which i believe is what im supposed to do, all video communication breaks and i have to restart the drone as well as the remote and app.

how can i ensure i keep a stable video connection ?
If i ever wanted to revert back to the official 1.11 firmware could i?

would it still be okay to mod the remote with a range extender as well?

Im mainly concerned about the usb file transfer bug i mentioned above, and making sure i went the steps right.

Files used when following the video i linked were the firmware 1.11 file, the dji_xv4_fvcon.py file, and the dji_flyc_param_ed.py file. 

Thanks for your help. all this would be good to know.

mefistotelis, 2019-07-28:
> When I extracted the firmware tools the first time, after modding, the video signal between the drone and the controller went dead.

Ok, I watched the video, it says to update m0306 only. So basically, the issue you're having is unrelated to the modding.

You might have overheated your gimbal. That usually causes loss of FPV.

> I noticed the remote has to be charged past 50% to keep the video signal good.

That would be unusual. I don't see how it could be related. Possibly misinterpretation.

> it seems to fly smoothly but how can i make sure what is stable and what to do different?

You may learn more on how the drone works on wiki pages of this project.
We don't have too much diagnosis tools; our service tool does not have issue diagnosis function yet. For no image transmission, there's a diagnosis chart on phantompilots forum, somewhere. The guy who made it collaborated with me when creating it, so I can't tell much more than you see on that chart.

> Another strange bug i have noticed is when i switch from "usb charging" to "usb file transfer" on a phone which i believe is what im supposed to do,

Never tried that, don't know.

> how can i ensure i keep a stable video connection ?

Find the issue and fix it. It has nothing to do with modding. Well, besides that you overheated your drone while modding. But any firmware update could've overheated it.

> If i ever wanted to revert back to the official 1.11 firmware could i?

The video I just watches shows how to downgrade firmware. The same technique can be used to re-flash existing FW version. Though I would just reflash the one module which was modded.

> would it still be okay to mod the remote with a range extender as well?

Up to you. But that has nothing to do with your current issue. Won't fix it.

>  all this would be good to know.

You can get in-depth knowledge on the wikis. The main purpose of having them is to make it easier for new people to get understanding on how DJI products work, for both repair and modding.

ramteen85, 2019-07-28:
cool makes sense. are there further steps i can take to prevent further
overheating the gimbal? i dont have to replace it do i?
i still manage to get image transmission. however just on usb charging

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Scotty-coder, 2019-08-22:
The Phantom 3 Professional is known to burn out the Gimbal AV Board despite having a fan if the drone is left on for long periods of time without a desk fan aim at the Gimbal.  We have had many come in for repairs after experiencing this issue.  General examples are leaving the drone on for 20+ minutes doing firmware updates or just having the drone powered on for hours while you play with app functions.  DJI never released a statement on this issue.  Hope this helps. 

Regards Scott

Scotty-coder, 2019-08-22:
Usually if the AV board burns out you lose image so perhaps that is not you issue.  But still take my advice with leaving the drone on without a desk fan. :)