o-gs/dji-firmware-tools #127: P3A P3S_FW_V01.11.0020.bin how to open FCC?

gaifeng8864, 2019-10-07:
When I exported JSON using "lightbridge_stm32_hardcoder.py", I found that "P3A P3S_FW_V01.11.0020_m0900.elf" had a lot of options that were not found, and the last generated "P3S_FW_V01.11.0020_m0900.bin" was burned into the drone. After that, my drone image transmission distance is smaller. When I use these tools on this version of P3S_FW_V01.11.0020, do I need to change those parameters? thanks

mefistotelis, 2019-10-07:
Here's a full log of testing a firmware with m0900 identical to yours.
Use it as a refernce to check whether something is wrong with your run.

The test include extraction, changing values in JSON and modifying the binary to JSON changes:

```
### TEST of hardcoders with P3X_FW_V01.11.0030 ###
fw/P3X_FW_V01.11.0030.bin: Opening for extraction
fw/P3X_FW_V01.11.0030.bin: Package format version 2016 detected
fw/P3X_FW_V01.11.0030.bin: Header:
{   'entry_count': 17,
    'hdrend_offs': 950,
    'magic': 305419896,
    'magic_ver': 1,
    'manufacturer': b'DJI',
    'model': b'P3X',
    'padding': '00000000000000000000',
    'timestamp': 1504073841,
    'ver_latest': 17498142,
    'ver_latest_enc': 160103691,
    'ver_rollbk': 17498132,
    'ver_rollbk_enc': 160103681}
fw/P3X_FW_V01.11.0030.bin: Module index 0
{   'decrypted_len': 905728,
    'decrypted_md5': 'a6f0e3375c3a8417d8090f1c27d1826d',
    'dt_offs': 950,
    'encrypt_type': 1,
    'reserved2': 1,
    'spcoding': 16,
    'splvalue': 0,
    'stored_len': 905728,
    'stored_md5': 'c0c9ee22ae23700e7f3d41d00a4a5ca8',
    'target': 'm0306',
    'target_name': 'main '
                   'controller '
                   "'A3' "
                   'app',
    'version': '02.04.5170'}
fw/P3X_FW_V01.11.0030.bin: Module index 1
{   'decrypted_len': 43776,
    'decrypted_md5': 'cc3baf52297f2eb67adf98f0270956f4',
    'dt_offs': 906678,
    'encrypt_type': 1,
    'reserved2': 1,
    'spcoding': 16,
    'splvalue': 0,
    'stored_len': 43776,
    'stored_md5': 'bbd8964489204dfdb8858d5824a116b0',
    'target': 'm0305',
    'target_name': 'main '
                   'controller '
                   "'A3' "
                   'ldr',
    'version': '34.02.0009'}
fw/P3X_FW_V01.11.0030.bin: Module index 2
{   'decrypted_len': 97032,
    'decrypted_md5': '391e3bd702bf6ef8cb1f85c2f5d92be3',
    'dt_offs': 950454,
    'encrypt_type': 0,
    'reserved2': 1,
    'spcoding': 0,
    'splvalue': 0,
    'stored_len': 97032,
    'stored_md5': '391e3bd702bf6ef8cb1f85c2f5d92be3',
    'target': 'm0400',
    'target_name': 'gimbal '
                   'mdl '
                   '0',
    'version': '01.48.0000'}
fw/P3X_FW_V01.11.0030.bin: Module index 3
[... 1792 lines cut ...]
At 0x800f24c, replacing 0221 -> 0021 to set tcx_config_update1.board_ar7_attenuation_tx1_cnup
Making code re: movs	r1, #(?P<board_ar7_attenuation_tx2_fcc>[0-9a-fx]+)
Compiling code: movs	r1, #0x0
At 0x800fedc, replacing 0221 -> 0021 to set board_ar7_attenuation_tx2_fcc
Making code re: movs	r1, #(?P<board_ar7_attenuation_tx2_init>[0-9a-fx]+)
Compiling code: movs	r1, #0x0
At 0x800da54, replacing 0221 -> 0021 to set init_fpga_config.board_ar7_attenuation_tx2_init
Making code re: movs	r1, #(?P<board_ar7_attenuation_tx2_cnup>[0-9a-fx]+)
Compiling code: movs	r1, #0x0
At 0x800f256, replacing 0221 -> 0021 to set tcx_config_update1.board_ar7_attenuation_tx2_cnup
Making code re: movs	r0, #(?P<packet_received_attenuation_value>[0-9a-fx]+)
Compiling code: movs	r0, #0x0
At 0x80121f6, replacing a078 -> 0020 to set packet_received_attenuation_override
P3X_FW_V01.11.0030_m0900.elf: Updated 22 out of 23 hardcoded values
+ arm-none-eabi-objcopy -O binary P3X_FW_V01.11.0030_m0900.elf P3X_FW_V01.11.0030_m0900.bin
+ set +x
### OK: Amount of changes in bin file, 46, is reasonable. ###
### SUCCESS: Binary file changes are within acceptable limits. ###
```

To run the test yourself, use:

```
./tests/test_hardcoders_p3x.sh
```

gaifeng8864, 2019-10-07:
> 这是使用与您的m0900相同的固件测试固件的完整日志。
> 使用它作为参考，检查您的跑步是否有问题。
> 
> 测试包括提取，更改JSON中的值以及将二进制文件修改为JSON更改：
> 
> ```
> ### TEST of hardcoders with P3X_FW_V01.11.0030 ###
> fw/P3X_FW_V01.11.0030.bin: Opening for extraction
> fw/P3X_FW_V01.11.0030.bin: Package format version 2016 detected
> fw/P3X_FW_V01.11.0030.bin: Header:
> {   'entry_count': 17,
>     'hdrend_offs': 950,
>     'magic': 305419896,
>     'magic_ver': 1,
>     'manufacturer': b'DJI',
>     'model': b'P3X',
>     'padding': '00000000000000000000',
>     'timestamp': 1504073841,
>     'ver_latest': 17498142,
>     'ver_latest_enc': 160103691,
>     'ver_rollbk': 17498132,
>     'ver_rollbk_enc': 160103681}
> fw/P3X_FW_V01.11.0030.bin: Module index 0
> {   'decrypted_len': 905728,
>     'decrypted_md5': 'a6f0e3375c3a8417d8090f1c27d1826d',
>     'dt_offs': 950,
>     'encrypt_type': 1,
>     'reserved2': 1,
>     'spcoding': 16,
>     'splvalue': 0,
>     'stored_len': 905728,
>     'stored_md5': 'c0c9ee22ae23700e7f3d41d00a4a5ca8',
>     'target': 'm0306',
>     'target_name': 'main '
>                    'controller '
>                    "'A3' "
>                    'app',
>     'version': '02.04.5170'}
> fw/P3X_FW_V01.11.0030.bin: Module index 1
> {   'decrypted_len': 43776,
>     'decrypted_md5': 'cc3baf52297f2eb67adf98f0270956f4',
>     'dt_offs': 906678,
>     'encrypt_type': 1,
>     'reserved2': 1,
>     'spcoding': 16,
>     'splvalue': 0,
>     'stored_len': 43776,
>     'stored_md5': 'bbd8964489204dfdb8858d5824a116b0',
>     'target': 'm0305',
>     'target_name': 'main '
>                    'controller '
>                    "'A3' "
>                    'ldr',
>     'version': '34.02.0009'}
> fw/P3X_FW_V01.11.0030.bin: Module index 2
> {   'decrypted_len': 97032,
>     'decrypted_md5': '391e3bd702bf6ef8cb1f85c2f5d92be3',
>     'dt_offs': 950454,
>     'encrypt_type': 0,
>     'reserved2': 1,
>     'spcoding': 0,
>     'splvalue': 0,
>     'stored_len': 97032,
>     'stored_md5': '391e3bd702bf6ef8cb1f85c2f5d92be3',
>     'target': 'm0400',
>     'target_name': 'gimbal '
>                    'mdl '
>                    '0',
>     'version': '01.48.0000'}
> fw/P3X_FW_V01.11.0030.bin: Module index 3
[... 1792 lines cut ...]
> At 0x800f24c, replacing 0221 -> 0021 to set tcx_config_update1.board_ar7_attenuation_tx1_cnup
> Making code re: movs	r1, #(?P<board_ar7_attenuation_tx2_fcc>[0-9a-fx]+)
> Compiling code: movs	r1, #0x0
> At 0x800fedc, replacing 0221 -> 0021 to set board_ar7_attenuation_tx2_fcc
> Making code re: movs	r1, #(?P<board_ar7_attenuation_tx2_init>[0-9a-fx]+)
> Compiling code: movs	r1, #0x0
> At 0x800da54, replacing 0221 -> 0021 to set init_fpga_config.board_ar7_attenuation_tx2_init
> Making code re: movs	r1, #(?P<board_ar7_attenuation_tx2_cnup>[0-9a-fx]+)
> Compiling code: movs	r1, #0x0
> At 0x800f256, replacing 0221 -> 0021 to set tcx_config_update1.board_ar7_attenuation_tx2_cnup
> Making code re: movs	r0, #(?P<packet_received_attenuation_value>[0-9a-fx]+)
> Compiling code: movs	r0, #0x0
> At 0x80121f6, replacing a078 -> 0020 to set packet_received_attenuation_override
> P3X_FW_V01.11.0030_m0900.elf: Updated 22 out of 23 hardcoded values
> + arm-none-eabi-objcopy -O binary P3X_FW_V01.11.0030_m0900.elf P3X_FW_V01.11.0030_m0900.bin
> + set +x
> ### OK: Amount of changes in bin file, 46, is reasonable. ###
> ### SUCCESS: Binary file changes are within acceptable limits. ###
> ```
> 
> 要自己运行测试，请使用：
> 
> ```
> ./tests/test_hardcoders_p3x.sh
> ```

closed 2019-10-07 by gaifeng8864

gaifeng8864, 2019-10-07:
> P3X_FW_V01.11.0030

Thank you for your help. My drone model is P3A. The latest firmware version displayed on the official website is "P3S_FW_V01.11.0020.bin". I saw that you are using P3X_FW_V01.11.0030. What is the difference between P3S_FW_V01.11.0020 and P3X_FW_V01.11.0030? In addition to the machine model, the numbers on the back are different. How was V01.11.0030 obtained?

mefistotelis, 2019-10-07:
P3X and P3S differ only in camera and FPV firmware (m010? and m0800).

Version V01.11.0030 is a fix to .0020 which reverts a "fix" in ESC firmware. The only difference is the ESC firmware. The "fix" DJI made enabled strict validation of ESC state, which made some completely functional drones to show "ESC Status Error".

V01.11.0030 was shared by a DJI employee on phantompilots forum.
Like all firmwares, you can get it with DankDroneDownloader.

gaifeng8864, 2019-10-08:
> P3X和P3S的区别仅在于相机和FPV固件（m010？和m0800）。
> 
> V01.11.0030版本是对.0020的修复，可修复ESC固件中的“修复”。唯一的区别是ESC固件。DJI的“修复”功能可以对ESC状态进行严格验证，从而使一些功能完备的无人机显示“ ESC Status Error”。
> 
> V01.11.0030由DJI员工在phantompilots论坛上共享。
> 像所有固件一样，您可以通过DankDroneDownloader获得它。

great job.
I re-created the P1765Fw3.bin file, but the transmission distance is still not much improved compared to normal, only about 1.7KM. Can I change the JOSN file to the following to open the FCC?

[
	{
		"index" : 0,
		"description" : "What to do when received a packet with transceiver power set request; 0 - use the received attenuation value, 1 - override the value with constant one",
		"minValue" : 0,
		"maxValue" : 1,
		"defaultValue" : 0,
		"setValue" : 1,
		"name" : "og_hardcoded.lightbridge_stm32.packet_received_attenuation_override"
	},
	{
		"index" : 1,
		"description" : "Constant attenuation value used when packet_received_attenuation_override is enabled; unit depends on OFDM board type",
		"minValue" : 0,
		"maxValue" : 255,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.packet_received_attenuation_value"
	},
	{
		"index" : 2,
		"description" : "Transceiver attenuation value for board type 4 with Analog Devices chip, change by 1 means 0.25 dBm",
		"minValue" : 0,
		"maxValue" : 255,
		"defaultValue" : 50,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.board_ad4_attenuation_tx1_ce"
	},
	{
		"index" : 3,
		"description" : "Transceiver attenuation value for board type 4 with Analog Devices chip, change by 1 means 0.25 dBm",
		"minValue" : 0,
		"maxValue" : 255,
		"defaultValue" : 44,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.board_ad4_attenuation_tx2_ce"
	},
	{
		"index" : 4,
		"description" : "Transceiver attenuation value for board type 5 with Analog Devices chip, change by 1 means 0.25 dBm",
		"minValue" : 0,
		"maxValue" : 255,
		"defaultValue" : 44,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.board_ad5_attenuation_tx1_ce"
	},
	{
		"index" : 5,
		"description" : "Transceiver attenuation value for board type 5 with Analog Devices chip, change by 1 means 0.25 dBm",
		"minValue" : 0,
		"maxValue" : 255,
		"defaultValue" : 45,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.board_ad5_attenuation_tx2_ce"
	},
	{
		"index" : 6,
		"description" : "Transceiver attenuation value for board type 6 with Artosyn chip, change by 1 means 1 dBm",
		"minValue" : 0,
		"maxValue" : 63,
		"defaultValue" : 20,
[... 123 lines cut ...]
		"description" : "Transceiver attenuation value for board type 7 with Artosyn chip, change by 1 means 1 dBm",
		"minValue" : 0,
		"maxValue" : 63,
		"defaultValue" : 2,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.board_ar7_attenuation_tx1_fcc"
	},
	{
		"index" : 21,
		"description" : "Transceiver attenuation value for board type 7 with Artosyn chip, change by 1 means 1 dBm",
		"minValue" : 0,
		"maxValue" : 63,
		"defaultValue" : 2,
		"setValue" : 0,
		"name" : "og_hardcoded.lightbridge_stm32.board_ar7_attenuation_tx2_fcc"
	},
	{
		"index" : 22,
		"description" : "Firmware version number",
		"minValue" : "00.00.00.00",
		"maxValue" : "99.99.99.99",
		"setValue" : "04.01.00.00",
		"name" : "og_hardcoded.lightbridge_stm32.mcu_firmware_version"
	}
]

mefistotelis, 2019-10-08:
Yes, that looks fine.

gaifeng8864, 2019-10-08:
> Yes, that looks fine.

thank you very much! Tomorrow I will replace a place to test again. I hope that the cause of the interference alone is less than 5KM instead of the P1765Fw3.bin file.