o-gs/dji-firmware-tools #129: P4A curious gimbal issue

guewald, 2019-10-23:
Hello, 
i have an P4A ho have an curious gimbal issue, never see this before.

When power the drone, the gimbal starts normally, stabilize then after about 10 sec de pitch goes about 30º down. It still working all axes but the compensations wasnt correct because the drone recognizes this 30º pitch down like the neutral point.
So after many many IMU and Gimbal calibrations , board replacements etc i discover that if i power the drone without de main controller board the gimbal works normally.

mefistotelis, 2019-10-23:
What if you won't connect it to RC?

guewald, 2019-10-23:
Same issue, with an without rc... 
Already try to refresh firmwares too..

mefistotelis, 2019-10-23:
If you're really into it, you could capture UART communication between FC and the gimbal - that would allow to later identify the packet which causes such reaction.

guewald, 2019-10-24:
Unfortunately i dont have this level of experience. Just know how to do basic readings with serial interfaces and putty.
I have an drone repair shop here in Brazil... we are upgrading our skills to be able to fix this tipe of issues as finding parts is becoming a very hard task.

If there is an simple way to capture this communications i can do this and place here to you guys, the real specialists !!!!

mefistotelis, 2019-10-24:
All you really need is to know where to solder the USB-to-TTL. Then you just run [comm_serial2pcap.py](https://github.com/o-gs/dji-firmware-tools#comm_serial2pcappy) to start capturing, and turn the drone on. Or you can just store raw data from serial into binary file, with any tool, and it can be later converted to PCAP format.

I am not very familiar with Ph4 though - don't know which test pads have that UART.

guewald, 2019-10-24:
Unfortunately my skills don't goes to this level =/

guewald, 2019-10-24:
https://forum.dji.com/thread-157997-1-1.html   some guys with the same issue here

guewald, 2019-10-28:
I have already identify that this issue is caused by the IMU gimbal board (the one is located behind the CMOS sensor board). Put another IMU board here and the gimbal works well but with another board im unable to calibrate and the yaw gets out of center.

mefistotelis, 2019-10-28:
That's good to know. Thanks for sharing.

I assume this is the board you replaced:
https://github.com/o-gs/dji-firmware-tools/wiki/WM330-Gimbal-Master-Control-board

guewald, 2019-10-28:
This is the Phantom 4 Standard board... The Phantom 4 Advanced is very similar. 
Do you guys believe that this is an hardware or firmware issue? 
If this is an hardware issue i of this boards here to get components and fix it

mefistotelis, 2019-10-28:
I'd vote hardware issue. Hard to be completely sure though.

Gimbal MCU works, so its programming is not damaged; but the area with calibration data could have some strange values stored.

But some kind of HW problem is more likely to cause such behavior.

guewald, 2019-10-28:
Do you have an ideia wich component can cause this situation?

mefistotelis, 2019-10-28:
I only have guesses. My guess would be the IMU (MP66), or one of the passive components near it.

guewald, 2019-10-28:
I will try to replace it and notify you guys if it work.

guewald, 2019-10-28:
Replaced the mp66 and unfortunately the issue still the same

robxs1, 2019-11-07:
Gimbal ESC & Yaw can cause similar issue

pratikalwaysrocks, 2019-11-19:
Hello all,

I am not in any way technically advanced like you guys but I do have some input.

While I used to buy P4P Yaw motor from Aliexpress, they used to put something like this in product description:

**Tips 3:

If the gimbal camera doesn’t fix in the center after install the Y motor, please follow the steps below to solve the problem.

Step 1: Measure the deviation angle of the gimbal camera;

Step 2: Measure and mark making on the counterclockwise direction of the metal plate (the one for motor, above the shock-absorbing plate, next to the flat cable);

Step 3: Drill a screw hole with an electric drill on the metal plate. (If the gimbal camera goes to the left deviation of 45 degrees, then please drill a hole of 45 degrees to the right.)

Step 4: Install the Y motor.**

Now I don;t know where to drill the hole or will it help in anyway but I thought to share it to you guys for any advancement in the procedure.

I really hope it helps

guewald, 2019-11-19:
Thanks for your collaboration but this yaw issue is very common and easy to solve. The issue in question is about pitch motor.

Clearvisionsenthil, 2021-12-16:
[image]

How to Read & write the firmware of the p4pro imu board?

openworldcontato-spec, 2026-10-05:
voce tem foto do outro lado dessa placa???