o-gs/dji-firmware-tools #130: Ambarella video processing unit on MMini

CarlBeek, 2019-11-13:
In the interests of keeping the wiki updated (specifically [this page](https://github.com/o-gs/dji-firmware-tools/wiki/WM160-Main-Processing-Core-Board#parts)), the Ambarella IC on the MM main board is a H22A35. Interestingly enough this IC is 4k@60 H265 capable.

[Ambarella H22A35 description](https://3vpstm1hc6e52739x31131p2-wpengine.netdna-ssl.com/wp-content/uploads/H22-Product-Brief.pdf)

[image]

[Sauce](https://mavicpilots.com/threads/mavic-mini-a-look-inside.75814/)

mefistotelis, 2019-11-13:
Thanks; updated the wiki.

I image using full power on that tiny drone would fry it quite fast. So this will be probably easy to unlock, but quite dangerous.

CarlBeek, 2019-11-13:
Awesome, love the project here! I agree that 4k@60 H265 is probably too much, but I think there may be headroom to push the default without overworking anything.

closed 2019-11-13 by CarlBeek

CarlBeek, 2019-11-13:
I think the board handles interfacing with the microSD too. I'm hoping the pads next to it are the JTAG header (I'd be (less) happy with a UART interface or SPI too).

mefistotelis, 2019-11-13:
In older drones, there was always an UART with standard shell on it.