o-gs/dji-firmware-tools #134: Old openssl issues decrypting dji firmware

Toomanyplanes, 2019-12-06:
I am not a software engineer. I just fly widebody jets worldwide but Github really caught my attention on fixing the Phantom 3 Professional. I am a self starter and not a whiny "please help me" because I don't do my own research. I have re-programmed via UART or FTDI upper gimbal boards and interface boards for the GL300A RC. My comment here is to say that Unbutu  Linux has version 1.1.1 of openssl and to decrypt oldr Dji firmware updates it does require md5 via the older openssl 1.1.0 version. Your Wiki files are extremely important and VERY appreciated but if anyone uses Ubuntu and it has the openssl V1.1.1 it WILL NOT WORK to decrypt Dji firmware via the md5 parameter. It comes out as garbage. I had to download and use openssl V01.01.00.

mefistotelis, 2019-12-06:
I see, you're referring to DM36x firmware which is OpenSSL encrypted.

Encryption algorithm does not change between OpenSSL versions; but default values of some parameters can change. Maybe it's a matter of changing digest algorithm which converts password into key? It did changed at some point in OpenSSL, from MD5 to SHA256. You can use the old algorithm with "-md md5".

Can you link to the wiki which fails to mention that?

Toomanyplanes, 2019-12-06:
There was no mention of using -md md5. That was very clear. I tried sha256 and it would not take that. when I used -md md5 the terminal response was  "*** WARNING : deprecated key derivation used. Using -iter or -pbkdf2 would be better." This is why I had to use the original openssl you mentioned in the code on the wiki guide. Once I used the older version, it worked fine but the newer build of Ubuntu includes the newer version of openssl and it will return that warning error and if I included the  "-pbkdf2" parameter (the -iter was a no go) it would turn out a tar.gz file but it was corrupted and would not read. I know not many people actually decrypt and use these files as the thread is, what, 2 years old or so? Just thought I would mention this so others might get confused as to why the terminal was putting out a tar.gz file that was unusable.

mefistotelis, 2019-12-06:
No reproduction. Issue caused by user error:
```
# openssl version
OpenSSL 1.1.1c  28 May 2019

# file P3X_FW_V01.11.0030_m0800.bin
P3X_FW_V01.11.0030_m0800.bin: openssl enc'd data with salted password

# openssl des3 -md md5 -d -k Dji123456 -in P3X_FW_V01.11.0030_m0800.bin -out P3X_FW_V01.11.0030_m0800_decrypted.tar.gz
*** WARNING : deprecated key derivation used.
Using -iter or -pbkdf2 would be better.

# file P3X_FW_V01.11.0030_m0800_decrypted.tar.gz
P3X_FW_V01.11.0030_m0800_decrypted.tar.gz: gzip compressed data, last modified: Wed Jul 17 09:57:10 2019, from Unix, original size modulo 2^32 7741440
```

Wikis for both m0800 and m1300 properly include decryption examples with MD5 param. No changes needed.

closed 2019-12-06 by mefistotelis

Toomanyplanes, 2019-12-07:
OK. I thought openssl was stuck because des3 is being retired. Just tried what's shown above and it is working. Thank you for the explaination and help.