o-gs/dji-firmware-tools #136: DJI GL300A RC interface board repair anomaly

Toomanyplanes, 2019-12-12:
I recently had an original DJI GL300A RC that had the DM368 and it's NAND memory corrupted due to shelf life issues (yes, it was still in the box and plastic!). I hooked up a UART (FTDI) and analyzed it with PuTTY and it had verified loss of the NAND boot image. (U-Boot and the primary kernel) so I used your resources to re-program the NAND from a user named "Quaddamage" on Phantom Pilots and got it working again with the iPad. HOWEVER, the DJI GO app says it needs a firmware update as it is at FW V01.02.006 and the update must be done with a USB stick. It completly IGNORED the USB stick (used a few different ones) albeit it was reading it for about 45 seconds and shut off. I extracted the firmware from RC FW V01.02.0006 and loaded it onto the NAND and no luck with that to accept an upgrade to V01.03.0020 ( the DJI GO app upgrade only). I was under the impression that the interface board handled the USB or DJI GO app firmware upgrade ability but whether or not the UBL1 or U-Boot is pre- or Post- DJI app upgrade it will not allow upgrade. The workaround was to install an HDMI module (I also used a GL300A interface board that had been currently upgraded) and upgrade the RC through the DJI GO app. That was successful in upgrading the RC main board but the interface board still thinks an upgrade is required! It does work just fine but if the firmware is upgraded, the interface board WILL NOT ALLOW IT! Anyway, this is an interesting problem and I really don't know if this "bug" can be fixed...

mefistotelis, 2019-12-13:
Make sure to read this:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-via-SD-card#flashing-rc-by-official-package-on-usb-stick

Now assuming you did know all that already and still have the issue:
The root FS I provide probably has firmware around 1.07 on it, can't remember anymore. Kernel and u-boot has minimal changes between FW versions and can't be the reason of what you described.

Toomanyplanes, 2019-12-13:
Well, that's interesting. My personal opinion is that there is some other piece of programming that stores the firmware version on that interface board that is telling the RC that it is still on FW V01.02.0006. It does work OK. Don't get me wrong. The really annoying thing is that the software is seeing something that says it still needs an upgrade. BTW, I have another untouched GL300A that has the same issue in the DJI box. I have not messed with it yet. Any ideas???

mefistotelis, 2019-12-13:
If you're interested in various discussions on drone repairs, you should join #hardware channel on Slack. Hopefully the invite here is valid - see slack link on top:
https://dji.retroroms.info/

Toomanyplanes, 2019-12-13:
I went there and it said link to invite is no longer valid...Thank you for the info. Looks like a great site!

Toomanyplanes, 2019-12-14:
Please send the invite again! It looks interesting!

mefistotelis, 2019-12-14:
Here's a valid link:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

Toomanyplanes, 2019-12-14:
Got it! Thanks!

closed 2019-12-14 by Toomanyplanes