o-gs/dji-firmware-tools #138: GL300F - controller works, but it is not possible to connect any USB device

petr222, 2020-01-18:
Hello, does anyone have experience with this problem, please ?

GL300F controller - works, charges, control drone, but there is no reaction to USB 
connection to anything - tried several phones, tablets, several cables. Tried several
USB boards which works in other controllers. All buttons react (beep), but controller
reset using C1, C2 and PHOTO does not work.

Regarding to previous owner - antenna mod kit was installed and after flex cables were
connected and disconnected several times, usb devices cannot be connected anymore.

USB modules provide 5V on output, but mobile device dont even charge from it.

Phantom 3 module P01213.02 should be compatible as one was already in controller
and it is propably original from DJI (sold with controller).

Is there a way to reflash firmware without working USB module ?
Is there UART output on board to check from terminal if there is a error message ?
Is there common problem like burned fuse, which can be checked ?

petr222, 2020-01-18:
OK, messing with flex cable get charging icon on phone, still no connection.

Looks like these connectors are not very reliable, as USB module weight itself
is enough to pull out flex cable from locked connector.

mefistotelis, 2020-01-18:
Fixing the connector will hopefully solve it.

If not - I don't have much experience with GL300F; but in GL300b/c, the USB output was handled by Cypress controller. So in case connections are ok and still device isn't detected, I'd check Cypress (assuming that chip is still there in GL300F). Not sure how though - would require reading Programming Guide for the chip. It has a very simple uC, so might not even have UART.

petr222, 2020-01-18:
GL300F is very similar to GL300b/c, damaged flex connector look like cause
as "it stopped working after flex was connected and disconnected several times".

After lifting flex up on the motherboard side, phone is now charging from controller battery,
Sadly my continuity probe is too big to fit inside connector to measure continuity
between flex cable and pads on motherboard.

Will change the connector, if it do anything - look like standard 30 pin flex connector.

GL300E controller (for parts, cannot turn on) is also on the way. I hope someday I can test
my Phantom 4 Pro which I repaired 2 years ago. Only working controller is needed.

pawelsky, 2020-01-18:
Noticed on my GL300B that when you press the C1+C2+Play combo (D-DDDD beeps) the controller is no longer visible by the app (RC slave mode?). To fix it you need to press C1+C2+Shutter combo (D-DDD beeps), release, switch the RC off, then switch it back on and it works fine again.

petr222, 2020-01-18:
C1+C2+Shutter combo do nothing, only C1+C2+Video record disconnect
controller from drone.

Other way to reset controller is from app, which is not possible without USB connection.

USB board end is holding well when locked, motherboard end is loose and slight touch 
to flex cable from side start pulling cable from locked connector, looks like these connectors
will not survive more cable connect/disconnect cycles.

pawelsky, 2020-01-18:
Did you try connecting to PC Assistant using the micro-USB port?

petr222, 2020-01-18:
No reaction, even after pulling flex cable in several directions.

petr222, 2020-01-18:
OK, after inspecting it under microscope - right side of this connector is cracked,
thats why flex cable is almost loose on mainboard end.

Now to find, order and solder replacement connector a we will see after that.

GL300E is on its way, that will be another challenge as it should be dead and only screen works.

pawelsky, 2020-01-19:
One not regarding key combos - depending on the firmware version (at least on GL300B) they are triggered a bit differently. If just pressing and holding the whole combo doesn't work start with pressing C1 and C2 first until you hear D-DD sound, and then (while still holding C1 and C2) press the third key (e.g. Shutter or Play).

There is also another combination (at least on GL300B) that puts the RC into forced firmware upgrade mode: switch the RC off, then press C1, C2, Rec and Wheel together and hold. While still holding then switch the RC.

petr222, 2020-01-19:
If anyone interested, it look like it is this connector: FH34S-32S-0.5SH

18mm is correct, 32 pins is correct

Mating cycles 20 explain why it is cracked.

petr222, 2020-01-19:
Or maybe XF2M32151AR100, it is 1,5mm wider, but insert size and pitch is the same.

 XF2M32151AR100 it is possible to buy locally, for FH34S-32S-0.5SH it is only one seller on AliExpress and is only selling 20 pcs.

petr222, 2020-01-22:
XF2M32151AR100 is the right one - larger, but pins and pin distance is correct - because of thicker plastic, it should be more reliable.

GL300E arrived today - cannot turn on, after power is pressed, all LEDs are on, but it is not possible to hold power as it start beeping and then turn off.