o-gs/dji-firmware-tools #139: Set/Get on MMINI only partially working ...

T69, 2020-01-20:
... any chance to fix this?

Description:
If I list all parameters, there're unknown packets, which are dropped, and I don't know if this is ok or not:

...
repared binary packet:
55 11 04 92 0a 03 d7 9c 40 03 e1 00 00 f6 04 d1 57
Sending packet...
Waiting for reply...
/dev/tty.usbmodem14103: Packet type 36 damaged - no data recognized; 4 bytes
Retrieved 1 packets (79b), dropped 1 fragments (4b)
Received response packet:
55 4f 04 39 03 0a d7 9c 80 03 e1 00 00 00 00 f6 04 08 00 04 00 cd cc cc 3e cd cc cc 3d 66 66 66 3f 67 5f 63 6f 6e 66 69 67 2e 6d 6f 64 65 5f 67 65 6e 74 6c 65 5f 63 66 67 2e 6c 69 66 74 5f 65 78 70 5f 6d 69 64 5f 70 6f 69 6e 74 00 da dc
Parsed response - DJIPayload_FlyController_GetParamInfoF2017Re:
          status: 0
        table_no: 0
     param_index: 1270
         type_id: 8
            size: 4
       limit_def: 0.4000000059604645
       limit_min: 0.10000000149011612
       limit_max: 0.8999999761581421
            name: b'g_config.mode_gentle_cfg.lift_exp_mid_point'
Prepared request - DJIPayload_FlyController_GetParamInfoByIndex2017Rq:
        table_no: 0
     param_index: 1271

If I do a get with parameter --alt, I do get all desired parameters, like
...
python3 comm_og_service_tool.py /dev/tty.usbmodem14103 MMINI FlycParam get g_config.mode_sport_cfg.tors_gyro_range --alt
...
Response on parameter 1284 indicates end of list despite larger size reported.
g_config.mode_sport_cfg.tors_gyro_range = 150.000000 range = < 5.000000 .. 300.000000 >

If I try to set this parameter
python3 comm_og_service_tool.py /dev/tty.usbmodem14103 MMINI FlycParam get g_config.mode_sport_cfg.tors_gyro_range
Error: Unrecognized response to read parameter value by hash request.

Get/set on other parmeters are working perfectly:
python3 comm_og_service_tool.py /dev/tty.usbmodem14103 MMINI FlycParam get g_config.mode_sport_cfg.vert_vel_up_0
mode_sport_cfg_vert_vel_up = 4.000000 range = < 1.000000 .. 10.000000 >

python3 comm_og_service_tool.py /dev/tty.usbmodem14103 MMINI FlycParam set g_config.mode_sport_cfg.vert_vel_up_0 7
mode_sport_cfg_vert_vel_up = 7.000000 range = < 1.000000 .. 10.000000 >

If I can contribute anything to fix this, just let me know ...

Thanks a lot.

P.S.: There is a chinese website, which can get/set the parameters using a web socket, but I don't trust them, which is why I opened this issue ...

T69, 2020-01-23:
get/set via --alt option works for me ... closing ...

T69, 2020-02-11:
I had some problems setting parameters with a lenght of 1, e.g.
MMINI  FlycParam set bat_enable_smart_bat_landing_protect 0 --alt ends with
Error: this type has no size
Traceback (most recent call last):
  File "comm_og_service_tool.py", line 1628, in <module>
    main()
  File "comm_og_service_tool.py", line 1618, in main
    do_flyc_param_request(po)
  File "comm_og_service_tool.py", line 988, in do_flyc_param_request
    do_flyc_param_request_2017_set_alt(po, ser)
  File "comm_og_service_tool.py", line 968, in do_flyc_param_request_2017_set_alt
    rplpayload = flyc_param_request_2017_write_param_value_by_index(po, ser, paraminfo.table_no, paraminfo.param_index, param_val)
  File "comm_og_service_tool.py", line 500, in flyc_param_request_2017_write_param_value_by_index
    payload.param_value = (c_ubyte * sizeof(payload.param_value)).from_buffer_copy(param_val)
TypeError: this type has no size

I've no idea of python (yet), but the pdb is similar to the one in perl (sort of). Long story short, after changing the definition of DJIPayload_FlyController_WriteParamVal1ByIndex2017Rq to
class DJIPayload_FlyController_WriteParamVal1ByIndex2017Rq(DJIPayload_Base):
  _fields_ = [('table_no', c_ushort),
              ('unknown1', c_ushort),
              ('param_index', c_ushort),
              ('param_value', c_ubyte * 1),
             ]
in comm_mkdupc.py, comm_og_service_tool.py was writing the value finally.

The more I dig into into the toolkit, the more I realize how much work this must have been, so thanks for this great toolkit ...

mefistotelis, 2020-02-11:
Yeah, there's currently a bug in ctypes; should be reported to Python project, but I have no time for that now..

The workaround is to define the type as 1-element array, we should do this as the error won't be fixed any time soon it it's not even reported.

kilrah, 2020-02-12:
Had same prob, changing ` c_ubyte` to ` c_ubyte*1` seems like a working solution indeed