o-gs/dji-firmware-tools #141: P3X device not detected after interrupted update

gdanas87, 2020-02-04:
Total noob at this, used digdat0's method (https://github.com/digdat0/P3I1_FWProject/tree/master/Phantom%203/P3X_FW_V01.11.0020%20height%2Bnfz%20%2B%20speed) for removing nfz, height and speed
accidentally powered down (such an idiot) and now i get constant motor beeping (escs not receiving signal i guess) can't link rc and can't update to latest 11.20 version. Results are as follows:

[03 06][00] device not detected.
[03 05][00] device not detected.
[04 00][00] v1.48.0.0 -> v1.48.0.0
[11 00][00] device not detected.
[11 01][00] device not detected.
[12 00][00] device not detected.
[12 01][00] device not detected.
[12 02][00] device not detected.
[12 03][00] device not detected.
[15 00][00] v1.1.2.0 -> v1.1.2.0
[17 00][00] device not detected.
[17 01][00] device not detected.
[19 00][00] v1.0.8.96 -> v1.0.8.96
[01 00][00] v1.32.5432 -> v1.32.5432
[01 01][00] v1.32.5432 -> v1.32.5432
[08 00][00] v0.13.0.7 -> v0.13.0.7
[09 00][00] v4.1.0.0 -> v4.1.0.0
[00141330]Packet upgrade failed at version checking.

Do I need a new ofdm module and mc with mainboard or will I be able to recover those undetected modules by incrementally rolling back to older firmwares? Also, read here after a long search and research that the debug method is not to be used for downgrading. I thought only with the debug file in place in the sd card along with the fw file I am able to rollback

mefistotelis, 2020-02-04:
> From what I understand, 0305 and 0306 are mainboard (MC) related, 11 through 12 are esc related (again, located in mainboard) and 1701 and 1702 are vps and sonar?

Yes. All those are accessible to Ambarella (m0100) via 0306(+0305, they're both on the same chip). so that's probably the only module that failed.

> will I be able to recover those undetected modules by incrementally rolling back to older firmwares?

No. How do you imagine that could work? You're inserting FW update into m0100. It then asks other modules to enter bootloader mode and accept transmission of new FW binary. If it cannot talk to a module, how could it update?

mefistotelis, 2020-02-04:
You need to either flash Flight Controller manually, or replace it.

gdanas87, 2020-02-04:
> You need to either flash Flight Controller manually, or replace it.

Thank you so much for the above clarifications. I have read so many posts and info you provide on this forum to everyone. I dont believe anyone had a problem like mine before (probably nobody stupid enough to do what I did :p), but I have seen similar issues where you contributed. I have no idea if its a job I could undertake but I am willing to try and to learn. I only have an ftdi adapter available and only have experience flashing cheap old simonk esc's with it by accesing the boot pads on the aforementioned escs. Can you point me to the right direction to study up on how to start about doing this and what software/hardware tools to use? It will either be this (the hard way. But I would love to learn a few things) or I will save up some money and get a mainboard. So you believe only replacement of the mainboard will fix this? Ofdm is not necessary to be replaced since I have a linking issue?

mefistotelis, 2020-02-04:
> Can you point me to the right direction to study up on how to start about doing this and what software/hardware tools to use?

You will need a programmer; for the tools - STLink is used for that.
You will need extracted m0306 firmware, and offset at which to flash it.

All I know about it is on wikis. You will find out which chip you're flashing, and a bit of info about m0305/m0306.

> So you believe only replacement of the mainboard will fix this? Ofdm is not necessary to be replaced since I have a linking issue?

All parts of OFDM are responding. If linking fails - that means FC is required to establish the link. I suspected that from previously encountered issues.

gdanas87, 2020-02-04:
> You will need a programmer; for the tools - STLink is used for that.
> You will need extracted m0306 firmware, and offset at which to flash it.
>
> All I know about it is on wikis. You will find out [...]

Thanks so much once again! I had read a bit in the wiki and thats how I id,d which devices are not detected anymore. I will buy an stlink and try to find out more how to program the MC using the offset you mention

gdanas87, 2020-02-04:
Allright I am reading up in the wiki ... https://github.com/o-gs/dji-firmware-tools/wiki/P3-Flight-Controller-aka-MC
Wow I can only imagine the immense amount of work it took to identify all these components. You have done an amazing job!!! So I will have to flash the m0306 thats zipped into the 11.20 bin fw from what I understand. I will use your dji tools to unpack it and then flash a hex to the actual stm32 f4 chip using the test pads! Can I use stm software to do this? Allright no more pointless stupid questions I guess I am too excited for this, I gotta keep on studying up. Thats easier than having to solder my wires straight to the chip's legs!!! Coming from apm, pixhawk, Betaflight and inav projects it feels familiar to use this chip. Never imagined the phantom uses it as well.

gdanas87, 2020-02-05:
Hello again, so this is a follow up, I see that I can use a serial connection usart to pads pb11 and pb10 for tx and rx, however do you have any idea how to set the stm32f4 chip into a fw upload state? Do I ground the boot0 pad to the gnd pad?

mefistotelis, 2020-02-05:
> you have any idea how to set the stm32f4 chip into a fw upload state? Do I ground the boot0 pad to the gnd pad?

Such things you usually learn from programming guide of the chip. I won't remember more than I put on the wiki.

gdanas87, 2020-02-05:
> > you have any idea how to set the stm32f4 chip into a fw upload state? Do I ground the boot0 pad to the gnd pad?
> 
> Such things you usually learn from programming guide of the chip. I won't remember more than I put on the wiki.

Thank you I looked at the datasheet for the chip, I have to pull high the boot0. I do not know what that means. There's also a guide out there how to flash stm32f4 with uart usb connection which states the same.

gdanas87, 2020-02-05:
I will test tomorrow and let you know how it went! I first have to tackle unpacking the fw file for 0306, I haven't studied up for that here on dji fw tools I remember there's extensive info for it

gdanas87, 2020-02-16:
Hello once again. Can I just flash the PMCAPPFw3.bin from digdat0's github project to the stm32f4 with the stlink or do I have to unpack the p3x firmware with the dji tools?
I will be finally receiving on monday the stlink v2 mini programmer. I will attach to swdio, swclk, 3v3 and gnd pads.
BTW let me make clear how this whole ordeal happened. I have never done such a stupid move in my life before. In my initial post I say that I accidentally powered down and thought that this is what caused the bricking of the mc. Most people can't believe that the p3x can get bricked easily on power down/interruption of fw updates, and they are correct. IT CAN'T. Upon further inspection (and after receiving an mc replacement unit from a local repair shop and soldering it in place) I found out that my p3x is functional once again. I then went to reflash the modded digdat0 PMCAPPFw3.bin. I let it finish succesfully. Then I restart the drone and I get the no-lights esc-motor beep beep forever pattern. 
I sit down and inspect the PMCAPPFw3.bin I inserted in the sd card to update through the notepad. 
THEN i realize my monumental and stupidest/noobest mistake I have ever made in my life!!!!!
all this time, instead of cloning the whole repository of digdat0 and downloading it and using the PMCAPPFw3.bin I want to use for updating, i did the following stupid action:
I was right clicking on the "P3X_FW_V01.11.0020_m0306.bin_MODDED" file and "save as" instead of going into the file and clicking the download button/raw.
So i inspect the "stupid download mistake P3X_FW_V01.11.0020_m0306.bin_MODDED" I updated to the poor p3x which has now bricked not one but two mc's!!!! It shows up on notepad as follows:

<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://github.githubassets.com">
  <link rel="dns-prefetch" href="https://avatars0.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars1.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars2.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars3.githubusercontent.com">
  <link rel="dns-prefetch" href="https://github-cloud.s3.amazonaws.com">
  <link rel="dns-prefetch" href="https://user-images.githubusercontent.com/">
  <link crossorigin="anonymous" media="all" integrity="sha512-ZUjVod2EvYMDbGqRSyW0rpfgBq3i+gnR/4PfrzLsy5f20oIcRfgFQFVKgi3Ztp917bP1K/kdP5q8+nAlJ3+cFA==" rel="stylesheet" href="https://github.githubassets.com/assets/frameworks-6548d5a1dd84bd83036c6a914b25b4ae.css" />
   <link crossorigin="anonymous" media="all" integrity="sha512-6yFc9FSTi8nXD5tHofjHQbrKCUrMwAMlwFuHYHW3o79xPCVJKjuHFqCpr8KAuORzB3RbyhBdv3A4A4rM30suqg==" rel="stylesheet" href="https://github.githubassets.com/assets/github-eb215cf454938bc9d70f9b47a1f8c741.css" />
       <meta name="viewport" content="width=device-width">
    <title>P3I1_FWProject/P3X_FW_V01.11.0020_m0306.bin_MODDED at master · digdat0/P3I1_FWProject</title>
    <meta name="description" content="Contribute to digdat0/P3I1_FWProject development by creating an account on GitHub.">
    <link rel="search" type="application/opensearchdescription+xml" href="/opensearch.xml" title="GitHub">
  <link rel="fluid-icon" href="https://github.com/fluidicon.png" title="GitHub">
  <meta property="fb:app_id" content="1401488693436528">
    <meta name="twitter:image:src" content="https://avatars1.githubusercontent.com/u/24786664?s=400&amp;v=4" /><meta name="twitter:site" content="@github" /><meta name="twitter:card" content="summary" /><meta name="twitter:title" content="digdat0/P3I1_FWProject" /><meta name="twitter:description" content="Contribute to digdat0/P3I1_FWProject development by creating an account on GitHub." />

Can you believe my stupidity???? I basically flashed the mc with html code!!!!!!!
The correctly downloaded digdat0 file is like this in notepad (just the start part of it, I won't copy the hole code):
  emo“•—                ™›    Ηγ΅΅΅΅΅΅±¤·¤½¤Γ¤Ι¤α¤η¤ν¤σ¤ω¤΅¤΅SSYS΅Ο¤΅΅΅΅΅΅΅¤¤“¤¥¤΅΅]¤c¤i¤Υ¤΅΅Ι-§΅΅¥΅C΅΅΅o¤u¤΅΅¥¥¥G΅΅΅΅½UΙUV΅;|¥#¥)¥{¤™¤«¤΅΅΅΅΅΅΅΅¤‡¤΅΅΅΅        ΅π¶πΣϋµπ«πpωπ    !πϊ@``   !πxΰ`½µ½΄χϋΌπΤϊπ7Ήµ7π­gπ.ύhFGπεpH !`A`Α`πψπΔψlH0πjH<0π™hFHπζψ  ½fH !A```Α`pGµπ2π£ψ¨π%ϋ/π½ώaπ`π}ψaπƒώπ ψ πPωπ·ϊπ€ϊπωπ«ϋTH !0RπώχΦhF%πύ OτzqϋρςN ™Pπ…ψP΅R PπψHI `½µOπΙFLahIa`OπΗΰh (ΠΤιBΡOπΊ !α`OπΊ h (Π½θ@ G½µOπbώπuϊ½θ@Ϊη4Hµ@hh±2HOτq0π΅ πςψπώ½θ@πsΊ#πIω½θ@&π¨Ό(H0π«Ώ&I H`pG$I  H`pG"H@hpG H€hpGµ%πόL h()Ρπΰh@ΰ` h@!°ϋρς `ϋ Ή π•ϋ hΐ!°ϋρςϋ Ή πΥό h@!°ϋρςϋ Ήπ”ψaπΖύ½θ@π"Έ½JΣh +Ρ (ΠSh`D ΒιpGΝ ν startup:%d.%d
 wm320v2 Board:"%s"  F―σ € Fχςώµ'πΛψ&πΧύ&πϊύ,πΜπϊ%π6ύ
πϊπ`όNπ§ύπ)ϊXπvϋdπΤψπaϋgπ3ϊπuωQπ½ώ½θ@π`Ύµ$πLψ$π€ψ½θ@#πυΏµ-HπNϊeπΎωeπΛωπόωπγω7πθόeπYω&Hψ°±$H h
πAϊ#πΦω+π¬ώπBϋRπύ½θ@ΣηµRπύπηωeπΎωπΥϋπΣϋπΕωH¤0qπ­ύRπύ½θ@7π0Όµπΰω½θ@?πΞΉµHπ9ώ
H½θ@ 0π3Ύµ	HπώH½θ@ 0π—Ύµχz½θ@εη     p    -ιAF   "iF F(πpϊoπ
€³η‰ !τ@ΰ
FOτ@UOφvΠ"πΠΰ\A@ΰΰZκ!#Υ…κA ²ΰκA[πσΡR²*δΣη΅{ πBΡ  °½θπη`Fωηpµ
F-ν‹Fνύνώ»°ξK
πξk
gπwόόL°ξ@πξ``‰bπ¨ύSμ+Rπ“ψAμphbπύSμ+QπψώAμνρ°ξK
πξk
[... cut ...]
 ωη'ηp„ψ4!hFOπVϋ`x­ψ  ρ4 ­ψp`x­ψ ­ψpΝψ€­ψ` k-uΡ!kΉHΘ` ka"iF kaπ ϊαxIαpΞηώµF  Πΰx (Ρ ΰp %ec¥c •••`x­ψ  ρ4 X­ψ Fp9F@QπZό k.uΠΰoπ
 ώ½ kΕ` ka"iF kaπjϊαxIαpώ½πµF-ν‹¥°#
! πϊP±bx!x’ Oπάωoπ
 %°½μ

Have you ever seen such stupid/noob mistakes being made before in the dji modding scene? I feel so ashamed! I will try to flash the mc's with the stlink v2 programmer I will be receiving in the mail tomorrow (monday), and now at least I have found the culprit of my problems!!!! Do you believe I am approaching this stlink flashing method correctly?

gdanas87, 2020-02-16:
Hello once again. Can I just flash the PMCAPPFw3.bin from digdat0's github project to the stm32f4 with the stlink or do I have to unpack the p3x firmware with the dji tools?
I will be finally receiving on monday the stlink v2 mini programmer. I will attach to swdio, swclk, 3v3 and gnd pads.
BTW let me make clear how this whole ordeal happened. I have never done such a stupid move in my life before. In my initial post I say that I accidentally powered down and thought that this is what caused the bricking of the mc. Most people can't believe that the p3x can get bricked easily on power down/interruption of fw updates, and they are correct. IT CAN'T. Upon further inspection (and after receiving an mc replacement unit from a local repair shop and soldering it in place) I found out that my p3x is functional once again. I then went to reflash the modded digdat0 PMCAPPFw3.bin. I let it finish succesfully. Then I restart the drone and I get the no-lights esc-motor beep beep forever pattern. 
I sit down and inspect the PMCAPPFw3.bin I inserted in the sd card to update through the notepad. 
THEN i realize my monumental and stupidest/noobest mistake I have ever made in my life!!!!!
all this time, instead of cloning the whole repository of digdat0 and downloading it and using the PMCAPPFw3.bin I want to use for updating, i did the following stupid action:
I was right clicking on the "P3X_FW_V01.11.0020_m0306.bin_MODDED" file and "save as" instead of going into the file and clicking the download button/raw.
So i inspect the "stupid download mistake P3X_FW_V01.11.0020_m0306.bin_MODDED" I updated to the poor p3x which has now bricked not one but two mc's!!!! It shows up on notepad as follows (just the start of it, I am copying it with parts missing so it wont mess with my post, its clearly html code):

!DOCTYPE html
<html lang="en">
  <headcharset="utf-8">
 ink rel="ns-prefetc" href="https://githb.githubassets.com"
  <link rel="dns-prefetch" href="https://avatar0.githubusercontent.com"
  <link rel="dns-prefetch href="https://avatars1.githubusercontent.com">
  <lindfghj

Can you believe my stupidity???? I basically flashed the mc with html code!!!!!!!
The correctly downloaded digdat0 file is like this in notepad (just the start part of it, I won't copy the hole code):
  emo“•—                ™›    Ηγ΅΅΅΅΅΅±¤·¤½¤Γ¤Ι¤α¤η¤ν¤σ¤ω¤΅¤΅SSYS΅Ο¤΅΅΅΅΅΅΅¤¤“¤¥¤΅΅]¤c¤i¤Υ¤΅΅Ι-§΅΅¥΅C΅΅΅o¤u¤΅΅¥¥¥G΅΅΅΅½UΙUV΅;|¥#¥)¥{¤™¤«¤΅΅΅΅΅΅΅΅¤‡¤΅΅΅΅        ΅π¶πΣϋµπ«πpωπ    !πϊ@``   !πxΰ`½µ½΄χϋΌπΤϊπ7Ήµ7π­gπ.ύhFGπεpH !`A`Α`πψπΔψlH0πjH<0π™hFHπζψ  ½fH !A```Α`pGµπ2π£ψ¨π%ϋ/π½ώaπ`π}ψaπƒώπ ψ πPωπ·ϊπ€ϊπωπ«ϋTH !0RπώχΦhF%πύ OτzqϋρςN ™Pπ…ψP΅R PπψHI `½µOπΙFLahIa`OπΗΰh (ΠΤιBΡOπΊ !α`OπΊ h (Π½θ@ G½µOπbώπuϊ½θ@Ϊη4Hµ@hh±2HOτq0π΅ πςψπώ½θ@πsΊ#πIω½θ@&π¨Ό(H0π«Ώ&I H`pG$I  H`pG"H@hpG H€hpGµ%πόL h()Ρπΰh@ΰ` h@!°ϋρς `ϋ Ή π•ϋ hΐ!°ϋρςϋ Ή πΥό h@!°ϋρςϋ Ήπ”ψaπΖύ½θ@π"Έ½JΣh +Ρ (ΠSh`D ΒιpGΝ ν startup:%d.%d
 wm320v2 Board:"%s"  F―σ € Fχςώµ'πΛψ&πΧύ&πϊύ,πΜπϊ%π6ύ
πϊπ`όNπ§ύπ)ϊXπvϋdπΤψπaϋgπ3ϊπuωQπ½ώ½θ@π`Ύµ$πLψ$π€ψ½θ@#πυΏµ-HπNϊeπΎωeπΛωπόωπγω7πθόeπYω&Hψ°±$H h
πAϊ#πΦω+π¬ώπBϋRπύ½θ@ΣηµRπύπηωeπΎωπΥϋπΣϋπΕωH¤0qπ­ύRπύ½θ@7π0Όµπΰω½θ@?πΞΉµHπ9ώ
H½θ@ 0π3Ύµ	HπώH½θ@ 0π—Ύµχz½θ@εη     p    -ιAF   "iF F(πpϊoπ
€³η‰ !τ@ΰ
FOτ@UOφvΠ"πΠΰ\A@ΰΰZκ!#Υ…κA ²ΰκA[πσΡR²*δΣη΅{ πBΡ  °½θπη`Fωηpµ
F-ν‹Fνύνώ»°ξK
πξk
gπwόόL°ξ@πξ``‰bπ¨ύSμ+Rπ“ψAμphbπύSμ+QπψώAμνρ°ξK
πξk
gπXό°ξ@πξ` ‰bπύSμ+RπuψSμ+OπϋνθSμ+QπΓAμνζ°ξK
πξk
gπ9ό°ξ@πξ` ‰bπkύSμ+RπVψSμ+OπάAμνΫ°ξK
πξk
gπ ό°ξ@πξ` bπRύSμ+Rπ=ψSμ+Qπ‘AμνΡ°ξK
πξk
gπό°ξ@πξ`ΰbπ9ύSμ+Rπ$ψSμ+OπAμνΖ°ξK
πξk
gπξϋ°ξ@πξ``bπ ύSμ+RπψSμ+Qπ_AμνΌ°ξK
πξk
gπΥϋ°ξ@πξ`νΈ°ξK
πξk
gπΙϋ°ξ@πξ`0hbπϋόSμ+QπζSμ+OπlSμ+QπLώSμ+OπdAμQμQπώh`QμQπώ(`½μ‹p½|µΆHF ρh‘BΡAh™BΠΐι #iFHχ›Hν 
Πν
 ξ 
…ν 
ν
Πν
 ξ 
…ν
•ν 
  „ν 
•ν
„ν
|½ ! aπV½! aπR½µF! aπLύU4a|Ά|	Aκ"α|ψU
Cα@τΐ )Π	!½θ@,π¤½!ωηµF! aπ0ύzI lBΡ  ΰ „ψt ½i ρFlπmΎ)ΠqI9di ρ<FlπbΎlIφηµF ,π&ύ!xAτΐ!BΡ „ψu   ½-ιπG’° &F€ψu`€x``  ` aπ²ύ aπsύoπ
 (}Ρ! aπλό`xlπώώ a (sΠ(!ρ Oπ‡ώρ< (!FOπώff¦fζfOHίψ8&gΔι€JHρe0Δι@ „ψ( ab„ψ) G ΔιHHΔιAH dOπ
±F„ψP D΅(Flπvύ '%>F(!hFOπTώψ8P­2F9FhQπ”<H©Νψ €1Α; ψ 	iF ilπώ qπ¶ώ ! F πΑψMΆ!(€ F π»ψh€¤! F π¶ψ¨€¦! F π±ψθ€¨! F π¬ψ(! F π§ψh¬! F πΆψ ΰcΰ¨®! F π›ψθCΰ       @       @z       7@     @@      @      1@      @      0@      .@      5@Θ Ι HΓΣ5607_a  5607_c  Λ5607_w  (Fχuύ FHΝι
FHΝι
Ρ x
@τΐ! ,π2ό x "@τΐ!	 ,π+ό°(F½θπ‡pµ;N $Δλλΐ χϋώ Πdδ²,σΣ(Fp½Δλλΐ *FCx!F1 Oπƒόςηpµ-N $Δλλΐ•ψu P±(FχΉώ•ψt (Π!(Fχ·ώdδ²,ιΣp½!υηµ " ’#jF πψ½ψ  @Ί½πµ“°FFFF+&Ψ(!hFOπ`ύa
¨Oπ\ύa¨OπXύψ(`H HΝι
¨`ψ ®ψ – 	iF(ilπύ"Fq8FQπ…ώ°π½  kQ Θ ms5607:%d ret:%d spi_id:%d init ok  ΓΛmax_r   IBΊ€RΊRBB€IIΊIB€ΑIΑ€F8@hπ?Ήιη-ιπA†°FF Πΰx (Ρ'ηp!s!hFOπςό`x­ψ  ρ ­ψp`x­ψ ­ψpΝψ€­ψ` h-uΠ	ΰoπ
 °½θπ΅hnHΘ` ha"iF haπ6όαxIαpξη8µ@h$!eJ# ”χΌ8½|µF xaπόϋ ` &&q F#jF
! π΅ψoπ
[... cut ...]
 ωη'ηp„ψ4!hFOπVϋ`x­ψ  ρ4 ­ψp`x­ψ ­ψpΝψ€­ψ` k-uΡ!kΉHΘ` ka"iF kaπ ϊαxIαpΞηώµF  Πΰx (Ρ ΰp %ec¥c •••`x­ψ  ρ4 X­ψ Fp9F@QπZό k.uΠΰoπ
 ώ½ kΕ` ka"iF kaπjϊαxIαpώ½πµF-ν‹¥°#
! πϊP±bx!x’ Oπάωoπ
 %°½μ

Have you ever seen such stupid/noob mistakes being made before in the dji modding scene? I feel so ashamed! I will try to flash the mc's with the stlink v2 programmer I will be receiving in the mail tomorrow (monday), and now at least I have found the culprit of my problems!!!! Do you believe I am approaching this stlink flashing method correctly?

mefistotelis, 2020-02-16:
Yeah, the mistake of getting HTML instead of binary isn't that rare. Most people notice the "raw" button the second time.

Though experienced users do some basic eyeing of the file before flashing - viewing it in any editor and checking whether it looks like what is expected.

You can fix that with STlink; just remember you are not writing the whole firmware, only app part. Do not overwrite the bootloader. Provide offset.

gdanas87, 2020-02-16:
Thanks so much once again. So I program the chip through stm32 st-link utility at memory offset of 0x08000000? I will only program and verify.

mefistotelis, 2020-02-16:
> I program the chip through stm32 st-link utility at memory offset of 0x08000000?

I doubt that. You'd have to study memory map of the chip. Or dump the firmware first, and find where your HTML begins.

mefistotelis, 2020-02-16:
Remember the loader part must fit before the app.

gdanas87, 2020-02-17:
Thanks once again for your input. I cant connect to the MC via stlink. I get can not connect to target from st link utility and I have tried every single combination in the settings. I have soldered wires to swdio, swclk, gnd, and 3.3V and connected the wires to their corresponding pins on the st link. I even upgraded the fw of the stlink usb stick but that did not solve the problem of unable to connect.
[image]
[image]

gdanas87, 2020-02-17:
Is there a chance the stm32f4 has gone into level 2 write protection?

mefistotelis, 2020-02-18:
Don't know the levels; but it's very possible that the read protection is enabled. It was enabled back in the Naza FCs, and they have encryption key in the loader, so they wanted to protect it.

We've got the encryption key from Github, not from duping the hardware; tut that's an unrelated story.

Anyway, if you can't read it - then write it. Just remember you need the offset - you don't want to replace whole content of the internal flash.

mefistotelis, 2020-02-18:
When the 'app' part is mapped into memory, it is visible at offset 0x8020000. Subtract the place where flash mapping starts, and you will get your app offset.

gdanas87, 2020-02-18:
Some progress, at Last. As I write this, I managed to gain access to the chip but not with the stlink. I managed to connect via uart of my trusty old ftdi adapter and the flash loader stm software (using these pads on mc: pb11=rx and I connect it to tx of ftdi, pb10=tx and I connect it to rx of ftdi, 3.3V of ftdi power to 3v3, and I shorted the boot0 to the 3v3 pad). Miraculously it connects but I am informed that the target is read and write protected(Is that why I could not connect with the st link?). In order to remove protection flash has to be wiped, it does not let me otherwise. I went ahead and flashed and sent PMCAPPFW3.BIN just for the heck of it to the sector 5 0x8020000 as you said previously. It was downloaded to the stm32f4 successfully. I also made an upload of the complete flash of the chip to a bin file and it is currently being uploaded to my computer. Not hoping for much since now the loader is gone after the wipe but it was a one way choice there was no alternative. The fact that I got this far with 
 your help, knowledge and intuition is amazing. I only have basic soldering skills and mostly mechanical knowledge (I am not good with coding, hence the reason I Stupidly bricked The MC in the first place). I will keep you informed 
[image]
[image]
[image]
[image]
[image]
[image]

gdanas87, 2020-02-18:
Of course it did not work but this whole process was a step forward. P3X still wont start up properly, 0305, 0306 devices still not detected in the upgrade log after a failed fw upgrade.

here are the contents of whats inside the stm32f4, as of now (after wiping the complete chip-i had to in order to enable read and write), and uploading P3MCAPPFw.bin to it at 0x8020000 offset.
[flashed stm32f4 contents (after complete wipe to read and uploading P3MCAPPfw).txt](https://github.com/o-gs/dji-firmware-tools/files/4221736/flashed.stm32f4.contents.after.complete.wipe.to.read.and.uploading.P3MCAPPfw.txt)

If only there was a way to compile a .bin file with the contents of the loader and the app I would get it working again I am sure. Time to scavenge for a new healthy mc and try to NOT upload html code to it again LOL.
Thank you for all your help!

mefistotelis, 2020-02-18:
Try flashing m0305 at the beginning. If m0305 is the only bootloader, then that's probably where it goes. And if it's not the only one, then we won't recover it anyway.

gdanas87, 2020-02-18:
I will do so. I will have to learn how to extract the m0305 from the p3x 11.20 fw using your tools first. Thanks so much once again, never in my life I imagined that I would get to this raw level of interaction with electronics!!!

mefistotelis, 2020-02-18:
Hm, taking a quick look at the bootloader, it seem to expect address 0x20000000.  From Memory Map:
0x2000 0000 - 0x2001 BFFF: SRAM (112kb aliased by bit-banding)

So that would suggest the m0305 gets loaded to RAM and executed from there?

From "Clocks and startup":
```
3.15 Boot modes
At startup, boot pins are used to select one out of three boot options:
• Boot from user Flash
• Boot from system memory
• Boot from embedded SRAM
The boot loader is located in system memory. It is used to reprogram the Flash memory
through a serial interface. Refer to application note AN2606 for details.
```
So.. where is that "system memory"? Back to mem map:

0x1FFF 0000 - 0x1FFF 7A0F: System memory

But looking into that application note:
```
The bootloader is stored in the internal boot ROM memory (system memory) of STM32 devices.
It is programmed by ST during production.
```
So that is not the m0305 location. that seem to be another bootloader. Also:

```
When readout protection Level2 is activated, STM32 does not boot
on system memory in any case and bootloader can't be executed
```
So that bootloader from system memory was inactive anyway. Also, you need to re-enable the protection to make that thing boot in the same way as before.

So it looks like DJI uses "Boot from embedded SRAM" option. But who loads the bootloader from flash into SRAM?

Looking at "Bootloader selection" graph, the m0305 bootloader needs to be either at 0x08080000 or at 0x08000000. Try both. But not both at once, if 0x08080000 will be programmed then the second address won't be even tried.

mefistotelis, 2020-02-18:
The graph suggests that the Protection level2 isn't really needed. It locks out bootloader, but if boot0=0, that bootloader won't be executed anyway (unless there's no m0305 flashed):
[image]

gdanas87, 2020-02-19:
ok I managed to unpack the latest dji p3x firmware using the cool dji_xv4_fwcon tool. I have located m0305 and m0306. I dont understand which memory address is bank1 and bank2. I am downloading the STM32F427IIH6 datasheet as I write this to figure out which bank refers to which address .
So what memory address am I supposed to flash the m0305 to and what address the m0306?
I tried flashing only the m0305 to 0x800000 but it didn't work (I get no lights on the p3 when starting up=no mc connection-mc is not starting up). I will also flash (without global erase) the m0306 to the 0x8020000

gdanas87, 2020-02-19:
[image]
is this the table i should be referring to? (found in the datasheet)

gdanas87, 2020-02-19:
[image]
btw, these are available when I connect with uart by means of ftdi. I still can't get the st link to connect using the swdio and swclk among with the stlink utility.

gdanas87, 2020-02-19:
https://tinyurl.com/yy4qdmfw this is the datasheet I am using. I am sure its the correct one for the stm32 on the mc

gdanas87, 2020-02-20:
replacement mc board arrived, so I installed it and moved on. Thanks to mefistotelis for his patience and help teaching me how to try and resolve this issue. For anyone who might run into a similar issue in the future, I propose you try and "clone" the stm32f427 memory content without fully erasing it. Yes, the flashloader demonstrator said that there is both read and write protection and there is no way to "look" at the inside of the stm32 target without global erase/wipe but I am sure there must be some way to "clone" a healthy mc's stm32 in order to fix one like mine. Without the memory adresses needed for loading the 0305 (loader) and the 0306 (app) it is only a wild gamble and time consuming process to try all the possible combinations. Btw, my bricked mc is available for anyone who might want to test/play with it and try to figure things out. Just ask and I will send to you for free, you just pay postage. Happy flying guys! and don't be stupid like me. Experience usually causes stupid mistakes like the one I made. Always be rested, focused and careful to read instructions well and check what you're flashing and where you're flasing it to. With the newer products its a little hard to "brick" something, but with older ones like the p3 series that will flash ANYTHING to its insides, as long as the filename is correct, you can do a lot of harm to the drone itself and to yourself by trying and trying to revive things. I leave this experience with a little more knowledge about stm32's and with the reminder that I always need to be careful and focused when working on fw and software things with products!

closed 2020-02-20 by gdanas87

CimoGt, 2020-02-20:
hi, I've been doing research for a few weeks to repair a phantom 3 Pro that spit out
I soldered the MC and I could restart the drone to update but the esc are still not detected
 [00028013][12 00][00] device not detected.
 [00040621][12 01][00] device not detected.
 [00053234][12 02][00] device not detected.
 [00065846][12 03][00] device not detected.
i read the log file with Wireshark avec thanks to mefistotelis dji-firmware-tools-master
hardware tower side seems normal
is there a way to flash them
if you can guide me
I am not very strong in electronics and sucks in programming but I am ready to make the effort it takes to repair it

sorry for my bad english

gdanas87, 2020-02-24:
> hi, I've been doing research for a few weeks to repair a phantom 3 Pro that spit out
> I soldered the MC and I could restart the drone to update but the esc are still not detected
> [00028013][12 [...]

Hello my friend what do you mean when you say the phantom 3 "spit out"?
If they are not detected they are either dead or brick on their chipsets?? You need to provide more info by explaining how this happened and what "spit out" is

mefistotelis, 2020-02-24:
If none of m120x are responding, but m0306 is properly responding, then the copper paths which connects FC with ESCs are probably broken. I mean two specific paths - UART RX and TX.

And I assure you're not a complete noob and you've checked the power supply circuits - voltages on pads with multimeter.

CimoGt, 2020-02-25:
Thank you for your answers gdanas87 and mefistotelis

''Hello my friend what do you mean when you say the phantom 3 "spit out"?''
i mean crached  google traduction :(

I've been researching for days
I just made a discovery , each esc is on an update
the 1200 is on P3X_fw_v01.07.0060  /  the 1201 is on P3X_Fw_V01.10.0090
1202 is on P3X_fw_v01.07.0060 and the 1203 is on P3X_FW_V01.06.0040
thanks to ''mefistotelis'' flashing by module I am doing now an update

at the moment, I'm in front of the PC and the drone is doing updates
I will keep you posted to inform you
many thanks for your help

CimoGt, 2020-02-27:
I'm lost
I managed to update by module but it did not hold
 [00026464][12 00][00] device not detected.
 [00026592][12 01][00] v1.11.0.0 -> v1.10.0.0 need upgrade.
 [00026751][12 02][00] v0.0.0.0 -> v1.10.0.0 need upgrade.
 [00026884][12 03][00] v0.0.0.0 -> v1.10.0.0 need upgrade.

 [01694193][12 00][00] device not detected.
 [01694330][12 01][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
 [01694453][12 02][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
 [01694589][12 03][05] v1.10.0.0 -> v1.10.0.0 need upgrade.
now i only have esc 3 and 4 running, beeping motors
1 and 2 do not want to update and i have on my log :
 [00520378][12 00][00] device not detected.
 [00533068][12 01][00] device not detected.
 [00545647][12 02][00] device not detected.
 [00558231][12 03][00] device not detected.
I know it's an update problem but how to fix it
is there an order to update the Esc?
can we use the P3X_FW_DEBUG file with the update by module
I believe that the problem comes essentially that we install an earlier update and that it causes a conflict
I peeled all the updates and tested it by module on esc but it does not work, can guide me ?

mefistotelis, 2020-02-27:
> I managed to update by module but it did not hold

I highly doubt that.

> can guide me ?

You seem to have ignored previous advice, is there a reason for that?
Do you have any reason to assume this is a software issue?

CimoGt, 2020-02-27:
> You seem to have ignored previous advice, is there a reason for that?
> Do you have any reason to assume this is a software issue?
noo ! i didn't ignored your adviece, I just tried to update by module and it worked the file is transformed from PESC0Fw3.bin to PESC0Fw3_updated.bin but when I wanted to do another update it did not work

>What exactly do you think this will change?
>The FC can't talk to these chips. So why are you considering solutions which require FC to talk to them?
for UART RX7 and TX7 I check it is connected with the dauther motherboard with the grande
what should i check i'm really lost

mefistotelis, 2020-02-27:
Make sure you read this:
https://github.com/o-gs/dji-firmware-tools/wiki/P3X-ESC-center-board

Test the TMS320F280 chips - are they powered? Are they connected to U7_TX/U7_RX? Is the connection stable? Maybe there is a tear visible on the solder somewhere?

If you were able to update something at some point, that might suggest faulty connection which sometimes starts to work, when you pressure the board in a correct way.

If the uCs of ESCs are working, but have no connection to FC, they should start beeping.
In your case the FC can't see them, and they're not beeping. Try to figure out why is that.

CimoGt, 2020-02-27:
> Make sure you read this:
> https://github.com/o-gs/dji-firmware-tools/wiki/P3X-ESC-center-board
> Test the TMS320F280 chips - are they powered? Are they connected to U7_TX/U7_RX? Is the connection stable? Maybe there is a tear visible on the solder somewhere?

I read and I tested the connections
I don't really have the equipment I need but just a multimeter the connections seem normal
I will read again and do other tests

> If the uCs of ESCs are working, but have no connection to FC, they should start beeping.
> In your case the FC can't see them, and they're not beeping. Try to figure out why is that.

that must be the problem, by uCs  you mean IESC2000 (TMS320F28027F) ?
i have tested the mosfet seem normal
thank you for your help and for the time you have given me

mefistotelis, 2020-02-27:
> by uCs you mean IESC2000 (TMS320F28027F) ?

Yes.

CimoGt, 2020-02-27:
> 
> 
> > by uCs you mean IESC2000 (TMS320F28027F) ?
> 
> Yes.

Thank you for your answer tomorrow I will read again `` P3X ESC center board ''
and try to understand