o-gs/dji-firmware-tools #142: GL300F Issue, not working with p4p

vserena82, 2020-02-09:
Hello,

I have a Phantom 4 Pro. I have a GL300F controller that works well.
I have a second GL300F (bought used) that can link to drone, can control it but I can't see video on smartphone.
When I turn on the RC and the GO App, It shows that I am connected to an Inspire 2 even though I do not own an Inspire 2 and no drone is turned on anywhere.
Entering Device (even though it's Inspire 2) I can link RC to Drone via GO App. Problem is tha video feed is not send to phone.
The screen is also presented for the Inspire 2 (as I see a setting for a master and slave remote and I can see options to move landing gear). The RC also does not display any message requiring a firmware upgrade. I Tried a lot of times to reset RC pressing C1+C2+Record+right ISO wheel but it does not reset.
I haven't "advanced" options in RC menu to set RC for Phantom 4 pro.
I'm guessing that the firmware is the issue.
How can I manually get the firmware on the RC?
I cannot upgrade it from the DJI GO App or from the DJI Assistant.

I can see a virtula COM port, there is a way to force fw update using this COM?
Thank you!

[Update menu 1](https://user-images.githubusercontent.com/60800755/74111991-418c1380-4b99-11ea-9097-b652597f605a.jpg)
[image]
[image]

mefistotelis, 2020-02-09:
You should ask on #hardware group on slack.

I don't think there is separate RC firmware depending on AC model. So there are two possibilities:
- Your RC firmware is too new/too old compared to AC
- One of the chips within RC isn't working at all

If you have HDMI output, there's a common issue with NAND for DaVinci getting de-programmed.
If not - I don't know much about this RC, no idea.

vserena82, 2020-02-09:
It hasn't HDMI output and I can't see RC FW version, on DJI GO it is reported "N/A".
AC has last FW version.
I try to change flat cable but nothing has changed.
It should be HW of FW problem.

Is it possible to refresh RC firmware using virtual COM?

mefistotelis, 2020-02-10:
I'm sure there is. DJI always makes it possible. But for how - try Slack.