o-gs/dji-firmware-tools #143: Disable barometer and IMU redundancy limits?

durka, 2020-02-24:
I'm trying to eliminate altitude limits on a Matrice 210. With my setup I don't need a barometer to fly -- I'd be happy if I could just cut it out of the system entirely. What settings should I be looking for? So far I've found:

```
baro_fdi_open_disconnect default = 1 range = < 0 .. 1 >
baro_fdi_open_range default = 1 range = < 0 .. 1 >
baro_fdi_open_abrupt default = 1 range = < 0 .. 1 >
baro_fdi_open_stuck default = 1 range = < 0 .. 1 >
baro_fdi_open_noise default = 1 range = < 0 .. 1 >
baro_fdi_open_walk default = 1 range = < 0 .. 1 >
baro_fdi_open_temper default = 1 range = < 0 .. 1 >
baro_fdi_open_temp_warnning default = 1 range = < 0 .. 1 >
baro_fdi_open_temp_high default = 1 range = < 0 .. 1 >
baro_fdi_open_temp_low default = 1 range = < 0 .. 1 >
baro_fdi_open_disagree default = 1 range = < 0 .. 1 >
baro_fdi_open_invalid_float default = 1 range = < 0 .. 1 >
```

But I still get a lot of errors like "Navigation system error: check redundancy state", "IMU height difference too big", etc. when changing altitude. What settings should I look at in order to eliminate those errors?

mefistotelis, 2020-02-24:
Most people do like this, no barometer change.
Also, I hope you do know how the barometer affects flight, ie. you used a racing drone.

```
diff --git a/flyc_param_infos b/flyc_param_infos
index bf5d20d..c9578e5 100644
--- a/flyc_param_infos	Sat Jun  3 16:13:21 2017
+++ b/flyc_param_infos	Sat Jun  3 16:08:43 2017
@@ -2262,3 +2262,3 @@
 		"maxValue" : 2,
-		"defaultValue" : 1,
+		"defaultValue" : 2,
 		"name" : "g_config.advanced_function.height_limit_enabled_0",
@@ -2601,5 +2601,5 @@
 		"attribute" : 11,
-		"minValue" : 20,
-		"maxValue" : 500,
-		"defaultValue" : 120,
+		"minValue" : 5000,
+		"maxValue" : 5000,
+		"defaultValue" : 5000,
 		"name" : "g_config.flying_limit.max_height_0",
@@ -3414,5 +3414,5 @@
 		"attribute" : 11,
-		"minValue" : 10,
-		"maxValue" : 60,
-		"defaultValue" : 30,
+		"minValue" : 100,
+		"maxValue" : 9000,
+		"defaultValue" : 9000,
 		"name" : "g_config.control.atti_limit_0",
@@ -3427,3 +3427,3 @@
 		"maxValue" : 60,
-		"defaultValue" : 30,
+		"defaultValue" : 55,
 		"name" : "g_config.control.atti_range_0",
@@ -3437,4 +3437,4 @@
 		"minValue" : 10,
-		"maxValue" : 60,
-		"defaultValue" : 15,
+		"maxValue" : 6000,
+		"defaultValue" : 3000,
 		"name" : "g_config.control.avoid_atti_range_0",
@@ -3712,4 +3712,4 @@
 		"minValue" : 1,
-		"maxValue" : 5,
-		"defaultValue" : 3,
+		"maxValue" : 10,
+		"defaultValue" : 10,
 		"name" : "g_config.control.vert_vel_down_bat_limit_0",
@@ -4219,3 +4219,3 @@
 		"maxValue" : 1,
-		"defaultValue" : 1,
+		"defaultValue" : 0,
 		"name" : "g_config.serial_api_cfg.advance_function_enable_0",
```

durka, 2020-02-24:
I don't seem to have those advanced_function settings on the Matrices 200
series. Is there another list for that series?

mefistotelis, 2020-02-24:
I never worked on any matrices. Try asking on slack.
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff