o-gs/dji-firmware-tools #144: Mavic 2 gimbal calibration

isseybor, 2020-03-01:
I saw several videos on how to calibrate the gimbal for the spark and mavic air but I haven't seen any videos on how to calibrate he mavic 2 pro after changing the yaw arm. Is this even possible to do?

mefistotelis, 2020-03-01:
As long as you don't have pairing issue, calibration is the same as for Spark and MA - use comm_og_service_tool.py.

isseybor, 2020-03-01:
What do you mean with pairing issue?

mefistotelis, 2020-03-01:
I mean chips not wanting to talk to each other because of unmatched encryption keys or serial numbers.

isseybor, 2020-03-01:
I ran your software and now the Gimball is calibrated again!
Thanks!

A bit of feedback below.
=> Allthough the calibration went fine, when running the software two errors are reported "Error:Unrecognized response...". 
<<
PS C:\DJI\dji-firmware-tools-master> python .\comm_og_service_tool.py com6 M2P GimbalCalib JointCoarse

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Error: Unrecognized response to calibration command JointCoarse request.
PS C:\DJI\dji-firmware-tools-master> python .\comm_og_service_tool.py com6 M2P GimbalCalib LinearHall

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Error: Unrecognized response to calibration command LinearHall request.
>>

mefistotelis, 2020-03-01:
We'd need to have the reply packet to fix that. Run the command with "-vv" parameter, and paste the full log. It should contain reply as hex string.

ss5690, 2020-03-26:
I have the same issue trying to calibrate Mavic Air - 
"Error: Unrecognized response to calibration command JointCoarse request."
See below with -vv, any help would be appreciated.

C:\DJI\dji-firmware-tools-master>python comm_og_service_tool.py -vv com4 MAVAIR GimbalCalib JointCoarse
Opened com4 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 7c 02 20 04 08 01 27 c9
Sending packet...
Waiting for reply...
Retrieved 3 packets (181b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 7c 02 a0 04 08 01 65 52
Error: Unrecognized response to calibration command JointCoarse request.

islonina, 2020-04-01:
Have a problem calibrating gimbal on M2P original sensor/camera and drivers board, only arm change and got this error

C:\DJI\dji-firmware-tools-master>comm_og_service_tool.py -vv com5 M2P GimbalCalib JointCoarse
Opened com5 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 7d 2f 20 04 08 01 e9 d1
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 7d 2f a0 04 08 01 ab 4a
Error: Unrecognized response to calibration command JointCoarse request.

C:\DJI\dji-firmware-tools-master>python .\comm_og_service_tool.py -vv com5 M2P GimbalCalib LinearHall
Opened com5 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 32 3f 20 04 08 02 6a 68
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 32 3f a0 04 08 01 b3 c1
Error: Unrecognized response to calibration command LinearHall request.

Manojnagapudi, 2020-10-05:
Hello 
I saw several videos regarding mavic 2 pro gimbal issue after changing yaw arm 
I m not able to callibrate the gimbal 
Please can anyone help me

Manojnagapudi, 2020-10-05:
> I ran your software and now the Gimball is calibrated again!
> Thanks!
>
> A bit of feedback below.
> => Allthough the calibration went fine, when running the software two errors are reported [...]

> We'd need to have the reply packet to fix that. Run the command with "-vv" parameter, and paste the full log. It should contain reply as hex string.

Hello 
I have same issue with mavic 2 pro 
Can u help me

heruich, 2021-01-12:
Hi, i have the same problem. i tried to calibration camera for mavic mini how on in this video https://www.youtube.com/watch?v=M09OgSrhqkk&feature=youtu.be&ab_channel=BilalRaybrig but i have errors: 
C:\mini\dji-firmware-tools-master\dji-firmware-tools-master>py comm_og_service_tool.py COM10 -vv MMINI GimbalCalib JointCoarse
Opened COM10 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 db d7 20 04 08 01 57 8c
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a db d7 80 04 08 01 46 98
Error: Unrecognized response to calibration command JointCoarse request.

C:\mini\dji-firmware-tools-master\dji-firmware-tools-master>py comm_og_service_tool.py COM10 MMINI GimbalCalib LinearHall

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Error: Unrecognized response to calibration command LinearHall request.

So , if you know how to fix it - i will be very happy.

History: mavic mini fell from 350m and the camera came off about 1 m. so i bought new body for camera and connect all cables. after install it the camera was very hot ( finger does not endure). i tried to calibration by djy fly but it cant to help me. after that  tried to calibration by python but it was fail. (((

logros, 2021-08-24:
Hi, need help 
with C:\DJI\dji-firmware-tools-master>python comm_og_service_tool.py -vvv com34 MAVAI
R  GimbalCalib JointCoarse
Opened com34 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tun
e its central position. It will take around 15 seconds.

Prepared request - DJIPayload_Gimbal_CalibRq:
         command: 1
Prepared binary packet:
55 0e 04 66 0a 04 99 7c 20 04 08 01 f8 2b
Sending packet...
Waiting for reply...
Received unrelated packet:
55 47 04 4f 03 0a d8 10 00 00 0e 2d 65 72 72 6f 72 3a 20 64 65 76 69 63 65 20 6e
 6f 74 20 69 6e 20 64 6d 20 77 68 65 6e 20 72 65 61 64 20 66 72 6f 6d 20 64 62 2
e 20 64 61 74 61 20 61 74 74 72 3a 31 37 36 00 c3 f5
Retrieved 2 packets (85b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 99 7c a0 04 08 01 ba b0
Error: Unrecognized response to calibration command JointCoarse request.

What i must to do, to work calabration proccess?

Pradipmaism, 2021-12-08:
> As long as you don't have pairing issue, calibration is the same as for Spark and MA - use comm_og_service_tool.py.

Hi from where I can download the tool or do you have link

Pradipmaism, 2021-12-08:
> I ran your software and now the Gimball is calibrated again! Thanks!
>
> A bit of feedback below. => Allthough the calibration went fine, when running the software two errors are reported [...]

From where you download the tool

ArtBaks02, 2022-03-27:
Hi, please help
C:\DJI\dji-firmware-tools-master>py comm_og_service_tool.py --port COM4 -vv WM160 GimbalCalib LinearHall
Opened COM4 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 40 08 20 04 08 02 30 41
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 40 08 80 04 08 01 ba 67
Error: Unrecognized response to calibration command LinearHall request.

Not able to callibrate the gimbal

Amiracle23, 2022-04-18:
I ran the software and an error occurred

./comm_og_service_tool.py --port /dev/tty.usbmodem14105 -vv WM240 GimbalCalib LinearHall
Opened /dev/tty.usbmodem14105 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 6a fa 20 04 08 02 70 64
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 6a fa a0 04 08 01 a9 cd
Error: Unrecognized response to calibration command LinearHall request.

Can you help with this, Performing on a MAC

PabloTri, 2022-06-19:
Hi, I have a mavic air 2, I replaced the gimbal (used), I installed the same camera, and 2 brand new cables.  I am running the firmware too on a Mac and when I run the LinearHall command it does not get any replies:

./comm_og_service_tool.py --port /dev/tty.usbmodem14544305 -vv MAVAIR2 GimbalCalib LinearHall
Opened /dev/tty.usbmodem14544305 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 32 c2 20 04 08 02 b8 8b
Sending packet...
Waiting for reply...
Timeout while waiting for reply.
Retrieved 0 packets (0b), dropped 0 fragments (0b)
Prepared binary packet:
55 0e 04 66 0a 04 32 c2 20 04 08 02 b8 8b
Sending packet...
Waiting for reply...
/dev/tty.usbmodem14544305: Packet type 00 damaged - no data recognized; 75 bytes
Retrieved 117 packets (3003b), dropped 1 fragments (75b)
Received response packet:
55 0e 04 66 04 0a 32 c2 80 04 08 01 32 ad
Error: Unrecognized response to calibration command LinearHall request.

Help would me much appreciated.

rcourtna, 2022-08-29:
Same with my Mini 2:

`dji-firmware-tools-master % ./comm_og_service_tool.py --port /dev/tty.usbmodem14203 -vv WM160 GimbalCalib JointCoarse
Opened /dev/tty.usbmodem14203 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tune its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 3f 7e 20 04 08 01 48 a5
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 3f 7e 80 04 08 01 59 b1
Error: Unrecognized response to calibration command JointCoarse request.`

and

`dji-firmware-tools-master % ./comm_og_service_tool.py --port /dev/tty.usbmodem14203 -vv WM160 GimbalCalib LinearHall
Opened /dev/tty.usbmodem14203 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 88 a7 20 04 08 02 23 e7
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 88 a7 80 04 08 01 a9 c1
Error: Unrecognized response to calibration command LinearHall request.`