o-gs/dji-firmware-tools #153: Lua version used for wireshark dissectors

pcabreracamara, 2020-04-13:
Hi all,

After compiling wireshark with Lua 5.2.4, when wireshark/Lua is loading the plugins, it founds some errors related to local variable declarations at the beginning of each lua file:

```
Lua: Error during loading:
 .../wireshark/plugins/dji-dumlv1-general.lua:4: attempt to index global 'DJI_DUMLv1_PROTO' (a nil value)

```
The same error is repeated when tries to load "dji-dumlv1-camera.lua", "dji-dumlv1-flyc.lua" and "dji-dumlv1-gimbal.lua" files. I guess must be related to specific changes from Lua versions (local variables scope), but I don't know how to adapt the local variables declaration causing these errors:

```
local f = DJI_DUMLv1_PROTO.fields
local enums = {}

```

Could someone please let me know a working version of Lua for these dissectors ? Any help to avoid the "nil value" error would be appreciated.

I just fixed a minor error in the "dji-dumlv1-proto.lua" plugin, adding the absolute paths in the dofile commands (lines 489 to 492), to avoid the error: `dofile: file does not exist`.

Thank you very much,
Pedro

mefistotelis, 2020-04-13:
I'm using Wireshark 3.0.2, which has Lua 5.2.4 as well.

My `init.lua` is:

```disable_lua = false

dofile('dji-dumlv1-proto.lua')

dofile('dji-p3-flyrec-proto.lua')
dofile('dji-p3-batt-proto.lua')
dofile('dji-p3.lua')

dofile('dji-mavic-flyrec-proto.lua')
dofile('dji-mavic.lua')

dofile('dji-spark-flyrec-proto.lua')
dofile('dji-spark.lua')

dofile('dji-write-kml.lua')
```

And in the same folder, I have symlinks to all the LUA files.

pcabreracamara, 2020-04-13:
Thank you, I'm going to update wireshark as mine is version 2.6.4, I'll let you know if this was the cause.

pcabreracamara, 2020-04-13:
Yep, moving lua files out of generic plugins folder and using your `init.lua` file in the personal configuration folder solved the issue, Thank you very much Mefistotelis.

closed 2020-04-13 by pcabreracamara

whal-e3, 2023-10-15:
@mrlnc 's solution also worked on windows 11. Thank you @mrlnc !
For example in `init.lua`...

`dofile('C:\\Users\\<user-name>\\AppData\\Roaming\\Wireshark\\dji\\dji-mavic.lua')`

will do!

fitfrost4, 2024-12-03:
same issue on linux and windows