o-gs/dji-firmware-tools #154: WM240 Modules un-signing

vladsol, 2020-04-13:
It is possible ?

mefistotelis, 2020-04-13:
Only if someone will take time to make it possible, and then publish it.

GitHub-Updates, 2020-12-06:
Can you post how some companies are able to downgrade from the anti rollback 400+ back down to .100? Is this a soldering serial/ UART thing?

Thanks

mefistotelis, 2020-12-06:
That is a know-how of said companies.

DJI now become much more sneaky and are trying to hide the FW code, which makes it harder to find vulnerabilities in public. This makes the modding scene do the same - the findings are hidden, because it was a lot harder even to get root.

While DJI didn't really fixed some of the flaws which we used before, they made additional conditions which makes them impossible to abuse without some soldering.

If you have the drone rooted, you can make it decrypt the firmware for you. Then you can analyze it. But since there is no public way of getting the binaries to analyze, there are also no public conclusions from said analysis. This includes the ways to rollback.

To keep in loop with what the community (including various companies) is doing, join the Slack channel. But don't expect someone will just tell the details to any unknown person who asks. That is a place for sharing, not a place for free support.

See Bin4ry/deejayeye-modder project for an invite.