o-gs/dji-firmware-tools #155: Is it possible to read errors/system logs via serial communication?

webhdx, 2020-04-13:
Hello. 
I've crashed my Mavic Air, it's not linking to the controller nor it shows up in WiFi (even after setting it to WiFi only mode). There is no obvious damage to the PCBs, no water damage, no burnt components etc.

I can't connect it to DJI GO to read drone status but I can connect it via USB to DJI Assistant and also use serial communication via python scripts. I was thinking maybe it is possible to read this data using some commands?

BTW. I also tried downloading update logs but I can't tell how accurate (when they were created) are they. I couldn't find anything suspicious in them.

Any ideas? Thanks in advance, I appreciate all the help ❤️

mefistotelis, 2020-04-13:
Yes, you'll probably be able to read some logs that way. Though I never did that personally.

You should be able to send proper packet to FC to either download logs or enter the mode where logs are presented as another drive in your PC.

There is also a packet which allows reading update logs.

And you can ping various modules to check if they're responding - this will give you an idea about which processor is disconnected or doesn't boot.

Anyway, I don't know details on much of these; try asking on Slack:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

webhdx, 2020-04-13:
Thanks a ton. I owe you at least 5 beers!