o-gs/dji-firmware-tools #157: flash A9 ambarella bootloader with dji P3S board in usb . its possible.

bulbo77, 2020-04-30:
Hi,

I have a problem with my ph3 adv gimbal.
No light, no boot.. i crashed the ambarella chip.

I have discover the ambatools inside the loader firmware with the uart0 interface. You can use this tools when the chip boot on the arm RTOS.

This tool can make erase nand, test nand test ram etc.. of ambarella

My original problem was no linux boot and i had an error 3, firmware upgrade in loop.

I had tested the ambatool and the nand memory in situe. 
For test the program read and write the nand...crash of system pattern etc...

Result: No led, no boot the board is burn?

No ambarella come with a mini booloader in the chip triger by a combinaison of gpio pinout. (on the doc the story talk about boot in usb mode, it's not the method for falsh usb mode.. no found document on the web.. 

I have search on the web and i d'ont found any document on the A9. But A7 and A2.

On the gimbal board we are two gpio pinout on pad available. Out6 and out8

For a slave usb boot :solder the two pad with two wire and solder the two wire on gnd.

power the board with usb cable in the usb gimbal board port.the board must powered with the phantom.

The device is unknown on windows but it's broadcast.

On the web two solution for load the bootloader in the ambarella
Directusb you can find tuto on web
Ambausb you can find tuto on the web
this tools is on the web for save a camera with ambarella chip inside.

Install soft and driver and the board is recogniton to ambarella dev tools

The problem is to found a file descriptor for the chip, there is not a A7 but a A9 the memory adress plan is not same of A7. there are A2, A7 A9 and A12 but no real support for A9

My problem is that i burn my board with a bad use of my usb port

My experiment stop here, if you wont to continue investigation it's open...
I share here my discovery of this if you are the same problem. No boot from ambarella A9 

Thanks for contributor of the tools project and for reverse engeeniring ! 

I make order for a new board but i have learn lot of thing about the dji board

Thanks to o-gs !

mefistotelis, 2020-05-04:
Good research! Will hopefully be continued by someone.

bulbo77, 2020-05-04:
Hi Mefistolis,

Thanks for the shared and your reverse engeeniering about dji and other!
Thanks for schematics etc big help for dji debug !

You can add the decscription of pad in your dji wiki project if you want..

Yes for the A9 doc but i don't find it on the web.. 
In comparaison A2 and A7 pin 6 and 8 are the same description but not same localisation on the ball of the chip. for trigger of usb slave flash i was process by "dicotomi" 

i desolder the ambarella chip, nand and dram of my dji gimbal board, because my board is dead.
I can take photo but the thing that i would like search for gpio out and possible pad on the board.

mefistotelis, 2020-05-04:
Having a photo with chips desoldered will help in the future, when we have the datasheets. Please share.

bulbo77, 2020-05-04:
[image]
[image]
[image]

All track are in under pcb in the layer... but i post the photos

bulbo77, 2020-06-12:
> Hi @bulbo77<https://github.com/bulbo77>
>
> Have you managed to get the proper fw for the A9? Ambausb sees my board, but I don't have any fw to upload with the tool.
>
> Thanks!

Hi Andras

What is your error with ambausb? Me i had a sdram error.. but i think thath the bga sdram chip is unsolder or dead... but perhaps it's false!

my A9 Ambarrela is dead before, but i have seen  we are the good firmware with the diferent soft for download and dispatch etc.
But the problem is the config file for the A9. With Amba USB there are no the description for the A9.I think that it's the real problem.

You musr found in a documentation on the net. ME i have not found on the net..

Best regards

Baxxter38, 2021-07-20:
Hi,

 have a problem with my gimbal top board of my Phantom 3 Advanced.

The voltages on the board are OK exept : 1V2, VCC3V3 and VCC1V8 who stay at 0V (maybe only for phantom 3 pro?)
The LED stay off when I turn on the drone.

When I use Putty on U0_T and GND, I can see "Cortex Boot-Up success" in loop.
[dji P3S.txt](https://github.com/o-gs/dji-firmware-tools/files/6850910/dji.P3S.txt)

I tried Ambausb and my chip Ambarella was detected.

Do you find a solution to use this tool or DirectUSB?

Thank you for your work.

mefistotelis, 2021-07-20:
You're on a good path. The chip works, as it displays something. This means power works.

But it reboots. Either its RAM is damaged, or its nand is de-programmed.
So AmbaUSB is your best bet. Though I doubt anyone here went that path already.
Make sure you report your findings.

Baxxter38, 2021-07-23:
Hi,

I tried today with AmbaUSB and DirectUSB.

With AmbaUSB the chip was reconized like a Ambarella S2(A9) I think is not good and i can't change this.
I think this tool is not suitable.

With DirectUSB I can choose between : a9bub, a9evk and a9bub-LPDDR2. I think the last reference is the most appropriate.

In the TAB Download, I can choose between : Boot Loader (Amboot), Firmware Programming and Kernel.
I'm not sure where I should put the files m100.bin and m101.bin.

With my version of DirectUSB, when I connect to my chip, i have a error message, after some research, the config file is missing for the a9bub-LPDDR2.
I will try to find this file.

mefistotelis, 2021-07-23:
board selection - no idea what 'bub' is, but evaluation kits are development boards for specific chip provided my chip maker. So engineers using the chip, tend to use the tools prepared for evk - as this is what they first learned. On the other hand, DJI did used LPDDR2 in their design. Dunno.

The module m0100 stores several partitions. You can actually extract them with this repo, ie:
```
ls -1 *.a9s
P3X_FW_V01.07.0060_m0100_part_dsp_fw.a9s
P3X_FW_V01.07.0060_m0100_part_lnx.a9s
P3X_FW_V01.07.0060_m0100_part_rfs.a9s
P3X_FW_V01.07.0060_m0100_part_rom_fw.a9s
P3X_FW_V01.07.0060_m0100_part_sys.a9s
```
So I'd say this qualifies as "firmware programming".
For "kernel" - hard to say whether they mean Linux kernel, or RTOS core kernel, or DSP core kernel. Stupid name.
All the 3 things are included as partitions within m0100, so I just wouldn't touch that "kernel" field.

For 'amboot' - it's most likely m0101, though I can't find this name in any of the firmwares. I'd flash it only as last resort, it is unlikely that the bootloader is damaged on your board. m0101 has the same container format as m0100, but there is only one partition in it - it looks like kind of bootloader, most of it is firmware upgrade code. It is based on ThreadX, like the RFS partition from m0100.

I hope DirectUSB sends the data to Ambarella as-is - the container format is based on Ambarella SDK, but it is modified by DJI. So if DirectUSB will try to verify the container before flashing, it will most likely fail.

Also, if there is an option to "reboot to bootloader" or something similar, you should do it before flashing. Ther reboot to normal after flashing is done.

Baxxter38, 2021-07-25:
Hi,

I can't find the configuration file for a9bub-LPDDR2 for DirectUSB.

I may have found another way

It seems to me that the Hero 4 gopro have a similar architecture (A9 with LPDDR2 8Gb)

I find the gopro-usb-tools [https://github.com/evilwombat/gopro-usb-tools](url)
tutorial : [https://www.tapatalk.com/groups/goprouser/how-to-reflash-firmware-onto-a-bricked-hero4-camer-t23149.html](url)

I tried the 1st step which tests the memory and which downloads Linux to install the firmware from the SD card.
[image]
I thinks it's OK.

It would be necessary to modify this tool to adapt it to our firmware, unfortunately I do not have the skills for that.

I also found the tools to prepare the firmware : gopro-fw-tools [https://github.com/evilwombat/gopro-fw-tools](url)
with this tool, i can unpack the firmware m0100.bin to 5 files : 
- section_0
- section_1
- section_2
- section_3
- section_4

Surely the partition of the firmware.

gmtandi, 2021-09-18:
Any update on it? I am experiencing the same, ambarella try to boot, infinite loop.. what does "detecting" stands for?

```
Cortex Boot-Up Success

Cortex freq: 504000000
ARM freq: 132000000
DDR freq: 528000000
Core freq: 216000000
iDSP freq: 264000000
AXI freq: 168000000
AHB freq: 108000000
APB freq: 54000000
UART freq: 1846153
SD0 freq: 15428571
SD1 freq: 15428571
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting
```

mefistotelis, 2021-09-19:
We don't have bootrom code, so we don't know what the firmware does at that moment.
It is nbootrom though, so it definitely doesn't detect any devices specific to DJI board. Most likely it only communicates with RAM and NAND.

Some details:
https://github.com/o-gs/dji-firmware-tools/wiki/Firmware-m0101

gmtandi, 2021-09-19:
I did reflow on the Ambarella and ddr module, no luck.... jost noticed that after sending [ESC] (key) trought serial, i have received "AmbaTools>" shell... there i found some commands..,  show command show no partition, while diag command causes instant "boot loop"

```
AmbaTools> help
help [command]
Get help on [command], or a list of supported commands if a command is omitted.
The following commands are supported:
show
erase
diag
writew
readw

show
show ptb       - flash partition table
show meta      - flash meta table

AmbaTools> show ptb

AmbaTools> show meta
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1
:       sblk:   -1      nblks:   -1

crc32:  0xffffffff

diag
Do Device diag
diag ddr [rotate|pattern] [test loop count(0 for infinite loop)] [Dram Size]diag nand verify all
```

gmtandi, 2021-09-20:
Ok, so i found some tools (ambausb) on this link https://is.gd/bRve3X ...  found out that there are some A9 images that matches what i saw (amboot)... maybe this is the point where i will gonna need a full dump for the flash or like that? i see only S9 (A9) images with DDR3 ram, in this case, what is the DDR of our board?

mefistotelis, 2021-09-20:
> maybe this is the point where i will gonna need a full dump for the flash or like that?

I don't think amba flashing tools accept raw image. They most likely accept partitions. Those you can get by extracting firmware update.

> what is the DDR of our board?

You have major chips identified on the project wiki.

> where can i get .elf image for this device/board?

You want to try reverse engineering? You can convert binary image back to ELF using tools in this repo.
README has an example for ,0100, and for m0101 it would be:
```
./amba_sys2elf.py -vv -e -l 0x6000000 -p P3X_FW_V01.08.0080_m0101_part_lnx.a9s
```

gmtandi, 2021-09-20:
well, i would like to try to reflash the bootloader in order to get the "sd card" updater back to work, however i am not sure if with tools we have by here we are able to... ambausb recognizes only DDR3 ram for A9 ...
[image]

Baxxter38, 2021-09-20:
Hi

Normaly, i think the partitions of the Nand look like this : 

[partition.txt](https://github.com/o-gs/dji-firmware-tools/files/7198113/partition.txt)

I find this in a P3XW, I think is the same for P3S ans P3X.

This board was updated in loop with the SD Card and see can't update the firmware after updating.

When I use the command "diag nand verify all" the diag was succeful, but the nand was cleaned. :-/

I think the best way is to find a full image of the nand.

Can you confirm that the following indications are correct? 
m100 file contain : System Software, DSP uCode, System ROM Data, Linux Kernel, Linux Root FS
and M101 contain : SD Firmware Uptate Code

gmtandi, 2021-09-20:
> When I use the command "diag nand verify all" the diag was succeful, but the nand was cleaned. :-/

Same with me, maybe it is good to advertise to nobody run this command, otherwise it make things even hard...

If anybody had a full dump would be great, but i think it is not simple to dump/reflasy, may need to remove from the board and insert the memory in a external programmer?

mefistotelis, 2021-09-20:
From the screenshot, I can tell you that selecting ELF to run is definitely not what you want.
It should be possible to either write a single partition, or all partitions. They should be written in the same format DJI provides.

The function of reading partition table looks interesting.

RingingResonance, 2022-07-04:
Sorry to wake an old thread, but I've got the same problem with my Ambarella chipset on my zenmuse X3. I ran the 'diag nand' command in the debugger and seemed to have erased my nand flash yay! :D

I've gotten as far as getting directUSB or AmbaUSB to detect the A9 but I'm at a loss as to what I'm supposed to upload. I can't seem to find a copy of amboot anywhere, but I have the m0100 and m0101 firmware and have unpacked it. I wish I could just upload the *.bin file directly to the nand flash, but there seems to be other problems as well as that. Anyone else make any progress on this? I know it's a cheap camera but I'm a broke collage student who got this for free and would love to finish fixing it and hopefully make some extra cash doing drone shots. (I'm a licensed pilot and drone pilot already = also why I'm broke. XD)

I've started my own thread here for fixing my remote controllers which I succeeded at and wanted to document what I did, but then I broke the camera firmware for the Ambarella chip trying to downgrade before upgrading and thought I would add to my post for anyone else who comes along. https://old.reddit.com/r/dji/comments/vphlcy/borked_firmware_on_inspire_1_remote_model_gl658a/