o-gs/dji-firmware-tools #161: Mavic Mini uses comm_og_service_tool can't work normally

9844201, 2020-05-15:
My Mavic Mini uses him and can't work normally

C:\Users\Martian\Downloads\dji-firmware-tools-master>py comm_og_service_tool.py
-vv com6 MMini GimbalCalib JointCoarse
Opened com6 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tun
e its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 89 9a 20 04 08 01 36 79
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 89 9a 80 04 08 01 27 6d
Error: Unrecognized response to calibration command JointCoarse request.

C:\Users\Martian\Downloads\dji-firmware-tools-master>py comm_og_service_tool.py
-vv com6 WM160 GimbalCalib LinearHall
Opened com6 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several tim
es. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 32 a2 20 04 08 02 0b 2a
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 32 a2 80 04 08 01 81 0c
Error: Unrecognized response to calibration command LinearHall request.

9844201, 2020-05-15:
C:\Users\Martian\Downloads\dji-firmware-tools-master>py comm_og_service_tool.py
-vv com6 MMini GimbalCalib JointCoarse
Opened com6 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tun
e its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 89 9a 20 04 08 01 36 79
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 89 9a 80 04 08 01 27 6d
Error: Unrecognized response to calibration command JointCoarse request.

C:\Users\Martian\Downloads\dji-firmware-tools-master>py comm_og_service_tool.py
-vv com6 WM160 GimbalCalib LinearHall
Opened com6 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several tim
es. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 32 a2 20 04 08 02 0b 2a
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 32 a2 80 04 08 01 81 0c
Error: Unrecognized response to calibration command LinearHall request.

C:\Users\Martian\Downloads\dji-firmware-tools-master>py comm_og_service_tool.py
-vv com6 MMini GimbalCalib JointCoarse
Opened com6 at 9600

Info: The Gimbal will move through its boundary positions, then it will fine-tun
e its central position. It will take around 15 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 56 2d 20 04 08 01 24 6f
Sending packet...
Waiting for reply...
Retrieved 13 packets (304b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 56 2d 80 04 08 01 35 7b
Error: Unrecognized response to calibration command JointCoarse request.

C:\Users\Martian\Downloads\dji-firmware-tools-master>py comm_og_service_tool.py
-vv com6 WM160 GimbalCalib LinearHall
Opened com6 at 9600

Info: The Gimbal will slowly move through all positions in all axes, several tim
es. It will take around 30 seconds.

Prepared binary packet:
55 0e 04 66 0a 04 2c 5d 20 04 08 02 43 a4
Sending packet...
Waiting for reply...
Retrieved 1 packets (14b), dropped 0 fragments (0b)
Received response packet:
55 0e 04 66 04 0a 2c 5d 80 04 08 01 c9 82
Error: Unrecognized response to calibration command LinearHall request.

pilgrimage80, 2020-11-03:
I am getting the same error on the Mavic mini. Did you solve the problem?

denverhan, 2020-12-15:
what do u expect?