o-gs/dji-firmware-tools #164: Mavic Mini support for replaced gimbal/camera

OlafWinkler, 2020-06-12:
Hello,
Great tool here, great work!
I had to replace my Mavic Mini's gimbal/camera after a crash. I bought a new system from another disassembled Mavic Mini on Ebay and also a new flat ribbon cable with the compass (?) sensor.
The drone is okay now and flies and records, but there are still errors 40011 and 40021 reported in the app.
Any chance that you can improve this toolset to fix this problem?
I have already run GimbalCalib JointCoarse and LinearHall. Each with bad response error, but the drone performed as expected, still errors 40011 and 40021 persist.
Thanks for the great work.
Best regards,
Olaf

kRtUk, 2020-09-13:
Exactly the same here mate.

yakaracolombia, 2020-10-29:
did you solve this issue? I had the same problem after gimbal+camera replacement in mavic mini

pilgrimage80, 2020-11-03:
> Hello,
> Great tool here, great work!
> I had to replace my Mavic Mini's gimbal/camera after a crash. I bought a new system from another disassembled Mavic Mini on Ebay and also a new flat ribbon cable [...]

Drone gimbal Not Working Did it work even though you gave this error?

There must be some way to get around this error? Before I did this calibration setting it was just giving 40002 error. After calibration error 40011 has been added.

OlafWinkler, 2020-11-03:
My Mavic Mini still shows the errors, but the gimbal is working ok. So, basically, no big thing here. But it would be nice to get rid of the error messages. My understanding is that the gimbal needs to be properly introduced to the main board. But that seems to be a complicated programming mechanism which is probably held secret by DJI in order to encourage customers to use DJI service centers (or to buy other drones next time).

nettunoroberto, 2020-12-06:
removing the error 40011 is easily solved, while for the error 40021 it is necessary to bind the core board with the gimbal using tool dji and vpn

denverhan, 2020-12-16:
try downgrade and upgrage,if any one made it fix pls leave a message here

jefsilv, 2021-04-09:
Here I am with the same problem after replacing the gimbal, the APP presents the error 40021, I am looking for a solution, many say that only official DJI software can solve it.

954426470, 2021-11-06:
> > removing the error 40011 is easily solved, while for the error 40021 it is necessary to bind the core board with the gimbal using tool dji and vpn
> 
> Can you solve 40021 

I can

reggispinheiro, 2021-11-12:
tenho interesse como fazemos 

> > > remover o erro 40011 é facilmente resolvido, enquanto para o erro 40021 é necessário ligar a placa do núcleo com o cardan usando a ferramenta dji e vpn
> > 
> > 
> > Você pode resolver 40021
> 
> eu posso

tenho interesse como podemos fazer ? meu zap 19997001644

954426470, 2021-11-12:
> tenho interesse como fazemos 
> 
> > > > remover o erro 40011 é facilmente resolvido, enquanto para o erro 40021 é necessário ligar a placa do núcleo com o cardan usando a ferramenta dji e vpn
> > > 
> > > 
> > > Você pode resolver 40021
> > 
> > eu posso
> 
> tenho interesse como podemos fazer ? meu zap 19997001644

Sorry, I want to know which contact information 19997001644 is for you? I'm From China

reggispinheiro, 2021-11-12:
> > tenho interesse como fazemos
> > > > > remover o erro 40011 é facilmente resolvido, enquanto para o erro 40021 é necessário ligar a placa do núcleo com o cardan usando uma ferramenta dji e vpn
> > > > 
> > > > 
> > > > Você pode resolver 40021
> > > 
> > > 
> > > eu posso
> > 
> > 
> > tenho interesse como podemos fazer? meu zap 19997001644
> 
> Desculpe, gostaria de saber quais informações de contato 19997001644 são para você? Eu sou da China

sim pra mim mesmo +5519997001644 meu zap, voce pode fazer isso remoto? ai posso fazer transferencia via paypal

reggispinheiro, 2021-11-13:
> sim pra mim mesmo +5519997001644 meu zap, voce pode fazer isso remoto? ai posso fazer transferencia via paypal

Quanto vc cobra pra fazer isso remotamente ?

954426470, 2021-11-13:
> 
RMB 50。I hope you can use a Chinese VPN so that I can operate your computer. Of course, do you suggest using Chinese remote software, because I have not used the contact information you mentioned

954426470, 2021-11-13:
> > > > 特尼奥 ·特内斯 ·科莫 ·法泽莫斯
> > > > > > > 40011 [ 40011 ] 文 · 特恩 · 特 · 特 · 塞恩 · 德维尔， 12 月 40 日 erro 40021 ]n [克莱奥] com o .乌桑多 ·乌马 ·费拉门塔 ·e vpn
> > > > > > 
> > > > > > 
> > > > > > 沃切波德解析器 40021
> > > > > 
> > > > > 
> > > > > 欧波索
> > > > 
> > > > 
> > > > 特尼奥 ·特内斯 ·科莫 ·波德莫斯 ·法泽尔？梅扎普 ·19997001644
> > > 
> > > 
> > > 德斯库佩， 萨伯奎斯 ·信息19997001644圣帕沃奇？欧苏达中国
> > 
> > 
> > 西姆 ·普拉 ·米姆 ·梅斯莫 ·5519997001644 ·梅扎普， 沃斯 ·波德 ·法泽尔 ·伊索 ·泰尔夫·阿夫特·阿夫特paypal
> 
> 特克 · 斯夫 · 埃夫特

RMB 50。I hope you can use a Chinese VPN so that I can operate your computer. Of course, do you suggest using Chinese remote software, because I have not used the contact information you mentioned

reggispinheiro, 2021-11-18:
ja estou usando VPN Chinesa

reggispinheiro, 2021-11-18:
> > > > > 特尼奥 · 特内斯 · 科莫 · 法泽莫斯
> > > > > > > > 40011 [40011] 文 · 特恩 · 特 · 特 · 塞恩 · 德维尔, 12 月 40 日 erro 40021] n [克莱奥] com o. 乌桑 多 · 乌马 · 费 拉门 塔 · e vpn
> > > > > > > 
> > > > > > > 
> > > > > > > 沃切波德 解析器 40021
> > > > > > 
> > > > > > 
> > > > > > 欧波索
> > > > > 
> > > > > 
> > > > > 特尼奥 · 特内斯 · 科莫 · 波德莫斯 · 法泽尔？ 梅扎普 · 19997001644
> > > > 
> > > > 
> > > > 德斯库佩 ， 萨伯奎斯 · 信息 19997001644 圣帕沃奇？ 欧苏达 中国
> > > 
> > > 
> > > 西姆 · 普拉 · 米姆 · 梅斯莫 · 5519997001644 · 梅扎普 ， 沃斯 · 波德 · 法泽尔 · 伊索 · 泰尔夫 · 阿夫特 · 阿夫特 paypal
> > 
> > 
> > 特克 · 斯夫 · 埃夫特
> 
> RMB 50。 Espero que você possa usar uma VPN chinesa para que eu possa operar seu computador. Claro, você sugere o uso de software remoto chinês, porque não usei as informações de contato que você mencionou

Ja estou usando VPN chinesa e software ANYDESK para coneção remoto 
codigo 621529374

954426470, 2021-11-18:
> try downgrade and upgrage,if anyone made it fixed pls leave a message here

me

reggispinheiro, 2021-11-18:
> > tente fazer downgrade e upgrage, se alguém corrigiu, por favor, deixe uma mensagem aqui
> 
> mim

Vou tentar obrigado

954426470, 2021-11-18:
> Ja estou usando VPN chinesa e software ANYDESK para coneção remoto
> codigo 621529374

I'll contact you as soon as I can to get rid of 40021 for you

reggispinheiro, 2021-11-18:
> > > > > > > 特尼奥 · 特内斯 · 科莫 · 法泽莫斯
> > > > > > > > > > 40011 [40011] 文 · 特恩 · 特 · 特 · 塞恩 · 德维尔, 12 月 40 日 erro 40021] n [克莱奥] com o. 乌桑 多 · 乌马 · 费 拉门 塔 · e vpn
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 沃切波德 解析器 40021
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 欧波索
> > > > > > > 
> > > > > > > 
> > > > > > > 特尼奥 · 特内斯 · 科莫 · 波德莫斯 · 法泽尔？ 梅扎普 · 19997001644
> > > > > > 
> > > > > > 
> > > > > > 德斯库佩 ， 萨伯奎斯 · 信息 19997001644 圣帕沃奇？ 欧苏达 中国
> > > > > 
> > > > > 
> > > > > 西姆 · 普拉 · 米姆 · 梅斯莫 · 5519997001644 · 梅扎普 ， 沃斯 · 波德 · 法泽尔 · 伊索 · 泰尔夫 · 阿夫特 · 阿夫特 paypal
> > > > 
> > > > 
> > > > 特克 · 斯夫 · 埃夫特
> > > 
> > > 
> > > RMB 50。 Espero que você possa usar uma VPN chinesa para que possa operar seu computador. Claro, você descartou o uso de software remoto chinês, porque não usei como informações de contato que você mencionou
> > 
> > 
> > Ja estou usando VPN chinesa e software ANYDESK para coneção 
> > remota codigo 621529374
> 
> Entrarei em contato com você assim que puder para me livrar de 40021 para você

teamviewer id 586 702 594 senha efih8mzk

954426470, 2021-11-18:
> teamviewer id 586 702 594 senha efih8mzk

Are you free now

reggispinheiro, 2021-11-18:
> > > > > > > > > 特尼奥 · 特内斯 · 科莫 · 法泽莫斯
> > > > > > > > > > > > 40011 [40011] 文 · 特恩 · 特 · 特 · 塞恩 · 德维尔, 12 月 40 日 erro 40021] n [克莱奥] com o. 乌桑 多 · 乌马 · 费 拉门 塔 · e vpn
> > > > > > > > > > > 
> > > > > > > > > > > 
> > > > > > > > > > > 沃切波德 解析器 40021
> > > > > > > > > > 
> > > > > > > > > > 
> > > > > > > > > > 欧波索
> > > > > > > > > 
> > > > > > > > > 
> > > > > > > > > 特尼奥 · 特内斯 · 科莫 · 波德莫斯 · 法泽尔？ 梅扎普 · 19997001644
> > > > > > > > 
> > > > > > > > 
> > > > > > > > 德斯库佩 ， 萨伯奎斯 · 信息 19997001644 圣帕沃奇？ 欧苏达 中国
> > > > > > > 
> > > > > > > 
> > > > > > > 西姆 · 普拉 · 米姆 · 梅斯莫 · 5519997001644 · 梅扎普 ， 沃斯 · 波德 · 法泽尔 · 伊索 · 泰尔夫 · 阿夫特 · 阿夫特 paypal
> > > > > > 
> > > > > > 
> > > > > > 特克 · 斯夫 · 埃夫特
> > > > > 
> > > > > 
> > > > > RMB 50。 Espero que você possa usar uma VPN chinesa para que possa operar seu computador. Claro, você descartou o uso de software remoto chinês, porque não usei como informações de contato que você mencionou
> > > > 
> > > > 
> > > > Ja estou usando VPN chinesa e software ANYDESK para coneção 
> > > > remota codigo 621529374
> > > 
> > > 
> > > Entrarei em contato com você assim que puder me livrar de 40021 para você
> > 
> > 
> > Teamviewer id 586 702 594 senha efih8mzk
> 
> Você está livre agora

sim

reggispinheiro, 2021-11-18:
> sim

pode conectar

reggispinheiro, 2021-11-19:
> pode conectar

Ola ta por ai ?

> pode conectar
Oi Amigo, preciso falar com voce!

mEfis-afk, 2021-11-27:
> Are you free now

your contact?

tecdrone, 2021-11-27:
> VOCE PODE CHAMa NO WeChat +5581987340010 ou pelo email tecdronedji@gmail.com

alxgacc, 2022-10-14:
Who needs remove error 40011 or 40021, you can write me on WhatsApp +62 812-8430-9143