o-gs/dji-firmware-tools #165: For give me...

KennethMcNutt, 2020-06-27:
But I started right here!! This is the beginning for me!!!  So Please forgive this erroneous post..  
 So here goes Mefisto I came back to my original  Og !!! If you got so clout.. if you still care about the P3?? 
Please help :) 
....You know I have been Crying for YEARS!!

To get someone to incorporate the P3 mods that are just like the other DJI stuff As far as NFZ and Speeds.....

I have supported Steve for as long I have known him.. Everone knows I LOVE Mefisto!! <<<<< ( this guy does NOT get enough love for what he has done )

I shoutout for the OGS' As much as I can..

I support NLD!!

AND I support Brett...

I support Jud..

It is a long list!! And I am sorry that I have NOT mentioned ALL of the MANY that have helped!!!

My true passion is the P3... Yet in all the years I asked and help others...

The P3 has always got forgotten :(

So I have one last call out??

nolimitdronez.com

drone-hacks.com

I don't know if you'll listen??

I have been ignored for years..

But come ON!!

Let's get a GUI based F/W mod out to the P3!!

(video for drama)
https://youtu.be/nqfVoTMEosw

mefistotelis, 2020-06-27:
I don't care for GUIs myself. I prefer people to learn while using my tools.

[Digdat0](https://www.youtube.com/user/digdat0/videos) worked on some wrappers to simplify patching Ph3.

KennethMcNutt, 2020-06-27:
I am up to date on most everything that came after I saw this post... I
have an MP1 Ect... It all started with the P3?? Yet it got lost in the GUI
:(  I am just trying to get my love for the P3 and everyone's  $$ driven
GUI to merge..   Thanks for your ear :)

KennethMcNutt, 2020-06-27:
mef

> I don't care for GUIs myself. I prefer people to learn while using my tools.
> 
> [Digdat0](https://www.youtube.com/user/digdat0/videos) worked on some wrappers to simplify patching Ph3.

Steve (Digdat0) was the next Guy I found after you..  I am a FB base P3 guy...  So I do all my "help" there...  Steve makes vids' I do everything in Posts or PMs'  What I learn here and after...  I just ask that person to Pay it forward :) But I am also NOT the guy that can make a P3 GUI happen!!! I can only ask ;)

coptersafe, 2020-10-05:
Stop crying Kennet you are not along in this world ))  It is me.. Coptersafe) i also be one who start hacking DJI .. but time..so time.

mefistotelis, 2020-10-05:
Having no GUI keeps the 'please bro' crowd away from me. So it's a feature.

I actually started working on one some time ago, even a mobile one, but then came to my senses and deleted everything:
https://www.youtube.com/watch?v=l4stERT4l3Y

closed 2021-03-10 by mefistotelis