o-gs/dji-firmware-tools #167: Mavic Mini gimbal calibration

johnsmotors, 2020-07-15:
Hello guys,

I dont find comport of my mavic mini.
i just have this: USBSTOR\DISK&VEN_DJI&PROD_WM160&REV_1000\0001&0
no com port. so i can't do this command:
python comm_og_service_tool.py com1 MMINI GimbalCalib JointCoarse

Please someone can help me?

ebarcelos, 2020-07-16:
Hi,
Have you already installed the DJI Assistant 2 For Mavic? It contains the necessary drivers to connect your mavic mini to the pc.
Once you install it and connect the drone to the computer, it should appear on your device manager (if you're using windows).
Regards

johnsmotors, 2020-07-17:
hello, good idea.. i ll try it!!
thanks

tranngoclinhbk, 2020-07-31:
> Hi,
> Have you already installed the DJI Assistant 2 For Mavic? It contains the necessary drivers to connect your mavic mini to the pc.
> Once you install it and connect the drone to the computer, it should appear on your device manager (if you're using windows).
> Regards
Not sure if i could post comment here.
I try to run 
py comm_og_service_tool.py com3 mmini GimbalCalib LinearHall

Info: The Gimbal will slowly move through all positions in all axes, several times. It will take around 30 seconds.

Error: Unrecognized response to calibration command LinearHall request.
i saw the gimbal moving, is that ok when this error appear. It not receving report back and pass like what i saw on some youtube video of spark.
thanks in advance

pilgrimage80, 2020-11-03:
> Hello guys,
> 
> I dont find comport of my mavic mini.
> i just have this: USBSTOR\DISK&VEN_DJI&PROD_WM160&REV_1000\0001&0
> no com port. so i can't do this command:
> python comm_og_service_tool.py com1 MMINI GimbalCalib JointCoarse
> 
> Please someone can help me?

You can solve this problem like this. Step 1 Dji Mavic assistant open.
Step 2 Open a drone.
Plug in Usb Cable in 3.20 Seconds. It will appear as USB Serial Com4 in Device Manager.
4. Close Dji Assistant.
5. Enter the relevant command.