o-gs/dji-firmware-tools #168: DJI Phantom 3 Standard gimbal pinouts to UART

andiboyshqipo, 2020-07-21:
Hello, 

I crashed my phantom 3 standard and after changing the ribbon I have no data and no camera preview on mobile. As I read the gimbal board needs to be flashed (p01117.t4). I have tried:
U1_RX and U1TX,
U1_RXT and U1TXT,
U2_RX and U2TX

I am not sure about the boot pinouts, there is only BOOT0 available which I connected to 3.3v but there is no BOOTME. 
Does anyone know where to connect the pins?

Thank you!

mefistotelis, 2020-07-22:
You do not flash boards, you flash specific chips (either their internal flash, or one connected to it). Which chip are you trying to flash?
Did you confirmed that the chip isn't booting?

andiboyshqipo, 2020-07-22:
The readable message that I get from those connections is:

▒▒▒▒▒▒▒ļ▒ǿ? ▒й▒▒▒▒▒▒Ҵ▒!
====== start from loader ======
= 55aa55aa !
app exist !

I cant see the chips, maybe is ROM (Micron 5GM12 D9NLC D8VC) or any Ambarella. I found some details here where you contributed :) 
https://github.com/o-gs/dji-firmware-tools/wiki/WM321-Gimbal-top-board

[image]
[image]