o-gs/dji-firmware-tools #169: not really an issue sorry. gimbal calibration on mavic air 2?

250fpv, 2020-07-26:
hi,
sorry for posting this in the issues section i couldnt find anywhere else to ask the question. can i use this to calibrate the mavic air 2 gimbal after replacement? and does anyone know if the gimbal needs calibrating on the mavic air 2 after replacement with this like on the mavic air 1 or can i just use the in app calibration once i have replaced the gimbal unit?
thanks
james

mingtaoxin, 2020-09-06:
Just find free or buy vpn acces to dji servers  and use djo dealer software.. i buy this software on ebay.. it cost about 150$

tecdrone, 2021-03-22:
> Basta encontrar gratuitamente ou comprar acesso VPN aos servidores dji e usar o software djo dealer .. eu compro este software no ebay .. custou cerca de 150 $

Olá amigo como compra este vpn. Passa o link

954426470, 2021-11-18:
I will, you can contact me

Saeidhhp, 2021-12-19:
> I will, you can contact me

Hello
How can I communicate with you?

954426470, 2021-12-21:
What chat software do you have

Saeidhhp, 2021-12-21:
> What chat software do you have 
> 
> 
> 
> ---Original---
> From: ***@***.***&gt;
> Date: Sun, Dec 19, 2021 12:03 PM
> To: ***@***.***&gt;;
> Cc: ***@***.******@***.***&gt;;
> 
> 
> 
> 
>   
> I will, you can contact me
>   
> Hello
>  How can I communicate with you?

Saeid.hosseinpour01@gmail.com

linoko, 2021-12-29:
> Just find free or buy vpn acces to dji servers and use djo dealer software.. i buy this software on ebay.. it cost about 150$

Hi , please give link for buying this software

mefistotelis, 2023-01-09:
No interesting info here, closing.

closed 2023-01-09 by mefistotelis