o-gs/dji-firmware-tools #170: Phantom 4 V0.00.0000

aeropacio, 2020-08-29:
Hi all! 

My name is patricio, i have 1 Board for phantom 4 standard with problem V0.00.0000 in dji assistant 2, (the live transmission don't working) , i try update with dji assistant 2, and dji go app but it does not work, the usb connection and the driver in my pc is ok, reading about how this work, te server RNDIS is the problem, windows say "network cable disconnected" , but is i unplug the usb, the RNDIS dissapear,so, i solder cable to board in TX,RX and GND(view photo) and connect they to usb-serial adapter, change baudrate to 921000 and power on (only board, not inside the phantom),and uboot system appear! y save all in the serial.txt file,  i cant find nothing in the forum about that problem but someone comment something about the EMMC memory,  in the serial.txt say something about it(could not import file,cannot open, not found,etc )  clearly some interrupt in the update process corrupts memory and destroy the files, i know very little about android and linux , but because i didn't find anything related on internet, i upload the few discoveries i made, hoping that someone with more knowledge than me can give me some idea how to go on maybe,or what i learn to try repair,at the moment i'm reading everything i can about uboot.
I hope someone is useful, and thanks a lot so the wiki and all the information in this git, y repair a lightbridge 1, phantom 2 vision plus and phantom 3 live transmision thanks to you!

ps: Sorry for my very bad english,is my first comment in this lenguage

Link to file in my dropbox:
https://www.dropbox.com/sh/1nuki0ozojl8g2n/AAB_OHZIz7pRZtmxl-3c-ONua?dl=0

mefistotelis, 2020-08-29:
I assume what we're looking at is LC1860 boot.

The issue is damaged initrd partition, sometimes also called initramfs:
`[    0.644863] c0 1 (swapper/0) Initramfs unpacking failed: junk in compressed archive`

aeropacio, 2020-08-30:
Hi megistotelis! , yes, the board is LC1860(sorry)

I'm reading about initrd partition, exist way to fix that without healthy copy? or i can get copy the file for a functional board? you know what version boot linux is? i read this board run android 4.4.4,its possible?  

Or maybe exist any way to keep open rndis0 ([1.304397] c1 83 (ueventd) init: add network interface: rndis0) to update all the firmware via FTP?
(Or am i saying trash haha)

Sorry for so many question, i found a new world to learn about! 

Thanks! 
Patricio

mefistotelis, 2020-08-30:
I have no idea about this board. Just looked at the logs you provided.

You can write most partitions via u-boot, assuming you can enter the bootloader shell.

Android usually has its own set of boot partitions - fastboot mode (aka download mode), recovery mode and normal boot. Normally boot method is chosen by buttons on the mobile device, so here I'd assume there are fields ob the board which work like those buttons.

aeropacio, 2020-09-01:
Hi megistotelis, i found a board with labels on pad's, , not all but this discard many options(none with reference to boot modes), so i try boot in recovery mode, for this i will jump the pads with 10k (or maybe 1kohm) resistor to GND, what you think? the problem is if i need jump two or three pad´s at the same time,how do you need to do on the phones..

also i try enter on the bootloader shell, but obtaing the screenshot in dropbox(in "new files"-> "shell"), i have not idea what is <username debuglevel password>..

mefistotelis, 2020-09-01:
If you are in fastboot mode and connect that chip USB to PC, 'fastboot' binary will list the device.
In 'recovery mode', 'adb' will list the device, I think..

Try joining Slack - maybe in #hardware you will find beople who really worked with these boards:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

sopacaliente, 2020-11-01:
> If you are in fastboot mode and connect that chip USB to PC, 'fastboot' binary will list the device.
> In 'recovery mode', 'adb' will list the device, I think..
>
> Try joining Slack - maybe in #hardware [...]

Yo eh probado esa función no funciona. Hay que encontrar los test point para ingresar el firmware a la placa 3 en 1 del Phantom 4

sopacaliente, 2020-11-02:
> ¡Hola a todos!
> 
> Mi nombre es patricio, tengo 1 placa para el estándar phantom 4 con el problema V0.00.0000 en el asistente dji 2, (la transmisión en vivo no funciona), intento actualizar con el asistente dji 2 y la aplicación dji go pero no funciona , la conexión USB y el controlador en mi PC están bien, leyendo sobre cómo funciona, el servidor RNDIS es el problema, Windows dice "cable de red desconectado", pero si desconecto el USB, el RNDIS desaparece, entonces, sueldo el cable para embarcar en TX, RX y GND (ver foto) y conectarlos al adaptador USB-serial, cambiar la velocidad en baudios a 921000 y encender (solo la placa, no dentro del fantasma), ¡y aparece el sistema uboot! y guarde todo en el archivo serial.txt, no puedo encontrar nada en el foro sobre ese problema pero alguien comenta algo sobre la memoria EMMC, en el serial.txt dice algo al respecto (no se pudo importar el archivo, no se puede abrir, no se encuentra,
> Espero que alguien sea de utilidad, y muchas gracias para que el wiki y toda la información en este git, y repare una transmisión en vivo de lightbridge 1, phantom 2 vision plus y phantom 3 gracias a ti!
> 
> ps: perdón por mi muy mal inglés, es mi primer comentario en este idioma
> 
> Enlace al archivo en mi dropbox:
> https://www.dropbox.com/sh/1nuki0ozojl8g2n/AAB_OHZIz7pRZtmxl-3c-ONua?dl=0

me interesa y quisiera ayudarte, necesito saber si usaste Programador Depurador Debugger St-link V2 Stm8 Stm32.
I am interested and I would like to help you, I need to know if you used Programmer Debugger St-link V2 Stm8 Stm32.

aeropacio, 2020-11-09:
Thanks so much for your answer! 
Yes, i work with STM32 m3 for some of hobbyist proyect, but, i dont find the relation woth the problem? Any help will be appreciated! 

Muchas gracias por tu respuesta, si, trabaje en algunos proyectos por hobby con STM32 y su programador, como se relaciona esto con este problema? Valoro mucho cualquier ayuda! Saludos!

sopacaliente, 2020-11-09:
> Thanks so much for your answer!
> Yes, i work with STM32 m3 for some of hobbyist proyect, but, i dont find the relation woth the problem? Any help will be appreciated!
>
> Muchas gracias por tu [...]

Mira estoy esperando me traigan una placa con ese fallo y voy a probar con el Programador Depurador Debugger St-link V2 Stm8 Stm32, si puedo inyectarle el firmware. Quizas este errado con el metodo pero probare. 
Look, I'm waiting for a board with that fault to be brought to me and I'm going to try with the St-link V2 Stm8 Stm32 Debugger Programmer, if I can inject the firmware into it. Maybe I am wrong with the method but I will try.