o-gs/dji-firmware-tools #172: Dji Phantom 3 advanecd

ivankika1982, 2020-09-10:
Can you explain to me in detail the procedure of connecting the gimbal board to the usb port on the computer.

mefistotelis, 2020-09-10:
Read wikis.

If you want better answer, show some effort on your side.

ivankika1982, 2020-09-10:
I have a problem with image transmission signal on my Phantom 3 advanced. The green LED starts flashing on my gimble board, over and over again. The hidden log file on the sd card. These are the only two lines

[00000679][01 00] Firmware upgrade finished failed (step = 3, err = 0xffffffff).
[00000678][01 00] Firmware upgrade finished failed (step = 3, err = 0xffffffff).

mefistotelis, 2020-09-11:
So you probably want to connect to UART, not to USB.

The micro-USB port is connected to Ambarella as well, if I remember properly. But I never tried doing anything with it.

ivankika1982, 2020-09-11:
I need instructions on how to connect the pins on the gimbal board to the TTL-USB converter. I want it to get a UART log file from the Ambarella chip. I guess I should use the UO_R and UO_T pins, but I would ask you for a more detailed explanation because with a beginner in this. Which pins on the gimble should I connect and with which pins on the TTL-USB converter.

mefistotelis, 2020-09-11:
The board gets 12V power, but immediately converts it to 5V. So connect power to 5V and GND. Remember you need more than 500mA current - no low power usb.

For UART - if you connect it incorrectly nothing bad will happen, if only you will use proper levels of logic signals (3.3V). Remeber you need GND for UART as well.

ivankika1982, 2020-09-12:
**Is it like this? At what point should I connect the 5v power supply.**

[image]

mefistotelis, 2020-09-12:
T is transmit, R is receive. Typically you want to receive what the other side it transmitting.

ivankika1982, 2020-09-12:
Where on the gimbal board to connect 5v and gnd?

ivankika1982, 2020-09-12:
Like this?

[image]

mefistotelis, 2020-09-12:
Yes.

ivankika1982, 2020-09-25:
[image]

I get these lines, where am I wrong?

ivankika1982, 2020-09-25:
Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.25.5432] Build Sep 23 2016 17:03:19

[00000329][CA9_0] [DJI] Stop WDT.

[00000329][CA9_0] [DJI] Firmware updating......

[00000329][CA9_0] Card In Slot.
[00000630][CA9_0] SD clock source, frequency = 216000000
[00000630][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000653][CA9_0] start to check media
[00000657][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000673][CA9_0] Packet Detected / c:\P3S_FW_V01.10.0090.bin /.

[00000676][CA9_0] [DJI] Try Ddr Param 0.

[00000676][CA9_0] Back up partion data ......
[00022827][CA9_0] Image "System Software" is found!
[00022827][CA9_0] Image "DSP uCode" is found!
[00022827][CA9_0] Image "System ROM Data" is found!
[00022827][CA9_0] Image "Linux Kernel" is found!
[00022827][CA9_0] Image "Linux Root FS" is found!
[00022827][CA9_0] Start firmware CRC check...

[00022827][CA9_0] Checking System Software
[00027302][CA9_0] Verifying image CRC ... done
[00027302][CA9_0]       length:         9154368
[00027302][CA9_0]       crc32:          0x845822c8
[00027302][CA9_0]       ver_num:        1.0
[00027302][CA9_0]       ver_date:       2016/9/23
[00027302][CA9_0]       img_len:        9154112
[00027302][CA9_0]       mem_addr:       0x0e600000

[00027302][CA9_0] Checking DSP uCode
[00029206][CA9_0] Verifying image CRC ... done
[00029206][CA9_0]       length:         3889408
[00029206][CA9_0]       crc32:          0x4f07273c
[00029206][CA9_0]       ver_num:        1.0
[00029206][CA9_0]       ver_date:       2016/9/23
[00029206][CA9_0]       img_len:        3889152
[00029206][CA9_0]       mem_addr:       0x0ff00000

[00029206][CA9_0] Checking System ROM Data
[00034532][CA9_0] Verifying image CRC ... done
[00034532][CA9_0]       length:         11624704
[00034532][CA9_0]       crc32:          0xc6dfedab
[00034532][CA9_0]       ver_num:        1.0
[00034532][CA9_0]       ver_date:       2016/9/23
[00034532][CA9_0]       img_len:        11624448
[00034532][CA9_0]       mem_addr:       0x03c00000

[00034532][CA9_0] Checking Linux Kernel
[00036963][CA9_0] Verifying image CRC ... done
[00036963][CA9_0]       length:         5038864
[00036963][CA9_0]       crc32:          0xf6a51577
[00036963][CA9_0]       ver_num:        1.0
[00036963][CA9_0]       ver_date:       2016/9/23
[00036963][CA9_0]       img_len:        5038608
[00036963][CA9_0]       mem_addr:       0x00508000

[00036963][CA9_0] Checking Linux Root FS
[00050003][CA9_0] Verifying image CRC ... done
[00050003][CA9_0]       length:         27001088
[00050003][CA9_0]       crc32:          0x65014d0d
[00050003][CA9_0]       ver_num:        1.0
[00050003][CA9_0]       ver_date:       2016/9/23
[00050003][CA9_0]       img_len:        27000832
[00050003][CA9_0]       mem_addr:       0x00001000

[00050477][CA9_0] Program "System Software" to NAND flash ...
[00053701][CA9_0] Initial bad block. <block 209>
[00053731][CA9_0] Done
[00053934][CA9_0] Program "DSP uCode" to NAND flash ...
[00055298][CA9_0] Done
[00055899][CA9_0] Program "System ROM Data" to NAND flash ...
[00059967][CA9_0] Done
[00060230][CA9_0] Program "Linux Kernel" to NAND flash ...
[00062010][CA9_0] Done
[00063414][CA9_0] Program "Linux Root FS" to NAND flash ...
[00072818][CA9_0] Initial bad block. <block 888>
[00072866][CA9_0] Done
[00073198][CA9_0] [DJI] Done!

[00073198][CA9_0] [DJI] Reboot......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 43200000
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x0 0x789aedcb
System WARM boot.
Continue to booting up ......

---------------------------------------------
    Cortex Boot-Up Success

Cortex freq: 504000000
ARM    freq: 17600000
DDR    freq: 396000000
Core   freq: 216000000
iDSP   freq: 264000000
AXI    freq: 168000000
AHB    freq: 108000000
APB    freq: 54000000
UART   freq: 1846153
SD0    freq: 30857142
SD1    freq: 24000000
---------------------------------------------

Start WDT timer OK.
Magic: 0x87651234 0x1 0x789aedca
System WARM boot.
Continue to booting up ......

sopacaliente, 2020-11-09:
https://youtu.be/K_rk9YyTyjg

sopacaliente, 2020-11-09:
Espero te ayude saludos

ivankika1982, 2020-11-10:
This tutorial is showing how to flash the Professional version, can i apply this to the Advanced version or is there other flash software?

mefistotelis, 2020-11-10:
Anyone: If you're making a tutorial, EXPLAIN WHAT YOU'RE F#@ING FIXING!

This video is for idiots who have no idea what they're doing. And what the #%$ is "black screen of death"??? Is the scenario within this video, drone is not dead, the camera is not dead, just FPV isn't working.

Anyway, the video shows a very specific repair. It concerns you if:
- You have an issue with DaVinci chip on Ph3 Pro gimbal
- The issue is that NAND connected to DaVinci chip was de-programmed due to high temperature
- The video only shows how to re-flash boatloader, not the whole programming of the NAND

So if your issue is only slightly different, then this video will lead you nowhere.

And btw, this is the proper set of information on which this video is based:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-on-DaVinci-media-processors

ivankika1982, 2020-11-10:
Advanced version does not have a DaVinci chip.

I get these lines of code via UART

1.  [with sd card.txt](https://github.com/o-gs/dji-firmware-tools/files/5519424/with.sd.card.txt)
2. [without sd card.txt](https://github.com/o-gs/dji-firmware-tools/files/5519428/without.sd.card.txt)

mefistotelis, 2020-11-10:
This is Ambarella UART.

The content is quite interesting. It looks like the chip is stuck in firmware update mode. It tries to update firmware, but update file is missing. It never goes beyond bootloader.

But the bootloader seem to be working, which is a good thing.

Extract Ambarella FW module from DJI firmware  package, and place on SD-card. When it will flash thefirmware successfully, it should break the loop.

Details:
https://github.com/o-gs/dji-firmware-tools/wiki/Flashing-firmware-via-SD-card#by-firmware-module-in-camera

ivankika1982, 2020-11-17:
Please help me how to Extract Ambarella FW module from DJI firmware package. I've been searching the internet for a week to come up with a solution on how to do this but I haven't been able to. I’m a beginner at this but I’m really trying to try to finish this to the end. Thank you to everyone who participates in this project.

mefistotelis, 2020-11-17:
Extracting modules from the FW package is pretty straightforward. What is the issue you're having?

ivankika1982, 2020-11-17:
Looks like you didn't understand me. I'm not a programmer, I make aerial videos. But I would like to try this because DJI service wants to change my gimbal board and charge 200e. I have a lot of skills with computers, so with someone's more detailed help, I think I could manage to extract the module from the framework.

ivankika1982, 2020-11-17:
Or maybe there is a download file or a tutorial video somewhere.

mefistotelis, 2020-11-17:
I see. ok. I'm sure if someone is interested in helping you personally, he or she will answer.

ivankika1982, 2020-11-23:
I was able to extract the module. If I understood correctly, it is the m0100 module. I renamed it to PFC300SFw3.bin and put it on the sd card. However, I now get these lines of code:
[UART.txt](https://github.com/o-gs/dji-firmware-tools/files/5586269/UART.txt)

mefistotelis, 2020-11-23:
I don't see any issue there. I mean the bootloader should be followed by application boot, and I can't see any messages printed by the application part. But the bootloader seem to have worked properly.

Not sure if you cut the log, or the application part freezes as soon as the code jumps to it.

ivankika1982, 2020-11-27:
The log is not cut, it repeats the same thing again.
Is there something wrong with this part of the code, specifically I mean the line of code where it says: **Initial bad block**

[00053501][CA9_0] Program "System Software" to NAND flash ...
[00056703][CA9_0] Initial bad block. <block 209>
[00056733][CA9_0] Done
[00056937][CA9_0] Program "DSP uCode" to NAND flash ...
[00058293][CA9_0] Done
[00058897][CA9_0] Program "System ROM Data" to NAND flash ...
[00062943][CA9_0] Done
[00063202][CA9_0] Program "Linux Kernel" to NAND flash ...
[00064971][CA9_0] Done
[00066373][CA9_0] Program "Linux Root FS" to NAND flash ...
[00075726][CA9_0] Initial bad block. <block 888>
[00075774][CA9_0] Done
[00076335][CA9_0] [DJI] Done!

To note that the gimbal board is not connected to the rest of the drone, does that affect the log file?

mefistotelis, 2020-11-27:
Initial bad block is not an issue. Todays systems can handle that. That is why the flashing continues regardless of this message.

If the board is disconnected from CMOS sensor and from OFDM, it should still boot into RTOS. Unless it doesn't have enough power, or it's overheating.

ivankika1982, 2020-11-27:
I supply the board with an external 5v 500mA power supply, but it had the same problem when it was connected to the drone and powered by electricity from the drone.
  Is there a hardware problem? I don't know what else to try to do.

mefistotelis, 2020-11-27:
> I supply the board with an external 5v 500mA power supply

I was powering my Pro board with 2 Amps supply.

If you've provided logs from normal run, why is there FW update at start?
Are you forcing the board to continuously re-flash?

ivankika1982, 2020-11-27:
Let's start from the beginning. I extracted the m0100 module from the DJI firmware (did you tell me to extract that module?) I put it on the sd card, the update started, after the file PFC300SFw3 was automatically renamed
in PFC300SFw3_updated. Attached I am sending you a log from Ambarel UART and a hidden log from the MISC / LOG folder:

[Ambarela UART.txt](https://github.com/o-gs/dji-firmware-tools/files/5609356/Ambarela.UART.txt)
[LOG from SD CARD.txt](https://github.com/o-gs/dji-firmware-tools/files/5609357/LOG.from.SD.CARD.txt)

In the meantime, I plugged in a 2A power supply.

mefistotelis, 2020-11-27:
I see. So it looks like the NAND got re-programmed (from the previous log), but it still doesn't work.

The CPU crashes as soon as it tries to load the firmware:

`--- ARM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc : [<0001134c>]    lr : [<00000000>]
sp : 0006f328  ip : 00000000  fp : 00000000
r10: 0006b400  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000000  r5 : 00000000
r4 : 0006f394  r3 : 00000000  r2 : 00000000
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  M`

So I'd say yes - there must he a hardware issue. The CPU itself seem to be working, but it cannot load the application, even though it seem to have been programmed properly.

If you still want to fix that board - try reflow.

If you give up - there is one experiment which I wonder about - try flashing m0101 the same way you did previously with m0100. It won't make the board work, but I wonder what will actually happen if you do that - whether the new bootloader will overwrite existing one, or just add a second stage.

ivankika1982, 2020-11-27:
Thanks a lot. Only if it's not a problem to give me a path on how to reprogram the CPU.

mefistotelis, 2020-11-27:
> Only if it's not a problem to give me a path on how to reprogram the CPU.

Are you asking for instructions on the m0101 flashing? The sentence is confusing.

You program m0101 in the same way you did with m0100 - extract modules, but instead of P3S_FW_*-m0100.bin use P3S_FW_*-m0101.bin. Rename it to PFC300SFw3.bin like before, and it should get flashed.

ivankika1982, 2020-11-27:
Okay, misunderstanding. I understood something different. I'll try to reflow, so I'll report the result. Should I reflow the whole board or just Ambarela chip?

mefistotelis, 2020-11-27:
The issue might be with Ambarella, its NAND and its RAM - 3 chips.

rsrmdp, 2021-02-11:
The truth is that I could not extract the module, I am trying, I am not a programmer like you, I already have a connection with my UBRT. But I can't extract the m0100 module from DJI firmware, I tried various commands from CMD, to no avail.

rsrmdp, 2021-02-11:
Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.32.5432] Build Sep 21 2016 13:26:46

[00000328][CA9_0] [DJI] Stop WDT.

[00000328][CA9_0] [DJI] Firmware updating......

[00000328][CA9_0] Card In Slot.
[00000629][CA9_0] SD clock source, frequency = 216000000
[00000629][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000663][CA9_0] start to check media
[00000668][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000672][CA9_0] Packet Detected / c:\P3X_FW_V01.11.0020.bin /.

[00000674][CA9_0] [DJI] Try Ddr Param 0.

[00000674][CA9_0] Back up partion data ......
[00012939][CA9_0] Image "System Software" is found!
[00012939][CA9_0] Image "DSP uCode" is found!
[00012939][CA9_0] Image "System ROM Data" is found!
[00012939][CA9_0] Image "Linux Kernel" is found!
[00012939][CA9_0] Image "Linux Root FS" is found!
[00012939][CA9_0] Start firmware CRC check...

[00012939][CA9_0] Checking System Software
[00016332][CA9_0] Verifying image CRC ... done
[00016332][CA9_0]       length:         9269056
[00016332][CA9_0]       crc32:          0x989b36d1
[00016332][CA9_0]       ver_num:        1.0
[00016332][CA9_0]       ver_date:       2016/9/21
[00016332][CA9_0]       img_len:        9268800
[00016332][CA9_0]       mem_addr:       0x0e600000

[00016332][CA9_0] Checking DSP uCode
[00017756][CA9_0] Verifying image CRC ... done
[00017756][CA9_0]       length:         3889408
[00017756][CA9_0]       crc32:          0xd97b99c4
[00017756][CA9_0]       ver_num:        1.0
[00017756][CA9_0]       ver_date:       2016/9/21
[00017756][CA9_0]       img_len:        3889152
[00017756][CA9_0]       mem_addr:       0x0ff00000

[00017756][CA9_0] Checking System ROM Data
[00022015][CA9_0] Verifying image CRC ... done
[00022015][CA9_0]       length:         11624704
[00022015][CA9_0]       crc32:          0xc6dfedab
[00022015][CA9_0]       ver_num:        1.0
[00022015][CA9_0]       ver_date:       2016/9/21
[00022015][CA9_0]       img_len:        11624448
[00022015][CA9_0]       mem_addr:       0x03c00000

[00022015][CA9_0] Checking Linux Kernel
[00023856][CA9_0] Verifying image CRC ... done
[00023856][CA9_0]       length:         5038864
[00023856][CA9_0]       crc32:          0xf6a51577
[00023856][CA9_0]       ver_num:        1.0
[00023856][CA9_0]       ver_date:       2016/9/21
[00023856][CA9_0]       img_len:        5038608
[00023856][CA9_0]       mem_addr:       0x00508000

[00023856][CA9_0] Checking Linux Root FS
[00033742][CA9_0] Verifying image CRC ... done
[00033742][CA9_0]       length:         27001088
[00033742][CA9_0]       crc32:          0x65014d0d
[00033742][CA9_0]       ver_num:        1.0
[00033742][CA9_0]       ver_date:       2016/9/21
[00033742][CA9_0]       img_len:        27000832
[00033742][CA9_0]       mem_addr:       0x00001000

[00034212][CA9_0] Program "System Software" to NAND flash ...
[00037453][CA9_0] Initial bad block. <block 126>
[00037521][CA9_0] Done
[00037721][CA9_0] Program "DSP uCode" to NAND flash ...
[00039087][CA9_0] Done
[00039679][CA9_0] Program "System ROM Data" to NAND flash ...
[00043756][CA9_0] Done
[00044015][CA9_0] Program "Linux Kernel" to NAND flash ...
[00045795][CA9_0] Done
[00047165][CA9_0] Program "Linux Root FS" to NAND flash ...
[00050410][CA9_0] Initial bad block. <block 581>
[00051508][CA9_0] Initial bad block. <block 606>
[00051553][CA9_0] Initial bad block. <block 608>
[00056604][CA9_0] Done
[00056937][CA9_0] [DJI] Done!

[00056937][CA9_0] [DJI] Reboot......

rsrmdp, 2021-02-11:
in my case it is a phantom 3 pro

mefistotelis, 2021-02-11:
> I already have a connection with my UBRT

UBRT is one of two russian tools for accessing and modifying data in BQ battery chips. The tool is a paid implementation of the features which BQEVSW from Texas Instruments gives for free. Anyway, it has nothing to do with this thread.

> But I can't extract the m0100 module from DJI firmware, I tried various commands from CMD, to no avail.

Then try harder, watch/read tutorials. Or give up.

General hint: It is far easier to get an answer to a question if you ask it.

rsrmdp, 2021-02-11:
sorry if I got confused it's not UBRT, I mean UART. sorry
Tutorials I did not see. I'm reading the README and I'm not going to give up. I'm interested, I'll keep trying.

rsrmdp, 2021-02-11:
D:\Drones\DJI\Liberaciones\dji-firmware-tools>dji_xv4_fwcon.py -vv -x -p P3X_FW_V01.11.0020.bin
Traceback (most recent call last):
  File "D:\Drones\DJI\Liberaciones\dji-firmware-tools\dji_xv4_fwcon.py", line 41, in <module>
    from Crypto.Cipher import AES
ModuleNotFoundError: No module named 'Crypto'

rsrmdp, 2021-02-11:
Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.32.5432] Build Sep 21 2016 13:26:46

[00000327][CA9_0] [DJI] Stop WDT.

[00000327][CA9_0] [DJI] Firmware updating......

[00000327][CA9_0] Card In Slot.
[00000628][CA9_0] SD clock source, frequency = 216000000
[00000628][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000662][CA9_0] start to check media
[00000667][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000671][CA9_0] Can't detect firmware.
[00000741][CA9_0] Firmware  program fail.RM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc : [<0001134c>]    lr : [<00000000>]
sp : 0006f328  ip : 00000000  fp : 00000000
r10: 0006b400  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000000  r5 : 00000000
r4 : 0006f394  r3 : 00000000  r2 : 00000000
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  Mï

rsrmdp, 2021-02-11:
> Are you asking for instructions on the m0101 flashing? The sentence is confusing.
>
> You program m0101 in the same way you did with m0100 - extract modules, but instead of `P3S_FW_*-m0100.bin` use [...]

try this same result
Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.32.5432] Build Sep 21 2016 13:26:46

[00000328][CA9_0] [DJI] Stop WDT.

[00000328][CA9_0] [DJI] Firmware updating......

[00000328][CA9_0] Card In Slot.
[00000629][CA9_0] SD clock source, frequency = 216000000
[00000629][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000663][CA9_0] start to check media
[00000669][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000673][CA9_0] Can't detect firmware.
[00000740][CA9_0] Firmware  program fail.RM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc : [<0001134c>]    lr : [<00000000>]
sp : 0006f328  ip : 00000000  fp : 00000000
r10: 0006b400  r9 : 00000000  r8 : 00000000
r7 : 00000000  r6 : 00000000  r5 : 00000000
r4 : 0006f394  r3 : 00000000  r2 : 00000000
r1 : fff90c5b  r0 : ffffffa4
Flags: NzCv
IRQs on  FIQs on  Mï

rsrmdp, 2021-02-11:
rename FC300XFw.bin or PFC300SFw3.bin

mefistotelis, 2021-02-11:
> ModuleNotFoundError: No module named 'Crypto'

Tried googling that? Maybe someone had this issue before. That's what you always do if you have any error message.

rsrmdp, 2021-02-11:
> ¿Intentaste buscarlo en Google? Quizás alguien haya tenido este problema antes. Eso es lo que siempre hace si tiene algún mensaje de error.

it's solved.
But I had no luck when I inject the m0100.bin which is renamed to PFC300XFw3.bin I get bad blocks
[00034212] [CA9_0] Program "System Software" to NAND flash ...
[00037453] [CA9_0] Initial bad block. <block 126>
[00037521] [CA9_0] Done
[00037721] [CA9_0] Program "DSP uCode" to NAND flash ...
[00039087] [CA9_0] Done
[00039679] [CA9_0] Program "System ROM Data" to NAND flash ...
[00043756] [CA9_0] Done
[00044015] [CA9_0] Program "Linux Kernel" to NAND flash ...
[00045795] [CA9_0] Done
[00047165] [CA9_0] Program "Linux Root FS" to NAND flash ...
[00050410] [CA9_0] Initial bad block. <block 581>
[00051508] [CA9_0] Initial bad block. <block 606>
[00051553] [CA9_0] Initial bad block. <block 608>
[00056604] [CA9_0] Done
[00056937] [CA9_0] [DJI] Donate!

[00056937] [CA9_0] [DJI] Reboot ......

Although in the P3X_FW_LOG_AB:

 [00055336] [01 00] Firmware upgrade finished successfully.
 [00055398] Done.

 [00000674] [01 00] Firmware upgrade finished failed (step = 3, err = 0xffffffff).
 [00000674] [01 00] Firmware upgrade finished failed (step = 3, err = 0xffffffff).
 [00139229] [01 00] Firmware upgrade finished failed (step = 4, err = 0xff).
 [00055256] [01 00] Firmware upgrade finished successfully.
 [00055318] Done.

 [00000673] [01 00] Firmware upgrade finished failed (step = 3, err = 0xffffffff).
 [00054991] [01 00] Firmware upgrade finished successfully.
 [00055053] Done.

When I mount the gimbal on the drone everything remains the same.

mefistotelis, 2021-02-11:
"Initial bad block." is not an error message, more like info/warning. Again, try googling it. It's a feature used in some NANDs.

Still, if you can't re-program your NAND, then the chip is either damaged or doesn't stick to the board properly.

rsrmdp, 2021-03-09:
> I see. So it looks like the NAND got re-programmed (from the previous log), but it still doesn't work.
>
> The CPU crashes as soon as it tries to load the firmware:
>
> ```
> --- ARM DATA ABORT EXCEPTION [...]

I tried:

Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000][CA9_0] [DJI] Start WDT of 20 seconds for boot up......

[00000000][CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000][CA9_0] [DJI] SD Updater Version [v01.32.5432] Build Sep 21 2016 13:26:46

[00000327][CA9_0] [DJI] Stop WDT.

[00000327][CA9_0] [DJI] Firmware updating......

[00000328][CA9_0] Card In Slot.
[00000629][CA9_0] SD clock source, frequency = 216000000
[00000629][CA9_0] SDCalcClk for SD card initiation:: Actual Clk = 140625, pCLK = 128
[00000663][CA9_0] start to check media
[00000667][CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000672][CA9_0] [DJI] Try Ddr Param 0.

[00000672][CA9_0] Back up partion data ......
[00053297][CA9_0] firmware model name doesn't match!
[00053297][CA9_0] firmware header is illegal!
[00053304][CA9_0] [DJI] Try Ddr Param 1.

[00053304][CA9_0] Back up partion data ......
[00105794][CA9_0] firmware model name doesn't match!
[00105794][CA9_0] firmware header is illegal!
[00105800][CA9_0] [DJI] Try Ddr Param 2.

[00105800][CA9_0] Back up partion data ......
[00130067][CA9_0] firmware model name doesn't match!
[00130067][CA9_0] firmware header is illegal!
[00130071][CA9_0] [DJI] Try Ddr Param 3.

[00130071][CA9_0] Back up partion data ......
[00177661][CA9_0] firmware model name doesn't match!
[00177661][CA9_0] firmware header is illegal!
[00177668][CA9_0] [DJI] Try Ddr Param 4.

[00177668][CA9_0] Back up partion data ......
[00229449][CA9_0] firmware model name doesn't match!
[00229449][CA9_0] firmware header is illegal!
[00229455][CA9_0] [DJI] Try Ddr Param 5.

ou program m0101 in the same way you did with m0100 - extract modules, but instead of P3S_FW _ * - m0100.bin use P3S_FW _ * - m0101.bin. Rename it to PFC300SFw3.bin like before, and it should get flashed.
And also change P3S_FW _ * - m0101.bin to PCAMLDRFw3.bin.
Unsuccessfully:
Start WDT timer OK.
Magic: 0x87651234 0x4 0x789aedcf
System WARM boot.
Detecting done ......

[00000000] [CA9_0] [DJI] Start WDT of 20 seconds for boot up ......

[00000000] [CA9_0] [Version 1.0.11824], Oct 16 2015, 14:06:46

[00000000] [CA9_0] [DJI] SD Updater Version [v01.32.5432] Build Sep 21 2016 13:26:46

[00000328] [CA9_0] [DJI] Stop WDT.

[00000328] [CA9_0] [DJI] Firmware updating ......

[00000328] [CA9_0] Card In Slot.
[00000629] [CA9_0] SD clock source, frequency = 216000000
[00000629] [CA9_0] SDCalcClk for SD card initiation :: Actual Clk = 140625, pCLK = 128
[00000663] [CA9_0] start to check media
[00000667] [CA9_0] AmbaSD_InitCard - SD_CARD OK
[00000672] [CA9_0] Can't detect firmware.
[00000766] [CA9_0] Firmware program fail.
--- ARM DATA ABORT EXCEPTION ---
Oops: CPU Exception!
pc: [<0001134c>] lr: [<00000000>]
sp: 0006f328 ip: 00000000 fp: 00000000
r10: 0006b400 r9: 00000000 r8: 00000000
r7: 00000000 r6: 00000000 r5: 00000000
r4: 0006f394 r3: 00000000 r2: 00000000
r1: fff90c5b r0: ffffffa4
Flags: NzCv
IRQs on FIQs on Me