o-gs/dji-firmware-tools #174: GL200a RC bricked after firmware update

BN-B, 2020-09-15:
Hi All,

Does anyone have any experience bringing RC's back from the dead?
My RC died after a botched firmware update. All that works now is the backlight. This is a known issue that DJI solves by replacing the RC's motherboard. Unfortunately, mine is no longer under warranty, so I'm on my own.

I've opened the rc up and I do not mind massacring it in the hope of bringing it back.

Any help would be appreciated.
Thanks

mefistotelis, 2020-09-15:
AFAIK the RC runs on Android, so all the tricks for phones should also work there.

You should try asking that on slack. Invite is here:
https://github.com/Bin4ry/deejayeye-modder#want-to-contribute-or-learn-some-stuff

If you learn anything, we should update the information on the wiki:
https://github.com/o-gs/dji-firmware-tools/wiki/GL200-Main-board

BN-B, 2020-09-15:
Thanks @mefistotelis 

I've check the slack group twice, but there never seems to be anyone online. What's the best time to catch someone?
Do you know anyone who has done this?

mefistotelis, 2020-09-15:
Not exactly sure what you were trying to do.. you don't have to catch anyone there.

Open Slack, browse channels, and join #general and #hardware. Ask a question in one of these channels, and if anyone happens to know something, they will respond. Check back after some time to see the answers.

I see messages added within the last hour, so the community is alive there.

asunder123, 2020-09-20:
The support team has taken a request for a different issue I have raised concerns over, not sure if your issues would be addressed as part of the same . 
Following is a mail thread from the community 
Dear Anand,

Thank you for contacting DJI Technical Support.

We appreciate your research and study regarding the DJI firmware updates codes lapses. Your feedback has been well received.
We really appreciate the time and efforts you have invested in providing these and ideas to us.
Your feedback will be definitely cascaded to the concerned team ( Developers team ) for their further review.
Please stay tuned for more updates through our official website.

Thank you for choosing DJI products.

Best Regards,

Jerome L.
DJI Customer Support
Website: http://www.dji.com/support
YouTube: http://s.dji.com/DJI-Tutorials