o-gs/dji-firmware-tools #175: Can I customize the UBOX-M8N GPS module on Phantom 3 to improve its performance?

gaifeng8864, 2020-09-27:
In the Phantom 3 series UAVs, does the flight controller perform other operations on the GPS module besides increasing the baud rate of the GPS module from 9600 to 115200 after startup, such as setting the GPS output protocol?
Can we customize some settings to improve GPS performance? For example, increase the output frequency of positioning information to 5Hz, or turn on GPS+BD. I am now in China and I am very interested in opening BD positioning. In addition, does the Phantom 3 series drones only use the UBX protocol to communicate with the GPS module?

mefistotelis, 2020-09-27:
Yes, these things can be changed within m0306. Enabling more GNSS types requires adding support of more UBX packet types.

There are such mods for Mavic. You should be able to develop something similar for P3P based on that.
You also need to update U-blox firmware - the drone never does that, and the old stock version lacks some features.

gaifeng8864, 2020-09-27:
> Yes, these things can be changed within m0306. Enabling more GNSS types requires adding support of more UBX packet types.
>
> There are such mods for Mavic. You should be able to develop something [...]

The firmware of UBOX-M8N was upgraded to version 3.01 by the way when I repaired the shell last year. But I didn't expect to change to GPS+BD. This year the BD system completed the global networking, and I began to consider this plan.
According to your research, can the current flight controller firmware correctly parse the UBX data packets output by the M8N in GPS+BD mode?
In addition to changing the baud rate, what settings have been made to the M8N GPS module after the flight controller is started?